lwfinger/rtw89 #275: Random failure on PopOS (Debian Base)

joshfaust, 2023-10-21:
## Issue

Driver enters an error state at non-deterministic times. Wifi becomes unresponsive and at times, not available. (No menu display). It can take several full re-boots to have the wifi chipset re-active (the driver to load properly). 

- **Hardware**: HP Elitebook 645 G9
- `Linux pop-os 6.5.6-76060506-generic #202310061235~1697396945~22.04~9283e32 SMP PREEMPT_DYNAMIC Sun O x86_64 x86_64 x86_64 GNU/Linux`

**Modinfo**:
```
╰─$ sudo modinfo rtw_8852b                                                                                                                                                           1 ↵
filename:       /lib/modules/6.5.6-76060506-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko
license:        Dual BSD/GPL
description:    Realtek 802.11ax wireless 8852B driver
author:         Realtek Corporation
firmware:       rtw89/rtw8852b_fw-1.bin
srcversion:     6E74AB6A8C61CF4D22CD2F0
depends:        rtw89core,cfg80211
retpoline:      Y
name:           rtw_8852b
vermagic:       6.5.6-76060506-generic SMP preempt mod_unload modversions 
```

**kernel.log**
```
Oct 20 09:15:56 pop-os kernel: [  622.693443] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693500] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693561] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693620] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693681] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693738] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693795] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693850] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693906] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.693962] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694019] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694074] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694130] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694186] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694242] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694298] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694354] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694409] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694464] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694519] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694575] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694634] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694694] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694751] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694808] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694863] rtw89_8852be 0000:02:00.0: read rf busy swsi
Oct 20 09:15:56 pop-os kernel: [  622.694919] rtw89_8852be 0000:02:00.0: read rf busy swsi
```

lwfinger, 2023-10-21:
You posted a meaningless modinfo.

Read the section under "Normally, none of these will be needed; however, if you are getting firmware errors, one or both of the disable_aspm_* options may help. Thay are needed when a buggy BIOS fails to implement the PCI specs correctly."

Most recent HP and Lenovo laptops have buggy BIOS code, and need both the disable_aspm_* options!!

joshfaust, 2023-10-22:
> You posted a meaningless modinfo.
>
> Read the section under "Normally, none of these will be needed; however, if you are getting firmware errors, one or both of the disable_aspm_* options may help. [...]

Ah, thank you! I'll do that today.

capdefra05, 2023-10-29:
Hello Larry,
I have exactly the same problem as joshfaust! I own a recent HP Pavilion 15 with a Realtek wifi 6 card which needs your RTL 8852be driver and , on my laptop, when I use kernels over 6.2, I have frequent disconnexions after minutes or hours. At each time, I have to restart the computer to recover wifi connexion.I have read your comments upper but it's impossible for me to modify the file POLICY beacause I have not the permission, even when I try on a Live session . Is'it possible to modify this file and how?
Since more than 6 months, I can correctly use my laptop because this problem with the wifi driver. I have tried with many Linux distros and It's always the same!
Before changing the wifi card, I try with you to have a chance to solve the problem.
Thanks in advance,

Franck

PS: Don't be affraid for my english, it's difficult for me, I speak french and only a little english

lwfinger, 2023-10-29:
Did you try wwhat was suggested above? I guess not. Run the command

sudo nano /etc/modprobe.d/rtw8852be.conf

In that file, add the lines
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89_core disable_ps_mode=y
options rtw89core disable_ps_mode=y

Save the resulting file, and reboot. These options will work for the in-kernel version, and this repo. If they help, the problem is with your BIOS, quite common with late=model HP and Lenovo.

capdefra05, 2023-10-29:
I haven't a file rtw8852be.conf in the folder /etc/modprobe.d.
I have created this file with the four lines you mentionned above.
Is it good?

lwfinger, 2023-10-29:
That is what I told you to do.

capdefra05, 2023-10-29:
It seems to be good! I have no problem since two hours but the test must continue a few hours to be sure that it's good.
In case of one kernel's upgrade (for example, kernel 6.2.0 --->6.2.1--->6.2.5) there is nothing more to do? The file rtw8852be is good for each active kernel?

lwfinger, 2023-10-29:
YES.

closed 2023-10-29 by lwfinger

vryy, 2025-01-01:
> Did you try wwhat was suggested above? I guess not. Run the command
>
> sudo nano /etc/modprobe.d/rtw8852be.conf
>
> In that file, add the lines options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y [...]

This somehow works. Can you briefly explain these parameters?