lwfinger/rtw89 #277: wifi removed afterr todays update commit #45dde70

salimp2009, 2023-11-03:
Hi
After todays update wifi is gone. During the update there were no errors but after the reboot wifi is gone

commit 45dde70
Manjaro  kernel 5.15.133-1

Let me know how i can help to diagnose it. 
my networkcard Realtek RTL8852AE

lwfinger, 2023-11-04:
Sorry, but I have bad news for you, but the driver builds perfectly well under kernel 5.15.0 on my system. That means that Manjaro has modified the API, and I do not support such backports as described in README.md. A search shows that Manjarro is using 6.X kernels. Why have you not upgraded? If it is not possible to upgrade, I could probably fix the build for you, but not at no cost - I have decided to fix it for $50, which is a real deal as I will likely have at least $500 worth of time in the fix. If you want to go this route, please let me know, and I will give you the details for payment, and I will get more info regarding your system.

salimp2009, 2023-11-04:
Thank you for looking into it I just installed Manjaro a year ago and did regular updates I did not select 5.15 on purpose Whatever manjaro releases i did Let me check if i can bump up too kernel 6 or maybe install Arch or Endevour . I use Arch on my desktop Your offer is kind In case i need i will get back to you Thnks or if there is older version the repo i can use it as well

salimp2009, 2023-11-04:
also one thing if Manjaro is playing with Kernel how come it was working until those recent changes I wonder ?