lwfinger/rtw89 #278: Fails to compile on long term kernel 6.1.30

mradcliffe, 2023-11-03:
I was updated the driver today and was unable to compile on a system using kernel 6.1.30 sources.

commit 39b6c53

      rtw89/core.c:2942:66: error: 'struct ieee80211_bss_conf' has no member named 'assoc'

It looks like some lines changed from `KERNEL_VERSION(6, 0, 0)` to `KERNEL_VERSION(6, 3, 0)`
  - https://github.com/lwfinger/rtw89/blame/main/core.c#L2942
  - https://github.com/lwfinger/rtw89/blob/main/ps.c#L280

Changing the directive to `KERNEL_VERSION(6, 1, 0)` allows for the driver to be compiled. I don't know if setting to `6, 0, 0` is appropriate or not like in previous versions.

lwfinger, 2023-11-03:
On my system, the change needed to be >= KERNEL_VERSION(6, 0, 0). Your change would have moved the erroring kernel.

The patches are pushed.

mradcliffe, 2024-02-09:
I forgot to come back and close this issue. Sorry. Thank you for the patch.

closed 2024-02-09 by mradcliffe