lwfinger/rtw89 #280: struct ieee80211_bss_conf’ has no member named ‘power_type’

netikras, 2023-11-05:
Following https://easylinuxtipsproject.blogspot.com/p/realtek.html

```
netikras@netikras-box:~/workspace/rtw89$ make
make -C /lib/modules/5.15.0-86-generic/build M=/home/netikras/workspace/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-86-generic'
  CC [M]  /home/netikras/workspace/rtw89/core.o
  CC [M]  /home/netikras/workspace/rtw89/chan.o
  CC [M]  /home/netikras/workspace/rtw89/mac80211.o
  CC [M]  /home/netikras/workspace/rtw89/mac.o
  CC [M]  /home/netikras/workspace/rtw89/phy.o
  CC [M]  /home/netikras/workspace/rtw89/fw.o
  CC [M]  /home/netikras/workspace/rtw89/cam.o
  CC [M]  /home/netikras/workspace/rtw89/efuse.o
  CC [M]  /home/netikras/workspace/rtw89/regd.o
/home/netikras/workspace/rtw89/regd.c: In function ‘rtw89_reg_6ghz_power_recalc’:
/home/netikras/workspace/rtw89/regd.c:518:39: error: ‘struct ieee80211_bss_conf’ has no member named ‘power_type’; did you mean ‘txpower_type’?
  518 |                 switch (vif->bss_conf.power_type) {
      |                                       ^~~~~~~~~~
      |                                       txpower_type
/home/netikras/workspace/rtw89/regd.c:519:22: error: ‘IEEE80211_REG_VLP_AP’ undeclared (first use in this function)
  519 |                 case IEEE80211_REG_VLP_AP:
      |                      ^~~~~~~~~~~~~~~~~~~~
/home/netikras/workspace/rtw89/regd.c:519:22: note: each undeclared identifier is reported only once for each function it appears in
/home/netikras/workspace/rtw89/regd.c:522:22: error: ‘IEEE80211_REG_LPI_AP’ undeclared (first use in this function)
  522 |                 case IEEE80211_REG_LPI_AP:
      |                      ^~~~~~~~~~~~~~~~~~~~
/home/netikras/workspace/rtw89/regd.c:525:22: error: ‘IEEE80211_REG_SP_AP’ undeclared (first use in this function); did you mean ‘IEEE80211_SEQ_TO_SN’?
  525 |                 case IEEE80211_REG_SP_AP:
      |                      ^~~~~~~~~~~~~~~~~~~
      |                      IEEE80211_SEQ_TO_SN
make[2]: *** [scripts/Makefile.build:297: /home/netikras/workspace/rtw89/regd.o] Error 1
make[1]: *** [Makefile:1909: /home/netikras/workspace/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-86-generic'
make: *** [Makefile:84: all] Error 2

```

```
netikras@netikras-box:~/workspace/rtw89$ cat /etc/lsb-release 
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=21.2
DISTRIB_CODENAME=victoria
DISTRIB_DESCRIPTION="Linux Mint 21.2 Victoria"
netikras@netikras-box:~/workspace/rtw89$ cat /etc/os-release 
NAME="Linux Mint"
VERSION="21.2 (Victoria)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 21.2"
VERSION_ID="21.2"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=victoria
UBUNTU_CODENAME=jammy

netikras@netikras-box:~/workspace/rtw89$ uname -r
5.15.0-86-generic
```

lwfinger, 2023-11-05:
Kernel 5.15.0 builds on my system. That means that your distro has ported API changess from newer kernels into 5.15.0-86. As I say in README.md, I do not fix such changes. You will need to do that.

closed 2023-11-05 by lwfinger

dariusj1, 2023-11-07:
Got my card working by installing the OEM kernel: https://easylinuxtipsproject.blogspot.com/p/hacks-mint.html#ID8.1

lwfinger, 2023-11-07:
I tested kernels back to 5.4.0, and all buiilt correctly. Has your distro backported some features?