lwfinger/rtw89 #282: Unstable Connection from rtw89_885be drivers

OakAtsume, 2023-11-06:
- Kernel-Version: 6.5.9-arch2-1
- OS: EndeavourOS Linux
- Hardware Model : 82XY IdeaPad Flex 5 16ABR8
Hello there, I've been experiencing issues with the  rtw89 drivers, as they seem to be rather unstable.
On boot they work perfectly for a seemly random amount of time (couple minutes); Until they randomly crash.
A snippet of the error dump
```log
[  498.036882] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.036943] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.037009] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.037070] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  498.037130] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.037190] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.037251] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.037312] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.037374] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  498.037434] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  498.043385] rtw89_8852be 0000:02:00.0: FW does not process h2c registers
[  498.043388] rtw89_8852be 0000:02:00.0: HW scan failed with status: -110
[  499.114743] rtw89_8852be 0000:02:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[  499.185572] rtw89_8852be 0000:02:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[  499.185576] rtw89_8852be 0000:02:00.0: mac init fail, ret:-110
```
### What I've attempted as a fix
I at first assumed it was an issue with the distribution or the installation, and after 2 re-installations I concluded it was the driver itself. </br>
I've tried Manually compiling the drivers by following the provided instructions;
```log
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make -j10
sudo make sign-install
```
to no avail, I tried signed and un-signed installation but both would still crash after a couple minutes or simply wouldn't load at all. (even tried blacklisting modules that might interfere)
```
blacklist rtw89core
blacklist rtw89_8852be
```
Even as a last ditch effort followed an old issue where I wrote to `50-rtw_pci.conf` but still to no luck.

- I'm truly out of options at this point, and really wanted to avoid creating an issue. But nothing I've read and tried has resolved the issue.

lwfinger, 2023-11-06:
The drivers are perfectly stable. The problem is with your BIOS that does not handle PCIe operations correctly. This well-known problem is handled by using the following parameters for the PCI module:

parm:           disable_aspm_l1:Set Y to disable PCI ASPM L1 support (bool)
parm:           disable_aspm_l1ss:Set Y to disable PCI L1SS support (bool)

The README.md tells you how to write the .conf file to fix this.

OakAtsume, 2023-11-06:
I understand, I've applied the changes and reboot.
If it isn't much of an inconvenience may the issue remain open to notify if it worked?

OakAtsume, 2023-11-06:
> I understand, I've applied the changes and reboot. If it isn't much of an inconvenience may the issue remain open to notify if it worked?

I applied the changes but still had the issue.
```
# /etc/modprobe.d/50-rtw_pci.conf
options rtw89 disable_aspm_l1=Y
options rtw89 disable_aspm_l1ss=Y
```
In this case I only lasted 2 minutes before being disconnected
```log
[  222.018259] rtw89_8852be 0000:02:00.0: read rf busy swsi
[  222.018317] rtw89_8852be 0000:02:00.0: write rf busy swsi
[  222.023402] rtw89_8852be 0000:02:00.0: FW does not process h2c registers
[  222.023404] rtw89_8852be 0000:02:00.0: HW scan failed with status: -110
[  223.114783] rtw89_8852be 0000:02:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[  223.185768] rtw89_8852be 0000:02:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[  223.185771] rtw89_8852be 0000:02:00.0: mac init fail, ret:-110
```
I re-tried by changing it to
```
```
# /etc/modprobe.d/50-rtw_pci.conf
options rtw89 disable_aspm_l1=true
option rtw89 disable_aspm_l1ss=true
```
and had a failure around 50 seconds on boot.

lwfinger, 2023-11-06:
Your .conf file is wrong. Thee family of drivers is rtw89, but the one you wnt to change is "rtw89_pci".

OakAtsume, 2023-11-06:
I changed it but to no luck still-
lasted around 1-2 mins before having the same issue
```
[   53.076723] rtw89_8852be 0000:02:00.0: FW does not process h2c registers
[   53.076727] rtw89_8852be 0000:02:00.0: HW scan failed with status: -110
```

lwfinger, 2023-11-06:
Verify that you got it right. Do 'sudo modprobe -rv rtw89_8852be; followed by 'sudo modprobe -v rtw89_8852be'. Do the new parameters show on the rtw89_pci line?

OakAtsume, 2023-11-06:
Running `sudo modprobe -rv rtw89_8852be` doesn't return anything (expected)
but `sudo modprobe -v rtw89_8852be`
returns 
```
insmod /lib/modules/6.5.9-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw89_8852be.ko.zst 
modprobe: ERROR: could not insert 'rtw89_8852be': Device or resource busy
```
I don't quiet understand what you mean by "Do the new parameters show on the rtw89_pci line?", would you like to see my current .conf?

lwfinger, 2023-11-06:
If you are running this driver, then your parameters are wrong. Add extra lines that use rtw89pci in place of rtw89_pci. That will cover you for either set. Then 'sudo modprobe -rv rtw_8852be' followed by 'sudo modprobe -v rtw_8852be'. When you get it right, the options will be printed on the pci line.

OakAtsume, 2023-11-06:
After re-running the command my output was:
```
insmod /lib/modules/6.5.9-arch2-1/updates/dkms/rtw89pci.ko.zst 
insmod /lib/modules/6.5.9-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/6.5.9-arch2-1/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko 
```
My current config is 
```
options rtw89_pci disable_aspm_l1=Y
options rtw89_pci disable_aspm_l1ss=Y
```
I'm sorry, I'm unsure if I'm doing this correctly

lwfinger, 2023-11-07:
You are loading rtw89pci, and trying to apply parameters to rtw89_pci!!!! The names MUST be the same, which is why I told you to create lines for BOTH names!

AnaxMorden, 2023-11-07:
Hey thanks for Helping so far,
i have another device with the same network chip

Im using the the Lenovo Ideapad Slim 3 16ABR8

Same issue here, tried the things out but still not working

In my case the flags for the module rtw89_pci are enabled (rtw89pci is not used) but the device puts out the same error again (even after reboot)

(If u need more logs, ill send them later if u need them)

closed 2023-11-08 by OakAtsume

NetSysFire, 2023-11-11:
@lwfinger Could you please pin this issue or make this more obvious in the README that this is a known issue? I *very* nearly opened another issue about this and I think either pinning this issue or mentioning this clearly in the README can prevent further duplicates.

OakAtsume, 2023-11-20:
A good idea; Since it seems this is a very common issue.