lwfinger/rtw89 #283: Build Debian package missing "/debian" directory

liorzilp, 2023-11-14:
The build of the Debian package is broken.
In order to create a Debian package using dpkg-buildpackage a specific structure is required.
Explanation and example can be found in:
https://github.com/FooBarWidget/debian-packaging-for-the-modern-developer/blob/master/tutorial-2/README.md#the-dpkg-buildpackage-workflow

Hareis, 2023-11-19:
yes, i hava some  problem，i download that [https://github.com/peasoft/rtw89-kali](url), that have debian directory

EnercoopJNE, 2023-11-29:
Same issue here, building the package would be very convenient... In the link provided by Hareis the repo is the one we are on here if I'm not mistaken.

lwfinger, 2023-11-29:
If you want a new feature, please submit an update to the  code. My time is limited.

PR #329 (2024-04-03): rtw89: Add missing debian dkms directory and fix build guide
Copy the `debian` directory from the old `test` branch to the latest `main` branch to fix #283.

Also fix `git deborig` not found in dkms building.
closed 2023-11-29 by lwfinger