lwfinger/rtw89 #284: make error in Ubuntu 20.04 and kernal version 5.15.0-88-generic

Venkata-Prashanth, 2023-11-17:
/home/user1/rtw89/regd.c: In function ‘rtw89_reg_6ghz_power_recalc’:
/home/user1/rtw89/regd.c:518:25: error: ‘struct ieee80211_bss_conf’ has no member named ‘power_type’; did you mean ‘txpower_type’?
  518 |   switch (vif->bss_conf.power_type) {
      |                         ^~~~~~~~~~
      |                         txpower_type
/home/user1/rtw89/regd.c:519:8: error: ‘IEEE80211_REG_VLP_AP’ undeclared (first use in this function)
  519 |   case IEEE80211_REG_VLP_AP:
      |        ^~~~~~~~~~~~~~~~~~~~
/home/user1/rtw89/regd.c:519:8: note: each undeclared identifier is reported only once for each function it appears in
/home/user1/rtw89/regd.c:522:8: error: ‘IEEE80211_REG_LPI_AP’ undeclared (first use in this function)
  522 |   case IEEE80211_REG_LPI_AP:
      |        ^~~~~~~~~~~~~~~~~~~~
/home/user1/rtw89/regd.c:525:8: error: ‘IEEE80211_REG_SP_AP’ undeclared (first use in this function); did you mean ‘IEEE80211_SEQ_TO_SN’?
  525 |   case IEEE80211_REG_SP_AP:
      |        ^~~~~~~~~~~~~~~~~~~
      |        IEEE80211_SEQ_TO_SN
make[2]: *** [scripts/Makefile.build:297: /home/user1/rtw89/regd.o] Error 1
make[1]: *** [Makefile:1909: /home/user1/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-88-generic'
make: *** [Makefile:93: all] Error 2

lwfinger, 2023-11-17:
On kernel 5.15.0, the driver builds fine. I think that Ubuntu has modified the kernel source, which means you will need to add local modifications to account for that. As I state in README.md, I cannot make these kinds of changes.

closed 2023-11-17 by lwfinger

aghand0ur, 2023-11-18:
I am using kernel `5.15.0-88-generic` and has the same issue..
so I would have to debug it myself or downgrade kernel?

lwfinger, 2023-11-18:
Yes. Upgrading to a newer Ubuntu is also an option.

Venkata-Prashanth, 2023-11-19:
Thank you for the reply, it works fine on Ubuntu 22.04 LTS.

aghand0ur, 2023-11-22:
@lwfinger 
I have upgraded kernel to `6.2.0-37-generic`
and tried to build but received the following error:
```
make -C /lib/modules/6.2.0-37-generic/build M=/home/ali/Downloads/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-37-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
  You are using:           gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-37-generic'
```

But I am not able to install `gcc-11-x86-64-linux-gnu`
```
The following packages have unmet dependencies:
 gcc-11-x86-64-linux-gnu:i386 : Depends: binutils-x86-64-linux-gnu:i386 (>= 2.38) but it is not going to be installed
                                Depends: libgcc-11-dev-amd64-cross:i386 (>= 11.4.0-1ubuntu1~22.04cross1) but it is not installable
                                Recommends: libc6-dev-amd64-cross:i386 (>= 2.13-0ubuntu6)
E: Unable to correct problems, you have held broken packages.

```

Any clue?

aghand0ur, 2023-11-22:
I had to simply pass the gcc as an argument for make `make CC=/usr/bin/x86_64-linux-gnu-gcc-11`

lwfinger, 2023-11-22:
You also need to learn the difference between warnings and errors.

aghand0ur, 2023-11-22:
> You also need to learn the difference between warnings and errors.

right and I am sure you could put it in nicer words!