lwfinger/rtw89 #285: Still not worked after success install. Arch | 8852ce

bushaev-denis, 2023-11-17:
### Specs:
- uname: `Linux server 6.6.1-arch1-1`
- mobo: `gygabyte aorus pro ax b550i 1.3`
- wifi: `realtek 8852ce`
- secure boot: `off`

### What was done:

```shell
Installed dkms:
git clone https://aur.archlinux.org/rtw89-dkms-git.git
cd rtw89-dkms-git
makepkg -sri

Installed drivers:
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install

Disable old pci:
sudo modprobe -rv rtw89_pci
Enable new one:
sudo modprobe -v rtw89_pci 

Installed firmware:
wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw89/rtw8852c_fw.bin
sudo cp rtw8852c_fw.bin /usr/lib/firmware/rtw89/rtw8852c_fw.bin
```

And i see nothing after rebooting, not errors, not adding wifi to the dm ui.

### Some additional information

```shell
lspci | grep Realtek
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
05:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device c852 (rev 01)
```

```shell
sudo dmesg | grep rtw
[   11.893368] rtw89_8852ce 0000:05:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[   11.893425] rtw89_8852ce 0000:05:00.0: enabling device (0000 -> 0003)
[   11.894271] rtw89_8852ce 0000:05:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 1
[   11.894274] rtw89_8852ce 0000:05:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 3
[   11.894279] rtw89_8852ce 0000:05:00.0: MAC has already powered on
[   11.899851] RIP: 0010:rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci]
[   11.900350]  ? rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci 65dd47a6bc8069c38b6a2d47f16ae044e3f098d0]
[   11.900391]  ? __pfx_rtw89_pci_ops_mac_pre_init+0x10/0x10 [rtw89pci 65dd47a6bc8069c38b6a2d47f16ae044e3f098d0]
[   11.900434]  rtw89_mac_partial_init+0x150/0x350 [rtw89core 27a3c47669f0fe62dd32ef7cb50384666b257788]
[   11.900491]  rtw89_chip_info_setup+0x9d/0x2c0 [rtw89core 27a3c47669f0fe62dd32ef7cb50384666b257788]
[   11.900542]  rtw89_pci_probe+0xa86/0x1750 [rtw89pci 65dd47a6bc8069c38b6a2d47f16ae044e3f098d0]
[   11.900791]  ? __pfx_rtw89_8852ce_driver_init+0x10/0x10 [rtw89_8852ce ed6ee5d673e78d197b110b7e9274d6d18a5f5d39]
[   11.901342] Modules linked in: rtw89_8852ce(+) rtw_8852c(OE) rtw89pci(OE) rtw89core(OE) r8125(OE) raid0 r8169 md_mod mac80211 8021q realtek intel_rapl_msr mdio_devres garp mrp intel_rapl_common stp libphy libarc4 overlay llc edac_mce_amd kvm_amd snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio kvm snd_hda_codec_hdmi snd_usb_audio(+) snd_hda_intel irqbypass crct10dif_pclmul snd_intel_dspcfg snd_usbmidi_lib crc32_pclmul snd_intel_sdw_acpi polyval_clmulni snd_ump polyval_generic snd_hda_codec snd_rawmidi gf128mul snd_seq_device ghash_clmulni_intel snd_hda_core sha512_ssse3 mc snd_hwdep aesni_intel snd_pcm crypto_simd cryptd snd_timer vfat rapl fat gigabyte_wmi wmi_bmof sp5100_tco acpi_cpufreq snd pcspkr ccp i2c_piix4 soundcore cfg80211 k10temp gpio_amdpt gpio_generic rfkill mac_hid i2c_dev crypto_user fuse loop dm_mod ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 usbhid amdgpu i2c_algo_bit drm_ttm_helper ttm video drm_exec drm_suballoc_helper amdxcp drm_buddy gpu_sched drm_display_helper nvme
[   11.901715] RIP: 0010:rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci]
```

Also after installing this driver i cannot normally reboot, stucked at `unable to unmount /oldroot` same as for this dude https://github.com/lwfinger/rtw89/issues/2#issuecomment-861106183

Feel free to ask me provide more information if needed.

lwfinger, 2023-11-17:
With a 6.6 kernel, there is no reason for you to be using this driver AT ALL. If you are trying to do this, but have not blacklisted the in-kernel version, then the accompanying conflicts will show exactly what you are seeing.

If you are not using the driver from this repo, then look at README.md on how to report problems with the kernel driver.

closed 2023-11-17 by lwfinger

bushaev-denis, 2023-11-17:
@lwfinger 
Perhaps this seems somewhat obvious to you, but not all people understand so well what exactly needs to be done for something to work.
I'm just asking for help on what I should do to make your drivers work.
Yes, as you correctly understood, mine didn’t work out of the box, I don’t know the reason and I don’t have time to deal with the kernel and its drivers right now.
Please tell me which built-in drivers I should blacklist so that your drivers will work.

lwfinger, 2023-11-17:
If you are going to use these out-of-kernel drivers, then you must read the README.md. It tells you how to blacklist the in-kernel versions. In your case, you should do nothing. Uninstall  this repo and  the dkms configuration, then reinstall the kernel. It should JUST WORK.

bushaev-denis, 2023-11-17:
If you use CAPS reality will not change.
I`ve deleted ur drivers and dkms 3 times already, it just dont work. Neither ur driver nor builtin.

bushaev-denis, 2023-11-17:
Whats the point to building something 3rd party drivers if mine one works well?

lwfinger, 2023-11-17:
Not everyone runs a 6.6 kernel. If you were using 5.19, then you need this one because the driver is not in the kernel.

The thing is that when you have a new kernel, use th internal drivers

Have you looked at the output of 'sudo dmesg | less'? Do you have the firmware installed?

bushaev-denis, 2023-11-17:
If you mean here - yes.
```
[   10.334164] rtw89_8852ce 0000:05:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[   10.334223] rtw89_8852ce 0000:05:00.0: enabling device (0000 -> 0003)
[   10.335098] rtw89_8852ce 0000:05:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 1
[   10.335101] rtw89_8852ce 0000:05:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 3
[   10.335106] rtw89_8852ce 0000:05:00.0: MAC has already powered on
[   10.340597] RIP: 0010:rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci]
[   10.341039]  ? rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci 8e49679efe1b763328b7b7e3e302370ad5bbe3be]
[   10.341075]  ? __pfx_rtw89_pci_ops_mac_pre_init+0x10/0x10 [rtw89pci 8e49679efe1b763328b7b7e3e302370ad5bbe3be]
[   10.341112]  rtw89_mac_partial_init+0x150/0x350 [rtw89core 0a0a16100048a54f77bb357ab0038e22659648d2]
[   10.341159]  rtw89_chip_info_setup+0x9d/0x2c0 [rtw89core 0a0a16100048a54f77bb357ab0038e22659648d2]
[   10.341202]  rtw89_pci_probe+0xa86/0x1750 [rtw89pci 8e49679efe1b763328b7b7e3e302370ad5bbe3be]
[   10.341443]  ? __pfx_rtw89_8852ce_driver_init+0x10/0x10 [rtw89_8852ce ed6ee5d673e78d197b110b7e9274d6d18a5f5d39]
[   10.342208] Modules linked in: rtw89_8852ce(+) snd_hda_intel snd_usb_audio(+) rtw_8852c(OE) irqbypass snd_intel_dspcfg rtw89pci(OE) crct10dif_pclmul snd_usbmidi_lib snd_intel_sdw_acpi crc32_pclmul rtw89core(OE) polyval_clmulni snd_ump polyval_generic snd_hda_codec snd_rawmidi gf128mul btusb snd_hda_core snd_seq_device ghash_clmulni_intel r8125(OE) mac80211 btrtl mc sha512_ssse3 snd_hwdep aesni_intel btintel r8169 snd_pcm libarc4 btbcm realtek crypto_simd btmtk cryptd snd_timer vfat mdio_devres snd sp5100_tco cfg80211 bluetooth fat rapl raid0 wmi_bmof gigabyte_wmi acpi_cpufreq pcspkr k10temp i2c_piix4 libphy ccp soundcore ecdh_generic rfkill gpio_amdpt md_mod gpio_generic mousedev joydev apple_mfi_fastcharge mac_hid i2c_dev crypto_user fuse loop dm_mod ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 uas usb_storage usbhid amdgpu i2c_algo_bit drm_ttm_helper ttm video drm_exec drm_suballoc_helper amdxcp drm_buddy nvme gpu_sched drm_display_helper crc32c_intel nvme_core xhci_pci cec nvme_common xhci_pci_renesas
[   10.351087] RIP: 0010:rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci]
``` 
if no - here is full dmesg output
[dmesg.txt](https://github.com/lwfinger/rtw89/files/13397200/dmesg.txt)
PS: i'm assume it's a built in driver, bc i removed dkms and in ur repo `sudo make uninstall`

bushaev-denis, 2023-11-17:
Dude, just sorry me, please. This is was buggy other dkms. When i removed it, all became good. 
And big thanks you for your response, idk how much time it would have taken to get there myself.

lwfinger, 2023-11-17:
No problem. The stuff that mattered from your dmesg was this:
[   11.851825] BUG: unable to handle page fault for address: 0000000100000001
[   11.851858] #PF: supervisor read access in kernel mode
[   11.851879] #PF: error_code(0x0000) - not-present page
[   11.851900] PGD 0 P4D 0 
[   11.851915] Oops: 0000 [#1] PREEMPT SMP NOPTI
[   11.851935] CPU: 11 PID: 413 Comm: (udev-worker) Tainted: G           OE      6.6.1-arch1-1 #1 be166a630cd909acf8820643140e9106c6ea80e6
[   11.851979] Hardware name: Gigabyte Technology Co., Ltd. B550I AORUS PRO AX/B550I AORUS PRO AX, BIOS FCc 09/20/2023
[   11.852016] RIP: 0010:rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci]
[   11.852049] Code: Unable to access opcode bytes at 0xffffffffc1847ff4.
[   11.852075] RSP: 0018:ffffc900016d39d0 EFLAGS: 00010282
[   11.852098] RAX: 0000000100000001 RBX: ffff8881071020a0 RCX: ffffc90003b01700
[   11.852125] RDX: 0000000000010008 RSI: ffffc90003b01700 RDI: ffff8881071020a0
[   11.852153] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[   11.852181] R10: 0000000000000001 R11: 0000000000000100 R12: 0000000000000000
[   11.852209] R13: ffff888101ced000 R14: 00000000000000ff R15: 00000000000000ff
[   11.852237] FS:  00007f1f677d1480(0000) GS:ffff888ffecc0000(0000) knlGS:0000000000000000
[   11.852268] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   11.852291] CR2: ffffffffc1847ff4 CR3: 00000001269a4000 CR4: 0000000000f50ee0
[   11.852320] PKRU: 55555554
[   11.852334] Call Trace:
[   11.852347]  <TASK>
[   11.852359]  ? __die+0x23/0x70
[   11.852377]  ? page_fault_oops+0x171/0x4e0
[   11.852401]  ? exc_page_fault+0x7f/0x180
[   11.852421]  ? asm_exc_page_fault+0x26/0x30
[   11.852445]  ? rtw89_pci_ops_mac_pre_init+0xe/0x20 [rtw89pci 65dd47a6bc8069c38b6a2d47f16ae044e3f098d0]
[   11.852485]  rtw89_mac_partial_init+0x150/0x350 [rtw89core 27a3c47669f0fe62dd32ef7cb50384666b257788]
[   11.852538]  rtw89_chip_info_setup+0x9d/0x2c0 [rtw89core 27a3c47669f0fe62dd32ef7cb50384666b257788]
[   11.852587]  rtw89_pci_probe+0xa86/0x1750 [rtw89pci 65dd47a6bc8069c38b6a2d47f16ae044e3f098d0]
[   11.852626]  local_pci_probe+0x45/0xa0
[   11.852647]  pci_device_probe+0xc1/0x260
[   11.852668]  ? sysfs_do_create_link_sd+0x6e/0xe0
[   11.852692]  really_probe+0x19e/0x3e0
[   11.852712]  ? __pfx___driver_attach+0x10/0x10
[   11.852733]  __driver_probe_device+0x78/0x160
[   11.852754]  driver_probe_device+0x1f/0x90
[   11.852775]  __driver_attach+0xd2/0x1c0
[   11.852794]  bus_for_each_dev+0x88/0xd0
[   11.852814]  bus_add_driver+0x116/0x220
[   11.852834]  driver_register+0x59/0x100
[   11.852851]  ? __pfx_rtw89_8852ce_driver_init+0x10/0x10 [rtw89_8852ce ed6ee5d673e78d197b110b7e9274d6d18a5f5d39]
[   11.852884]  do_one_initcall+0x5d/0x320
[   11.852903]  do_init_module+0x60/0x240
[   11.852920]  init_module_from_file+0x89/0xe0
[   11.852941]  idempotent_init_module+0x120/0x2b0
[   11.852961]  __x64_sys_finit_module+0x5e/0xb0
[   11.852980]  do_syscall_64+0x60/0x90
[   11.852998]  ? srso_alias_return_thunk+0x5/0x7f
[   11.853018]  ? ksys_lseek+0x6c/0xb0
[   11.853035]  ? srso_alias_return_thunk+0x5/0x7f
[   11.853054]  ? syscall_exit_to_user_mode+0x2b/0x40
[   11.853073]  ? srso_alias_return_thunk+0x5/0x7f
[   11.853092]  ? do_syscall_64+0x6c/0x90
[   11.853107]  ? do_syscall_64+0x6c/0x90
[   11.853123]  entry_SYSCALL_64_after_hwframe+0x6e/0xd8
[   11.853144] RIP: 0033:0x7f1f6732c73d
[   11.853170] Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c3 95 0c 00 f7 d8 64 89 01 48
[   11.853223] RSP: 002b:00007ffc19c4a538 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   11.853248] RAX: ffffffffffffffda RBX: 00005618da0ee230 RCX: 00007f1f6732c73d
[   11.853272] RDX: 0000000000000004 RSI: 00007f1f67874376 RDI: 0000000000000013
[   11.853296] RBP: 00007f1f67874376 R08: 0000000000000001 R09: fffffffffffffe90
[   11.853320] R10: 0000000000000050 R11: 0000000000000246 R12: 0000000000020000
[   11.853344] R13: 00005618da1262f0 R14: 0000000000000000 R15: 00005618da11f4c0
[   11.853370]  </TASK>
[   11.853381] Modules linked in: rtw89_8852ce(+) rtw_8852c(OE) rtw89pci(OE) rtw89core(OE) r8125(OE) r8169 8021q raid0 realtek mac80211 garp mdio_devres intel_rapl_msr mrp md_mod intel_rapl_common stp libphy libarc4 llc overlay edac_mce_amd kvm_amd snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio kvm snd_hda_codec_hdmi irqbypass snd_hda_intel crct10dif_pclmul snd_usb_audio(+) crc32_pclmul snd_intel_dspcfg snd_intel_sdw_acpi polyval_clmulni snd_usbmidi_lib polyval_generic snd_hda_codec snd_ump gf128mul snd_rawmidi ghash_clmulni_intel snd_seq_device snd_hda_core sha512_ssse3 mc snd_hwdep aesni_intel snd_pcm crypto_simd cryptd snd_timer vfat rapl fat snd sp5100_tco acpi_cpufreq wmi_bmof pcspkr gigabyte_wmi ccp soundcore k10temp i2c_piix4 cfg80211 gpio_amdpt gpio_generic rfkill mac_hid i2c_dev crypto_user fuse dm_mod loop ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 usbhid amdgpu i2c_algo_bit drm_ttm_helper ttm video drm_exec drm_suballoc_helper amdxcp drm_buddy gpu_sched nvme drm_display_helper
[   11.853441]  crc32c_intel nvme_core cec xhci_pci xhci_pci_renesas nvme_common wmi
[   11.853724] CR2: 0000000100000001
[   11.853738] ---[ end trace 0000000000000000 ]---

Obviously, that other dkms was the problem. Glad it it working for you now.