lwfinger/rtw89 #286: RTW8852AE After suspend : Unable to change power state from D3cold to D0, device inaccessible

zhinu, 2023-11-19:
Hello, I'm clogging this thread just as a last resort.

Wifi won't work after waking up from suspend.

## My Specs

My computer :
```
Host: 82L5 IdeaPad 5 Pro 16ACH6 
OS: Nobara Linux 38 (KDE Plasma) x86_64 
Kernel: 6.5.9-201.fsync.fc38.x86_64
``` 

My Wifi kernel:
```
RealTek 8852AE
``` 

## Steps I've tried

### 1. Multiple scripts in the Systemd System-Sleep Directory

```
sudo nano /usr/lib/systemd/system-sleep/suspend_wifi
```

following scripts 

```
#!/bin/sh
if [ "${1}" == "pre" ]; then
  modprobe -rv rtw_8852ae
  modprobe -rv rtw89core
elif [ "${1}" == "post" ]; then
  modprobe -v rtw_8852ae
fi
``` 

```
#!/bin/sh
case $1 in
  pre)
    modprobe -r rtw_8852ae
    modprobe -r rtw89pci
    modprobe -r rtw_8852a
    modprobe -r rtw89core
    ;;
  post)
    modprobe rtw89core
    modprobe rtw_8852a
    modprobe rtw89pci
    modprobe rtw_8852ae
    ;;
esac

``` 

Conclusion : Not working Device busy

### 2. Modifying the module parameters

     ```bash
     sudo nano /etc/modprobe.d/rtw89pci.conf
     ```

     ```
     options rtw89pci disable_aspm_l1=1 disable_aspm_l1ss=1
     ```

     ```bash
     sudo nano /etc/modprobe.d/rtw89core.conf
     ```

     ```
     options rtw89core disable_ps_mode=1
     ```

   - Update the initial RAM filesystem:
     ```bash
     sudo dracut -f
     ```

   - Reboot for the changes to take effect:
     ```bash
     sudo reboot
     ```

conclusion : Still no wifi after waking up from suspend

### 3. Use tlp to change the PCIe Power Management parameters

Idea taken from : (https://bbs.archlinux.org/viewtopic.php?id=281984)[https://bbs.archlinux.org/viewtopic.php?id=281984]

Based on the findings from the Arch Linux Forums, the issue with the Realtek RTL8852AE network controller not being able to change its power state from D3cold to D0 after waking from sleep can be addressed by modifying the power management settings using `tlp`.

Here's a framework based on mpdeimos' solution:

1. **Modify `/etc/tlp.conf` for PCIe Power Management:**
   - Open the `tlp` configuration file:
     ```bash
     sudo nano /etc/tlp.conf
     ```
   - Set PCIe runtime power management to 'auto':
     ```
     RUNTIME_PM_ON_AC=auto
     RUNTIME_PM_ON_BAT=auto
     ```
   - Add your WiFi adapter's PCIe address to deny the runtime PM 'auto' configuration. You'll need to replace `01:00.0` with your device's specific PCIe address. You can find this address using `lspci | grep Network`.
     ```
     RUNTIME_PM_DENYLIST="01:00.0"
     ```
   - Save and close the file.

2. **Apply the Configuration:**
   - Restart the `tlp` service to apply the changes:
     ```bash
     sudo systemctl restart tlp
     ```

3. **Verify the Configuration:**
   - Check the PCIe power management settings:
     ```bash
     tlp-stat --pcie
     ```
   - Verify that the power control for your WiFi device is set to 'on'.

This approach essentially configures the system to not allow the PCIe power management to turn off the WiFi card, thus keeping it in a ready state even after waking up from sleep.

Conclusion : doesn't work

### 4. Reinstalling driver and testing everything with and without the driver
Obviously for all the steps 
**I blacklisted the distro driver** 

### 5. Check BIOS Settings
Installed latest windows 11, upgraded the bios and deleted the malware.
Unfortunately, there aren't that many options in my bios.
Bios spec 
```
Version : GSCN38WW
```

## What I haven't yet tried 

### 1. Explore Additional Power Management Settings

Linux provides various mechanisms to control power management settings. We can try adjusting these:

1. **Kernel Boot Parameters:**
   - Edit the GRUB configuration:
     ```bash
     sudo nano /etc/default/grub
     ```
   - Find the line starting with `GRUB_CMDLINE_LINUX` and add `pci=noaer pcie_aspm=off`. It should look something like:
     ```
     GRUB_CMDLINE_LINUX="quiet splash pci=noaer pcie_aspm=off"
     ```
   - Save the file and update GRUB:
     ```bash
     sudo grub2-mkconfig -o /boot/grub2/grub.cfg
     ```
   - Reboot your system.

2. **Temporary Disablement for Testing:**
   - You can temporarily disable PCIe power management to test if it's causing the issue:
     ```bash
     echo off | sudo tee /sys/module/pcie_aspm/parameters/policy
     ```

### 4. Consider Linux Kernel Parameters

Adjusting specific Linux kernel parameters can sometimes resolve hardware-related issues:

- Edit the GRUB configuration as mentioned above and try adding `acpi=force` to the `GRUB_CMDLINE_LINUX` line.

please help <3

lwfinger, 2023-11-19:
This is a problem with your computer's BIOS, not the driver.

Your first script in /usr/lib/systemd/system-sleep/suspend_wifi should be all you need as long as you are using the driver from this repo. All others should be deleted. You also must be certain that the script is executable. I think that is likely your problem. 

Everything else you tried to irrelevant or harmful.

zhinu, 2023-11-21:
So I figured it out.
My computer the Lenovo 16ach6, has a very limited bios.

But thanks to DavidS95, we can access some bios settings through the browser they created.
Hop there [https://github.com/DavidS95/Smokeless_UMAF](https://github.com/DavidS95/Smokeless_UMAF)

Boot from the USB created, then 
Device Manager > AMD PBS > Enable D3cold > true

closed 2023-11-21 by zhinu