lwfinger/rtw89 #291: rtw89_8852ce mac init fail when suspend

binli, 2023-12-01:
Linux version is 6.7.0-060700rc3-generic, the wifi could not work after resume.

Dec 01 18:41:50 tb15 kernel: rtw89_8852ce 0000:05:00.0: loaded firmware rtw89/rtw8852c_fw.bin                                           
Dec 01 18:41:50 tb15 kernel: rtw89_8852ce 0000:05:00.0: enabling device (0000 -> 0003)                                                  
Dec 01 18:41:50 tb15 kernel: rtw89_8852ce 0000:05:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 1                   
Dec 01 18:41:50 tb15 kernel: rtw89_8852ce 0000:05:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 3                   
Dec 01 18:41:50 tb15 kernel: rtw89_8852ce 0000:05:00.0: chip rfe_type is 1                                                              
Dec 01 18:41:50 tb15 systemd-udevd[572]: Using default interface naming scheme 'v249'.                                                  
Dec 01 18:41:50 tb15 kernel: rtw89_8852ce 0000:05:00.0 wlp5s0: renamed from wlan0                                                       
Dec 01 18:41:51 tb15 systemd-udevd[581]: Using default interface naming scheme 'v249'.

Dec 01 18:42:37 tb15 kernel: r8169 0000:04:00.0 enp4s0: Link is Down                                                                    
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: xtal si not ready(W): offset=90 val=10 mask=10
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: xtal si not ready(W): offset=90 val=10 mask=10                                  
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: mac init fail, ret:-110                                                         
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: xtal si not ready(W): offset=90 val=10 mask=10                                  
Dec 01 18:42:37 tb15 wpa_supplicant[883]: Could not set interface wlp5s0 flags (UP): Connection timed out                               
Dec 01 18:42:37 tb15 wpa_supplicant[883]: nl80211: Could not set interface 'wlp5s0' UP                                                  
Dec 01 18:42:37 tb15 wpa_supplicant[883]: nl80211: deinit ifname=wlp5s0 disabled_11b_rates=0                                            
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: xtal si not ready(W): offset=90 val=10 mask=10                                  
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: mac init fail, ret:-110                         
Dec 01 18:42:37 tb15 kernel: Bluetooth: hci0: RTL: fw version 0x040d7225                                                                
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: xtal si not ready(W): offset=90 val=10 mask=10                                  
Dec 01 18:42:37 tb15 wpa_supplicant[883]: Could not set interface wlp5s0 flags (UP): Connection timed out                               
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: xtal si not ready(W): offset=90 val=10 mask=10                                  
Dec 01 18:42:37 tb15 kernel: rtw89_8852ce 0000:05:00.0: mac init fail, ret:-110                                                         
Dec 01 18:42:37 tb15 wpa_supplicant[883]: WEXT: Could not set interface 'wlp5s0' UP

lwfinger, 2023-12-01:
Is this problem a regression with kernel 6.7?

Comments on your log: There are some BIOSs that do not handle the PCIe bus correctly. For those, the solution is to unload the driver before suspending, and reload on resume. README.md has instructions on how to do this. Note: you will need to change the "rtw_8852ae" in file suspend_rtw89 to "rtw89_8852ce" or "rtw_8852ce". The first one is needed if you are using the in-kernel driver - the second is for the one from this repo.

pma1, 2023-12-15:
After resume the PCIe bus is accessible, I think it's a BIOS bug.

lwfinger, 2023-12-15:
Yes it is. Read the section ##### Problem with recovery after sleep or hibernation
 in README.md. Before you copy the file into /usr/lib/systemd/system-sleep/, edit suspend_rtw89 to refer to your device.

binli, 2023-12-18:
@lwfinger, thanks for your feedback, we could not use workaround in ubuntu. Cause it's related BIOS, I will close it, thanks!

closed 2023-12-18 by binli