lwfinger/rtw89 #292: make error at 5.10 kernal

dawangll, 2023-12-07:
[image]
my debian kernal is 5.10 still make error

lwfinger, 2023-12-07:
By the time your kernel gets older than 5.15, so much of wifi 6 is disabled that you might as well be using wifi4 devices.

If possible, you should update your kernel. If that is not possible, then let me know, and I will see if I can find a fix for 5.10. I will not promise a date.

lwfinger, 2023-12-07:
I had some time and fixed the builds through kernel 5.5. If you still get errors, then your distro has backported changes. Those errors will need to be fixed by you.

rinkedevries, 2023-12-12:
I have the same issue on kernel 5.10.104-tegra. Is there a fix?

rinkedevries, 2023-12-12:
I dont understand why it suddenly breaks. Only updating this repo breaks the whole thing. Also going back in commits is no use since the history seems to be completely rewritten? Ive fixed it by going back and undoing the changes. But why is the repo history being rewritten? To make sure people cant easily use it on older kernels?

Just fyi, Jetsons cannot currently use anything above 5.10 so updating the kernel is not an option.

lwfinger, 2023-12-12:
I had to rebuild the repo from scratch, thus I lost the local history; however, I got the entire history from the kernel.

I looked at the code, and found a problem. Do a 'git pull' and it should build.

rinkedevries, 2023-12-13:
> I had to rebuild the repo from scratch, thus I lost the local history; however, I got the entire history from the kernel.
> 
> I looked at the code, and found a problem. Do a 'git pull' and it should build.

Thanks. I had to do the exact same thing to get it working again. Ive already forked it so I am not affected if the history is messed with again.