lwfinger/rtw89 #293: my ubantu version is 20.04 and kernel version is 5.15

903859154, 2023-12-08:
should i update the kernel ?
[image]

903859154, 2023-12-08:
or should i change to use rtw88?  In the picture i use is rtw 89

903859154, 2023-12-09:
So should i also need to update the kernel? Because you said you fixed the error

903859154, 2023-12-09:
[image]
this is my Graphics card model number

903859154, 2023-12-09:
[image]

lwfinger, 2023-12-09:
I thought I made it clear. I did fix the error, but you should still update your kernel as the throughput is lower and lower the older your kernel. On my system, rtw_8852be gets upload and download speeds of over 700 Mbps running LibreSpeed with kernel 6.7-rc4. Using kernel 5.14, the throughput is 210 Mbps with nothing else changed. You do not NEED to update your kernel, but you will get wifi4 performance from your wifi6 device if you do not.

valerio-bozzolan, 2024-06-28:
(Could you please rename the title from "ubantu" to "Ubuntu" since my eyes are melting ihih - thanks)