lwfinger/rtw89 #296: Very slow WIFI since recently - Bus 003 Device 003: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]

stefanofornari, 2023-12-23:
Hello I have installed the driver a couple of months ego and everything was working like a charm giving me 30Mb/s download speed. Since a while it dropped to 1.2-2Mb/s ... I pulled the latest and installed it, but it did not improve.

I am available to provide any information it can help to solve the problem.

lwfinger, 2023-12-23:
I am unable to do any debugging on the vendor driver. As the driver has not changed, what changes did you make in your ssystem?

stefanofornari, 2023-12-23:
Just upgrading regularly... maybe the kernel? sorry if I am not accurate, but I am tech linux user, but not really into the details of the system. Is there anything we can do to at least investigate? 
Just to exclude it is the WIFI itself it is worth to mention I then plugged in a USB dongle with a different chipset and I reach the max speed of my ADSL

closed 2024-03-28 by lwfinger