lwfinger/rtw89 #301: Last update broke wifi

BUGOVER, 2024-01-05:
After last update wifi is not working on ubuntu 22.04 and kernel "6.2.0-39"

OakAtsume, 2024-01-05:
I've also been experiencing issues with the latest update of rtw89;
with issues ranging from:
- Unable Connection
- Reports of Beacon loss from wpa_supplicant every second
```
Jan 05 13:17:53 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:17:54 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:17:55 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:17:56 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:17:57 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:17:58 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:17:59 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:18:00 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:18:01 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:18:02 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:18:03 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS
Jan 05 13:18:04 [Redacted] wpa_supplicant[828]: wlan0: CTRL-EVENT-BEACON-LOSS

```

to simply an un-usable connection.

Running EndeavourOS `6.6.9-arch1-1`

lwfinger, 2024-01-05:
Have you tried bisecting the set of updates? It works perfectly on my openSUSE Tumbleweed running kenel 6.7-rc8.

OakAtsume, 2024-01-06:
I personally haven't (a bit unsure on what you mean by "bisecting"); but I've noticed that this issue is happening to other's as well; aside from this issue.
Where the connection *is* established but the actual networking is unusable (for my case due to beacons being lost?)

alou-S, 2024-01-06:
@lwfinger Problem is caused by commit f247fefa2138a0efe0de0f95330e3670e7165a43 at 
https://github.com/lwfinger/rtw89/blob/139b1477900f7f13f0c1111c8919a3bf4aeb4a80/mac80211.c#L1150-L1155
https://github.com/lwfinger/rtw89/blob/139b1477900f7f13f0c1111c8919a3bf4aeb4a80/mac80211.c#L1174-L1177

Where the author added the `#if` preprocessor which reserves the options under it exclusively to Red Hat Linux builds. 

Also I'm pretty sure the commit author meant `RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(10, 0)` and not `RHEL_RELEASE_CODE > RHEL_RELEASE_VERSION(10, 0)` on line 1150

lwfinger, 2024-01-06:
@alou-S - Does that change fix your system?

@OakAtsume - Bisection is a feature of git. If you can identify a commit that works (good) and one that fails (bad), then bisect will find one halfway between good and bad. You then build and test that revision, and report if this is the new good, or bad. Repeat until you find the bad one. See https://stackoverflow.com/questions/4713088/how-do-i-use-git-bisect#4714297 for another explanation.

lwfinger, 2024-01-06:
I just fixed some more errors in the RHEL9 patches. Please pull and try again.

OakAtsume, 2024-01-06:
Hello I wanted to confirm that the given patch from @alou-S worked for me.

BUGOVER, 2024-01-06:
It works, thanks, I think we can close this topic

weaki233, 2024-03-12:
@lwfinger ,Hi~ I’ve encountered a problem with the rtw89 and **rtw_8852ce** drivers following a kernel update. The drivers functioned well with kernel version **5.15.0.97,** but they no longer work even after I followed the update steps in the README for kernel version **5.15.0.100**. Has there been an adaptation for the newer kernel, or is it possible I’ve overlooked something that remain to be done to solve it?

lwfinger, 2024-03-12:
I need to see some error messages, or the full output of dmesg to help you. Do the command 'sudo dmesg > dmesg.txt' and attach dmesg.txt to this issue. I build the drivers with 5.15.0 - I have no idea what was done with your system.

weaki233, 2024-03-13:
Thank you for your prompt reply and assistance, here is the dmesg.txt attached.
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14580682/dmesg.txt)
It seems like the NVIDIA taints the kernel but I don't know how to solve it.

weaki233, 2024-03-13:
@lwfinger Thank you for your prompt reply and assistance, here is the dmesg.txt attached.
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14580720/dmesg.txt)
It seems like the NVIDIA taints the kernel but I don't know how to solve it.

weaki233, 2024-03-13:
@lwfinger Thank you for your prompt reply and assistance, here is the .txt file attached.
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14580738/dmesg.txt)

It seems like the NVIDIA taints the kernel but I don't know how to solve it.
The following are the lines that I think may cause the problem or you can see the file in detail.
 `[    2.025612] rtw89_8852ce 0000:08:00.0: failed to register hw
[    2.025625] rtw89_8852ce 0000:08:00.0: failed to register core hw
[    2.025634] rtw89_8852ce 0000:08:00.0: failed to register core
[    2.031723] nvidia: module license 'NVIDIA' taints kernel.
[    2.031725] Disabling lock debugging due to kernel taint
[    2.050508] usb 3-10: new full-speed USB device number 5 using xhci_hcd
[    2.079866] rtw89_8852ce: probe of 0000:08:00.0 failed with error -22
[    2.081240] nvidia-nvlink: Nvlink Core is being initialized, major device number 509`

weaki233, 2024-03-13:
@lwfinger Thank you for your prompt reply and assistance, here is the .txt file attached.
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14580738/dmesg.txt)

It seems like the NVIDIA taints the kernel but I don't know how to solve it.
The following are the lines that I think may cause the problem or you can see the file in detail.
'''
[    2.025612] rtw89_8852ce 0000:08:00.0: failed to register hw
[    2.025625] rtw89_8852ce 0000:08:00.0: failed to register core hw
[    2.025634] rtw89_8852ce 0000:08:00.0: failed to register core
[    2.031723] nvidia: module license 'NVIDIA' taints kernel.
[    2.031725] Disabling lock debugging due to kernel taint
[    2.050508] usb 3-10: new full-speed USB device number 5 using xhci_hcd
[    2.079866] rtw89_8852ce: probe of 0000:08:00.0 failed with error -22
[    2.081240] nvidia-nvlink: Nvlink Core is being initialized, major device number 509
'''

lwfinger, 2024-03-13:
You had a problem uploading dmesg.txt. When I click on it, I get a 404 error.

weaki233, 2024-03-13:
> You had a problem uploading dmesg.txt. When I click on it, I get a 404 error.

I am sorry about that but I don't know why either. What about this:

[dmesg](https://github.com/weaki233/dmesg.txt/blob/main/dmesg.txt)

lwfinger, 2024-03-13:
That worked, but with a problem. The first warning shows WARNING: CPU: 10 PID: 476 at net/wireless/core.c:879 wiphy_register+0xa5a/0xa80 [cfg80211]. Before I try to chase this one down, have you done a ;git pull' recently?

Any reason you are stuck at Ubuntu 20.04? There are a lot of improvements in later versions.

weaki233, 2024-03-13:
Yes I have done it following this about 10 hours ago:

> When your kernel changes, then you need to do the following:
```
cd ~/rtw89
git pull
make clean
make
sudo make install
```
Because I'm now collaborate with the team in my school and they told me to use this version to compatible with their current work.

parithy, 2024-05-27:
becon loss issue seems to be present in 6.8.0-31-generic  in ubuntu derivative..