lwfinger/rtw89 #305: ubuntu20.04 Linux 5.16.5 how can I solve this problem?

majiaqi669, 2024-01-18:
ubuntu20.04 Linux 5.16.5 how can I solve this problem?
[image]
[image]
thank you so much!

majiaqi669, 2024-01-18:
if I use old kernel I will meet another error
[image]

lwfinger, 2024-01-18:
I fixed the problem with kernel 5.10. Your problem with 5.15 is that your version of gcc is a lot older than the one used to compile the kernrl. That is not allowed.

majiaqi669, 2024-01-19:
thank you so much,It's solved!

closed 2024-01-19 by lwfinger