lwfinger/rtw89 #306: Need help installing driver for Openwrt OS

kinhsman, 2024-01-24:
Noob here! I attempted to install the driver on openwrt following the provided instructions but faced challenges as I couldn't locate the linux-headers and build-essential packages. Could someone kindly help me with this?
Thank you!

lwfinger, 2024-01-24:
What CPU are you using? Does it have a PCI interface? A quidck DuckDuckGo search located a number of articles.

More details please.

kinhsman, 2024-01-24:
The CPU is Intel N95. I was able to install the driver for Ubuntu on the same system, wifi working perfectly. I just cant seem to figure out how to install drivers on openwrt. 

> Does it have a PCI interface?

Yes the Wifi card is M2 E key with RTL8852BE chipset

I noticed rtw88 was available under kmod-rtl88 for openwrt, but I cannot find anything for rtw89

Below is the error im getting when trying to install the required packages
[image]

lwfinger, 2024-01-24:
try 'opkg | grep header'. Does that give you any clue?

razvanphp, 2024-02-12:
Is this driver supposed to work with M2?

I successfully compiled it, but it's not detecting the card as wifi card.

```
$ lsmod | grep rtw
rtw_8852be             16384  0
rtw_8852b             368640  1 rtw_8852be
rtw89pci               61440  1 rtw_8852be
rtw89core             479232  2 rtw89pci,rtw_8852b
mac80211              913408  2 rtw89pci,rtw89core
cfg80211              856064  3 rtw89core,mac80211,rtw_8852b
$
$ sudo iwconfig
lo        no wireless extensions.

dummy0    no wireless extensions.

eth0      no wireless extensions.

eth1      no wireless extensions.

l4tbr0    no wireless extensions.

rndis0    no wireless extensions.

usb0      no wireless extensions.

docker0   no wireless extensions.

$ lspci
0004:02:03.0 PCI bridge: Pericom Semiconductor Device b404 (rev 01)
0004:03:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
0004:04:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
$
```

closed 2025-04-13 by kinhsman