lwfinger/rtw89 #307: Issue with Kernel 6.6.x

gpastal24, 2024-01-24:
Hello,

The drivers do not seem to work for Linux Kernel 6.6.x.
When I try to modprobe the module it says "error: device is busy".
When I load kernel 5.15.x they work fine.

Do you have any idea what is going on here?

lwfinger, 2024-01-24:
Yes.  You have interference between the drivers in the kernel and those in this repo. With 6.6, you do NOT need this repo. Use the in-kernel drivers.

gpastal24, 2024-01-24:
You are the best , thanks

closed 2024-01-24 by gpastal24

privacyfreak84, 2024-02-11:
How do I use the in kernel drivers after building this one?

gpastal24, 2024-02-11:
@privacyfreak84
Sudo make uninstall inside rwt889

privacyfreak84, 2024-02-11:
I did that but it didn’t fix anything. WiFi still isn’t back. I just made an issue about it: https://github.com/lwfinger/rtw89/issues/311