lwfinger/rtw89 #308: rtw89_8852be: after resume WIFI is broken:

femiveys, 2024-01-30:
# Problem
Resume from suspend doesn't work for me with this driver.
When I suspend its seems to work.
When I hit resume (enter or mouse click), it takes a bit more than a minute before the laptop resumes.
When resumed the WIFI (and Bluetooth) doesn't work anymore.

# Context

- Laptop: HP 15-fd0008nb
- OS: Ubuntu 23.10

```
$ uname -r
6.5.0-15-generic
```

```
$ lshw -class network
  *-network                 
       description: Wireless interface
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0000:01:00.0
       logical name: wlo1
       version: 00
       serial: <PRIVATE>
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_8852be driverversion=6.5.0-15-generic firmware=N/A ip=192.168.0.173 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:134 ioport:3000(size=256) memory:80500000-805fffff
```

I tried it with the driver as is and also with the tweaks I could find in the [documentation](https://github.com/lwfinger/rtw89).

# journalctl log with driver loaded as is (clean)
- [journalctl right after suspend](https://github.com/lwfinger/rtw89/files/14101799/clean.suspend.log)
- [journalctl right after resume](https://github.com/lwfinger/rtw89/files/14101803/clean.resume.log)

(I don't know how to keep the colors)

The first error is: `rtw89_8852be 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible`

# Driver loaded with modprobe.conf file and systemd script

## systemd script (executable)
I have added following to `/usr/lib/systemd/system-sleep/suspend_rtw89_8852be` as explained [here](https://github.com/lwfinger/rtw89#problem-with-recovery-after-sleep-or-hibernation).

```
#!/bin/sh

# https://github.com/lwfinger/rtw89?tab=readme-ov-file#option-configuration
echo "Before if"
echo $1
if [ "${1}" = "pre" ]; then
  echo "FEV: in PRE1"
  sudo modprobe -rv rtw_8852be
  echo "FEV: in PRE2"
  sudo modprobe -rv rtw89core
elif [ "${1}" = "post" ]; then
  echo "FEV: in POST"
  modprobe -v rtw_8852be
fi
```

## modprobe configuration
I have added following to `/etc/modprobe.d/rtw8852be.conf` as explained [here](https://github.com/lwfinger/rtw89#option-configuration).

```
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq
options rtw89_core disable_ps_mode=y
options rtw89core disable_ps_mode=y
```
- [journalctl log right after suspend](https://github.com/lwfinger/rtw89/files/14101795/tweaked.suspend.log)
- [journalctl log right after resume](https://github.com/lwfinger/rtw89/files/14101796/tweaked.resume.log)

Any idea why resume doesn't work?

lwfinger, 2024-01-30:
Dmesg says EXACTLY why resume fails. Your BIOS refuses to change power state - that is a BIOS bug.

If you look at README.md, and follow the explicit instructions, you will find a fix.

closed 2024-01-30 by lwfinger

femiveys, 2024-01-31:
I see that I attached wrong logs. Due to a kernel update, the original driver was loaded and not the one of this repo. I'm very sorry for the confusion.

Here are the correct logs:
- Without tweaks
  - suspend: [clean.suspend.log](https://github.com/lwfinger/rtw89/files/14105440/clean.suspend.log)
  - resume: [clean.resume.log](https://github.com/lwfinger/rtw89/files/14105441/clean.resume.log)
- With tweaks
  - suspend: [tweaked.suspend.log](https://github.com/lwfinger/rtw89/files/14105443/tweaked.suspend.log)
  - resume: [tweaked.resume.log](https://github.com/lwfinger/rtw89/files/14105442/tweaked.resume.log)

I applied the tweaks that are clearly stated in the Readme.md (thanks for that):
- [Problem with recovery after sleep or hibernation](https://github.com/lwfinger/rtw89#problem-with-recovery-after-sleep-or-hibernation)

> Some BIOSs have trouble changing power state from D3hot to D0. If you have this problem, then
> 
> `sudo cp suspend_rtw89 /usr/lib/systemd/system-sleep/`
> 
> That script will unload the driver before sleep or hibernation, and reload it following resumption.

- [Option configuration](https://github.com/lwfinger/rtw89#option-configuration)

> IMPORTANT: If you have an HP or Lenovo laptop, Their BIOS does not handle the PCIe interface correctly. To compensate, run the following command: sudo cp 70-rtw89.conf /etc/modprobe.d/. Then unload the drivers and reload. You should see the options appended to the end of the rtw89_pci or rtw89pci load line.

What I see is:
- without the tweaks it is `rtw89_8852be` that complains:
`kernel: rtw89_8852be 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible`
- with **both tweaks applied** it is `pci` that complains:
`kernel: pci 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible`
`

If we only look at the errors (priority === 3) during resume I have these errors:
- without the tweaks:
```
jan 30 23:59:34 Cataleya kernel: rtw89_8852be 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible
jan 30 23:59:34 Cataleya kernel: ACPI Error: No handler for Region [CMS0] (00000000273e5f7b) [SystemCMOS] (20230331/evregion-130)
jan 30 23:59:34 Cataleya kernel: ACPI Error: Region SystemCMOS (ID=5) has no handler (20230331/exfldio-261)
jan 30 23:59:34 Cataleya kernel: ACPI Error: Aborting method \_SB.PC00.LPCB.EC0._Q33 due to previous error (AE_NOT_EXIST) (20230331/psparse-529)
jan 30 23:59:34 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
jan 30 23:59:34 Cataleya wpa_supplicant[797]: Could not set interface wlo1 flags (UP): Connection timed out
jan 30 23:59:34 Cataleya wpa_supplicant[797]: nl80211: Could not set interface 'wlo1' UP
jan 30 23:59:34 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
jan 30 23:59:35 Cataleya wpa_supplicant[797]: Could not set interface wlo1 flags (UP): Connection timed out
jan 30 23:59:35 Cataleya wpa_supplicant[797]: WEXT: Could not set interface 'wlo1' UP
jan 30 23:59:35 Cataleya wpa_supplicant[797]: wlo1: Failed to initialize driver interface
jan 30 23:59:35 Cataleya NetworkManager[789]: <error> [1706655575.0958] device (wlo1): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
jan 30 23:59:35 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
jan 30 23:59:42 Cataleya bluetoothd[699]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
jan 30 23:59:45 Cataleya wpa_supplicant[797]: Could not set interface wlo1 flags (UP): Connection timed out
jan 30 23:59:45 Cataleya wpa_supplicant[797]: nl80211: Could not set interface 'wlo1' UP
jan 30 23:59:45 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
jan 30 23:59:45 Cataleya wpa_supplicant[797]: Could not set interface wlo1 flags (UP): Connection timed out
jan 30 23:59:45 Cataleya wpa_supplicant[797]: WEXT: Could not set interface 'wlo1' UP
jan 30 23:59:45 Cataleya wpa_supplicant[797]: wlo1: Failed to initialize driver interface
jan 30 23:59:45 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
jan 30 23:59:45 Cataleya NetworkManager[789]: <error> [1706655585.6909] device (wlo1): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
jan 30 23:59:56 Cataleya wpa_supplicant[797]: Could not set interface wlo1 flags (UP): Connection timed out
jan 30 23:59:56 Cataleya wpa_supplicant[797]: nl80211: Could not set interface 'wlo1' UP
jan 30 23:59:56 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
jan 30 23:59:56 Cataleya wpa_supplicant[797]: Could not set interface wlo1 flags (UP): Connection timed out
jan 30 23:59:56 Cataleya wpa_supplicant[797]: WEXT: Could not set interface 'wlo1' UP
jan 30 23:59:56 Cataleya wpa_supplicant[797]: wlo1: Failed to initialize driver interface
jan 30 23:59:56 Cataleya NetworkManager[789]: <error> [1706655596.6929] device (wlo1): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
jan 30 23:59:56 Cataleya kernel: rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
```
- with **both tweaks applied**:
```
jan 30 23:39:05 Cataleya kernel: pci 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible
[... cut ...]
jan 30 23:39:12 Cataleya bluetoothd[699]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

There are already much less errors, but resuming still doesn't work:
- `pci` complains about being `Unable to change power state from D3cold to D0`
- this module (`rtw89_8852be`) complains about the same and about `no suitable firmware`

Is this a bug in the `pci` module or am I doing something wrong? Any idea how I could fix this?

I don't think it is time yet to close this issue as apparently something is still wrong.

lwfinger, 2024-01-31:
I never said anything about the options, they lead to different dmesg errors.

##### Problem with recovery after sleep or hibernation
Some BIOSs have trouble changing power state from D3hot to D0. If you have this problem, then

sudo cp suspend_rtw89 /usr/lib/systemd/system-sleep/.

That script will unload the driver before sleep or hibernation, and reload it following resumption.

You will need to edit the file to select your device.

femiveys, 2024-01-31:
As explained in my initial post, I have done this and in the logs it can be seen this works perfectkly, but resume still doesn't work.

This is how I changed the script:

```sh
#!/bin/sh

# https://github.com/lwfinger/rtw89?tab=readme-ov-file#option-configuration
echo "Before if"
echo $1
if [ "${1}" = "pre" ]; then
  echo "FEV: in PRE1"
  sudo modprobe -rv rtw_8852be
  echo "FEV: in PRE2"
  sudo modprobe -rv rtw89core
elif [ "${1}" = "post" ]; then
  echo "FEV: in POST"
  modprobe -v rtw_8852be
fi
```

femiveys, 2024-01-31:
In the meanwhile I have found the solution/workaround and I want to share it here so others can also benefit from it.

[This post](https://bbs.archlinux.org/viewtopic.php?pid=2136349#p2136349) pointed me to a solution that works.

The problem is actually the PCIe as a whole.

By adding following [kernel parameter](https://wiki.ubuntu.com/Kernel/KernelBootParameters) resuming works like a charm:

```
pcie_port_pm=off
```

With this kernel parameter actually also none of the tweaks are needed. Those tweaks fix it for the WIFI and Bluetooth but apparently power management for PCIe as a whole is broken. This also explains what I saw:

> What I see is:
>
> - without the tweaks it is `rtw89_8852be` that complains:
> `kernel: rtw89_8852be 0000:01:00.0: Unable to change power state from D3cold to D0, device inaccessible`
> - with both tweaks [...]

Now I'm not sure what the effect is of turning off the Power Management on  PCIe, but suspening and resuming works fine now.

joeyparrish, 2024-12-29:
@femiveys, thank you for this!  I struggled for ages before I found this incredibly useful post.  This should be shown somewhere much more prominently!