lwfinger/rtw89 #314: rtw89_8852be  firmware failed to ack for entering ps mode...

pbadun, 2024-02-18:
Good afternoon!
What could be the problem with these messages?

```
[40608.541770] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40610.529865] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40612.513905] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40614.526122] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40616.801907] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40619.613951] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40623.712553] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40625.570490] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
[40629.599827] rtw89_8852be 0000:2d:00.0: firmware failed to ack for entering ps mode
```
~# uname -r
6.1.0-18-amd64

~# ls -l /lib/firmware/rtw89/
-rw-r--r-- 1 root root 1423232 мая  1  2023 rtw8852a_fw.bin
-rw-r--r-- 1 root root 1242808 янв  5 13:40 rtw8852b_fw-1.bin
-rw-r--r-- 1 root root 1035232 мая  1  2023 rtw8852b_fw.bin
-rw-r--r-- 1 root root 1532656 мая  1  2023 rtw8852c_fw.bin

lwfinger, 2024-02-18:
I suspect that you have a Lenovo or HP laptop. Their problem is an improperly coded BIOS. Read the section of README.md that starts with ##### Option configuration and follow the instructions there.

pbadun, 2024-02-18:
Yes, Lenovo ThinkPad E15 , I know rtw_8852be, can you tell me what name of the configuration file is needed and what to write in the inside...

options disable_ps_mode=true ???