lwfinger/rtw89 #318: After last update wifi not wortking

BUGOVER, 2024-03-15:
Ubuntu 22.04 
Lenovo Thinkbook 14p Gen3 Amd

ghostintheshell0, 2024-03-15:
the same problem
i see all networks but get "Activation of network connection failed".
Lenovo ideapad gaming 3

lwfinger, 2024-03-15:
Did you change anything else? Whenever you report a problem, a synopsis of the output of dmesg is useful.

ghostintheshell0, 2024-03-15:
@lwfinger 
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14618760/dmesg.txt)
[make.log](https://github.com/lwfinger/rtw89/files/14618770/make.log)

lwfinger, 2024-03-15:
What version of Ubuntu are you running? Your system issues an authenticate, but never receives a reply.

ghostintheshell0, 2024-03-15:
@lwfinger Ubuntu 22.04.4 LTS

555cider, 2024-03-17:
I’m in the same situation. My OS version is Ubuntu 22.04 LTS, and the kernel version is 6.5.0-25-generic. When I downgraded the kernel version to 6.5.0-21-generic, my internet connection was restored.

lwfinger, 2024-03-17:
There are a couple of things you need to do: 1. Install firmware file rtw8852b_fw-1.bin. Your system is too old to handle the latest firmware. 2. There is a problem with the latest version on Ubuntu 20.04.6, which I installed. You need to do a 'git checkout https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61' and build and install that version, which works on my Ubuntu installation. When I have time, I will try to sort out the problem and notify you when that happens.

njutn95, 2024-04-22:
I have to confirm that the latest commit does not work on OpenSUSE Tumbleweed either. Wifi networks appear, but trying to connect to any of them leads to an error that it cannot connect to as reported in the OP. Rolling back to the commit mentioned above (https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61) fixed it for me as well. 

Kernel in question (if relevant): 6.8.6-1-default

lwfinger, 2024-04-22:
I have more info. I updated my local copy up to that of Realtek's repo and checked line-by-line with the diff command. The resulting code works with mainline 6.9.0-rc5, but fails with openSUSE's 6.8.7. Something is wrong with the 6.8.X versions. OK, another bisection.

njutn95, 2024-04-22:
If you need any SUSE testing, on kernel 6.8.x, let me know.

bill-mcgonigle, 2024-05-20:
6dc9441 also works for Debian Bookworm (6.1.0-21-amd64) on a GMKtec NucBox G3.

I couldn't quickly figure out how to get dkms to work with git checkout.

dhanak, 2024-06-23:
> [6dc9441](https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61) also works for Debian Bookworm (6.1.0-21-amd64) on a GMKtec NucBox G3.
> 
> I couldn't quickly figure out how to get dkms to work with git checkout.

I'm glad I found this thread. It worked for me as well! To make the dkms work again, just cherry-pick `5f6b2f9` on the top of `6dc9411`.