lwfinger/rtw89 #319: Seems to build correctly, still not appearing at all. 8852ce

cmf2201, 2024-03-15:
Hello!
I have the 8852ce driver, and when I run
`lsmod | grep rtw`
I get
`rtw_8852ce             16384  0 rtw_8852c             974848  1 rtw_8852ce rtw89pci               81920  1 rtw_8852ce rtw89core             761856  2 rtw_8852c,rtw89pci mac80211             1249280  2 rtw89pci,rtw89core cfg80211              970752  3 rtw_8852c,rtw89core,mac80211 `
so it seems like everything should be working, but the wireless connection option still does not exist. I've been going at this for a few hours now and was looking for some help.
Here's the output of dmesg :
[dmesg.txt](https://github.com/lwfinger/rtw88/files/14619691/dmesg.txt)

lwfinger, 2024-03-15:
As I wrote there, your firmware is missing.

cmf2201, 2024-03-15:
I think I installed the firmware, but I am not sure. I did 
`git clone git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git`
will this install the firmware?

lwfinger, 2024-03-15:
No, you still need to 'sudo cp ~/linux-firmware/rtw89/* /lib/firmware/rtw89/.'

cmf2201, 2024-03-15:
Got it! Seems that the firmware is now properly installed, but I think now there is a problem with the computer registering the hardware? On startup, I see these three lines:
`rtw89_8852ce 0000:04:00.0: failed to register hw
rtw89_8852ce 0000:04:00.0: failed to register core hw
rtw89_8852ce 0000:04:00.0: failed to register core`

Here is the new dmesg:

[dmesg.txt](https://github.com/lwfinger/rtw89/files/14620771/dmesg.txt)

lwfinger, 2024-03-15:
The problem is a WARNING from mac80211 that a channel offset is wrong. I installed Ubuntu 20.04.6, and it worked correctly for an RTW8852BE - I do not hace a CE, but the BE uses the same core as the CE. Have you installed the package wireless-regdb?

ghdeleo, 2024-03-15:
Hello! Writing from a different Github account. I have installed that package.

cmf2201, 2024-03-21:
I have installed the package

lwfinger, 2024-03-21:
I need to see the dmesg output. Do 'sudo dmesg > dmesg.txt' and attach file dmesg.txt to this issue.

JalorOo, 2024-03-22:
I have installed that wireless-regdb. and my dmesg.txt is 
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14717260/dmesg.txt), more detail is in #323 , i seems having the same problems [cmf2201](https://github.com/cmf2201) meet.

cmf2201, 2024-03-24:
Here is the 
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14733984/dmesg.txt)

lwfinger, 2024-03-24:
Your dmesg has the line WARNING: CPU: 2 PID: 420 at net/wireless/core.c:879 wiphy_register+0xa5a/0xa80 [cfg80211], which is the problem. That is a common problem with Ubuntu 20.04.2. I have duplicated the issue, but have not had time to research the problem, but 'git checkout 6dc944' followed by a 'make' and 'sudo make install' has helped for some people.

jiangtaohu1206, 2024-03-24:
@lwfinger I also encountered the same problem above, tried the method you said and then it works!! Thank you very much for your timely help.

cmf2201, 2024-03-24:
This worked! Thank you so much.

closed 2024-03-24 by cmf2201

eggehad, 2025-06-07:
The first release of Red Hat Enterprise Linux 10, just released, has a corrupted rtw89 image in the firmware_files that comes with that release.  I can't find anyplace (that i don't have to pay for), to alert them to this problem either.   But thanks to you all, I was able to figure out the problem in a few hours.  The following steps fix RHEL 10 :

1. git clone git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
2. sudo cp ~/linux-firmware/rtw89/* /lib/firmware/rtw89/
3. sudo rm /lib/firmware/rtw89/*.xz
4. reboot