lwfinger/rtw89 #320: rtw89_8852be: Unknown firmware header version 10

krausemann, 2024-03-16:
Hi!

I am new to compiling kernel modules and it is somehow exciting. For reasons i am stuck to using _Ubuntu 20.04_ with their kernel `5.15.0-83-generic`. Wifi module is `Realtek RTL8852BE` on _Lenovo_ Laptop. I followed the instruction in `README.md` but i am stuck at this point. From taking a peek in `fw.c`, the version should be `1` or `0`. Can you help me to get it working? See below my `kern.log` and what i have done so far.

kind regards!

# kern.log

during boot, `/var/log/kern.log` shows following error:
```
Mar 16 23:01:54 my-pc kernel: [   31.264741] rtw89_8852be 0000:09:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
Mar 16 23:01:54 my-pc kernel: [   31.265179] rtw89_8852be 0000:09:00.0: enabling device (0000 -> 0003)
Mar 16 23:01:54 my-pc kernel: [   31.267814] rtw89_8852be 0000:09:00.0: Unknown firmware header version 10
Mar 16 23:01:54 my-pc kernel: [   31.267819] rtw89_8852be 0000:09:00.0: failed to recognize firmware
Mar 16 23:01:54 my-pc kernel: [   31.267821] rtw89_8852be 0000:09:00.0: failed to setup chip information
Mar 16 23:01:54 my-pc kernel: [   31.269016] rtw89_8852be: probe of 0000:09:00.0 failed with error -2
```

# What i have done so far:

## Disable Fast Startup in Windows (dual boot)
It has been reported it can impact the availability of wifi chip. https://askubuntu.com/a/1303340

## get vmlinux to get rid of `Skipping BTF generation`
`make` printed `Skipping BTF generation [my module name] due to unavailabilty of vmlinux.`
Since my ubuntu version seems to use different vmlinux location, i did the following to successfully compile:

```
sudo apt install dwarves
sudo cp /sys/kernel/btf/vmlinux /usr/lib/modules/$(uname -r)/build/\
```

## download binary blobs to solve Direct firmware load
Kernel log gave error: `Direct firmware load for rtw89/rtw8852b_vw-1.bin failed with error -2` because my os does not ship the required binaries. As instructed in `README`, i obtained them from [git.kernel.org](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89) and copied them: `sudo cp *.bin /lib/firmware/rtw89/`

## Arguments to fix issues with lenovo BIOS
as instructed in `README` i did following:
`sudo cp 70-rtw89.conf /etc/modprobe.d/` and reloaded the modules.

lwfinger, 2024-03-16:
There are a couple of things you need to do: 1. Install firmware file rtw8852b_fw-1.bin. Your system is too old to handle the latest formware. 2. There is a problem with the latest version on Ubuntu 20.04.6, which I installed. You need to do a 'git checkout 6dc9441698a7f2' and build and install that version, which works on my Ubuntu installation. When I have time, I will try to sort out the problem and notify you when that happens.

krausemann, 2024-03-17:
Hi thanks!

> 1. Install firmware file rtw8852b_fw-1.bin.

apart from copying it to `/lib/firmware/rtw89/`, is there anything i need to do?

> 2. There is a problem with the latest version on Ubuntu 20.04.6, which I installed. You need to do a 'git checkout https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61' and build and install that version, which works on my Ubuntu installation.

Unfortunately gives me the same `/var/log/kern.log` messages. I did following:
```
git checkout 6dc9441
git clean -d -x -f
make clean
make
make sign-install
```
reboot and signing MOK procedure

also tried reloading the module `sudo modprobe rtw_8852be`

as well as disabling it in recovery mode, uninstalling it, newly installing it and reloading it.

My OS is 20.04.6 as well. Is there something i missed or do you have other hints to try?

Greetings!

lwfinger, 2024-03-17:
If it still does not work, then yu need to try the various options as your BIOS is certainly buggy. On my system, those steps resulted in a working wifi.

krausemann, 2024-03-17:
Thanks for your help. 
unfortunately it did not work. btw, laptop is _lenovo loq 15IRH8_. 

* updating BIOS did not do the trick. 
* using `options rtw89core debug_mask` gives `modprobe: ERROR: could not insert 'rtw_8852be': Invalid argument.
* using all other options together, (which were also showing up the `modprobe -v` calls) gave the same `kern.log` output as above.

```
# set options for faulty HP and Lenovo BIOS code
# options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y
options rtw89core disable_ps_mode=y           
```
```
user@my-pc:~$ sudo modprobe -v rtw_8852be
insmod /lib/modules/5.15.0-83-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko disable_ps_mode=y 
insmod /lib/modules/5.15.0-83-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw89pci.ko disable_aspm_l1=y disable_aspm_l1ss=y disable_clkreq=y 
insmod /lib/modules/5.15.0-83-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852b.ko 
insmod /lib/modules/5.15.0-83-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8852be.ko 
```

I appreciate your effort! Greetings!

lwfinger, 2024-03-17:
Updating the BIOS will not help. Those two families of laptops have a BIOS that handles the PCIe bus strangely, and they do not get along with the Realtek devices. Report this to linux-wireless@vger.kernel.org to see if Realtek has any fixes that I do not know about.

mad-cat, 2024-03-30:
Hi, @krausemann
I've faced the same issue on Lenovo Thinkbook 15 G4.
lwfinger's driver worked fine, but after installation Debian 12.5.0 it failed.
Have you resolved the problem with Realtek RTL8852BE?

krausemann, 2024-03-31:
Hi @rtw89 unfortunately i have not resolved the issue. Didn't find time also to report the issue as suggested above. For now, i arranged with using good old ethernet cable. If you found a solution, let me know!

lwfinger, 2024-03-31:
One user has reported that updating to sid resolved their issues.

It appears that Ubuntu 20.04.4 and some variants of Debian have problems in their user-space code. I am trying to resolve the Ubuntu issue.

a5a5aa555oo, 2025-07-08:
@ackPeng 

Try these two drivers.

For kernel 5.15 ~ 6.5 ==> https://github.com/a5a5aa555oo/rtw89
For kernel 6.6 ~ 6.15 ==> https://github.com/morrownr/rtw89

ackPeng, 2025-07-09:
@a5a5aa555oo  I'm really sorry to hear this news, thank you very much for your help