lwfinger/rtw89 #322: wifi and driver issues with kernel 6.8.1

mauricioquintela, 2024-03-20:
Recently updated to kernel 6.8.1 on arch and I'm having some authentication issues. To better debug and fix, I have started from the beginning in attempts to find the root cause. Lenovo laptop, so the section on those in the README applies. 

If I attempt to use the in-kernel drivers by doing the opposite of the README arch instructions: unninstalling both https://aur.archlinux.org/rtw89-dkms-git.git and doing a sudo make unninstall in the git root folder (followed by reinstalling the kernel+headers just in case), no wifi adapter shows up in network manager and dmesg logs show some errors regarding unknown symbols [dmesg_kernel-drivers.txt](https://github.com/lwfinger/rtw89/files/14667106/dmesg_kernel-drivers.txt).

Upon installing [rtw89-dkms-git](https://aur.archlinux.org/rtw89-dkms-git.git) as per the README and rebooting, wifi adapter shows up again in network manager and the unknown symbols dissapear from dmesg (see [dmesg_dkms-only.txt](https://github.com/lwfinger/rtw89/files/14667180/dmesg_dkms-only.txt)), although connecting to known networks isn't possible. According to journal, issue is regarding "wpa_supplicant[1026]: wlan0: SME: Authentication request to the driver failed", see line 2577 onwards: [journal_dkms-only.txt](https://github.com/lwfinger/rtw89/files/14667176/journal_dkms-only.txt).

After make-ing this repo's driver and rebooting while blacklisting kernel drivers (although I'm still unsure about this step as no drivers appear to exist without having [rtw89-dkms-git](https://aur.archlinux.org/rtw89-dkms-git.git) installed) via
```
> cat /etc/modprobe.d/70-rtw89.conf
# blacklist kernel driver 
blacklist rtw89_8852be
# set options for faulty HP and Lenovo BIOS code
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss=y
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss=y
```
no issues are present in dmesg logs (see [dmesg_rtw89.txt](https://github.com/lwfinger/rtw89/files/14667429/dmesg_rtw89.txt)). Nonetheless, the same issue is present in journal, where "wlan0: SME: Authentication request to the driver failed" still shows up and connecting to a network is not possible, see line 2396 onwards: [journal_rtw89.txt](https://github.com/lwfinger/rtw89/files/14667507/journal_rtw89.

No error for "wrong wifi password" or anything similar is presented, with the network manager widget just spinning as it attempts to connect multiple times.

lwfinger, 2024-03-20:
It is not in my job description to debug ARCH kernels. I can assure you that kernel 6.8.0 works fine, at least when it was built from the mainline sources and run with openSUSE user-mode networking. Did anything else change when you "upgraded" to 6.8.1?

The changes that you observed by implementing DKMS, particularly problems with global symbols, were most likely a problem with mixing modules from this repo with those from the kernel. By the way, the 6.8.X kernel drivers are the same as those from the repo and you should not need to switch away from the kernel versions.

mauricioquintela, 2024-03-21:
thank you very much. i've fixed the issue by getting a new pcie wifi card, becoming free of realtek hell. thx for the support until now

closed 2024-03-21 by mauricioquintela

lwfinger, 2024-03-21:
Just a comment - 6.8.1 from openSUSE is working just fine. T=You may not be free of ARCH hell.

FunkyloverOne, 2024-05-04:
Hey @lwfinger, I have a question about this:
> By the way, the 6.8.X kernel drivers are the same as those from the repo and you should not need to switch away from the kernel versions.

I have noticed that when I am on Ubuntu (and derivatives) and check my speed at https://www.speedtest.net/, the graph looks perfect - I instantly get the maximum speed, and it stays that way. (And I did not manually install the driver).
However, whenever I use any other distro, like Fedora, Arch, and currently openSUSE Tumbleweed (I have just installed it today) - the graph does not look so good, the download speed and ping are not stable. (Again, this is out of the box, I did not install the driver from the repo).

Also when I'm at my parents house, with not so good Wi-Fi - Ubuntu handles that connection way better than other distros (it was Fedora the last time I checked).
I would like to be not dependent on Ubuntu only, therefore I am curious if installing your driver makes sense on fresh openSUSE Tumbleweed. Or maybe you have any suggestions? My Wi-Fi card is RTL8852AE.

I would appreciate any feedback, and I am very grateful for your work! I used your driver way back when my older system was based on kernel 5.15 or something like that.
The reason I'm asking - is that I do not know how to follow your current README instructions properly yet, I haven't blacklisted any kernel modules before.

FunkyloverOne, 2024-05-08:
UPDATE: I have noticed this issue here: https://github.com/lwfinger/rtw89/issues/324
It looks like my firmware is very old, here's a sample from the `dmesg` output:
```
[    7.049228] rtw89_8852ae 0000:01:00.0: loaded firmware rtw89/rtw8852a_fw.bin
[    7.049324] rtw89_8852ae 0000:01:00.0: enabling device (0000 -> 0003)
[    7.051735] rtw89_8852ae 0000:01:00.0: Firmware version 0.13.36.0 (c33d3f88), cmd version 0, type 1
[    7.051744] rtw89_8852ae 0000:01:00.0: Firmware version 0.13.36.0 (c33d3f88), cmd version 0, type 3
```