lwfinger/rtw89 #323: It seems can't work in my ubuntu 20.04

JalorOo, 2024-03-22:
I use the `lsmod | grep rtw`, the output as follow:
```
rtw_8852ce             16384  0
rtw_8852c             974848  1 rtw_8852ce
rtw89pci               81920  1 rtw_8852ce
rtw89core             765952  2 rtw_8852c,rtw89pci
mac80211             1249280  2 rtw89pci,rtw89core
cfg80211              970752  3 rtw_8852c,rtw89core,mac80211
```
the dmesg.txt is follow:
[dmesg.txt](https://github.com/lwfinger/rtw89/files/14717212/dmesg.txt)

What should i do?

JalorOo, 2024-03-22:
run `lspci -nn | grep -i net`
```
05:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller [10ec:8125] (rev 05)
06:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:c852] (rev 01)
```

lwfinger, 2024-03-22:
There is a problem with Ubuntu 20.04. The workaround is in https://github.com/lwfinger/rtw89/issues/320. The clue was the line that said:
[    3.832328] WARNING: CPU: 8 PID: 567 at net/wireless/core.c:879 wiphy_register+0xa5a/0xa80 [cfg80211]
I am still trying to sort that out, but I have more pressing problems at the moment.

JalorOo, 2024-03-23:
> There is a problem with Ubuntu 20.04. The workaround is in https://github.com/lwfinger/rtw89/issues/320. The clue was the line that said:
>
> [    3.832328] WARNING: CPU: 8 PID: 567 at [...]

Ok,thank you, If you had solved this problem, please let me know. Although my problem has not been solved at present, thank you very much for developing this driver

JalorOo, 2024-03-24:
in the rtw89 folder use terminal input `git chekcout 6dc944`, then `make` and `sudo maker install` it work!!!