lwfinger/rtw89 #324: on the kernel version 6.8.X wifi works with delays

nudimannui4e, 2024-03-29:
```
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=1 ttl=114 time=45.4 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=2 ttl=114 time=12.8 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=3 ttl=114 time=87.3 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=4 ttl=114 time=11.3 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=5 ttl=114 time=35.6 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=6 ttl=114 time=54.0 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=7 ttl=114 time=279 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=8 ttl=114 time=12.3 ms
64 bytes from waw07s06-in-f14.1e100.net (142.250.203.142): icmp_seq=10 ttl=114 time=12.4 ms

--- google.com ping statistics ---
10 packets transmitted, 9 received, 10% packet loss, time 9029ms
rtt min/avg/max/mdev = 11.301/61.175/279.410/80.865 ms

and to my local wifi-router
PING 192.168.68.1 (192.168.68.1) 15000(15028) bytes of data.
15008 bytes from 192.168.68.1: icmp_seq=1 ttl=64 time=56.6 ms
15008 bytes from 192.168.68.1: icmp_seq=2 ttl=64 time=127 ms
15008 bytes from 192.168.68.1: icmp_seq=3 ttl=64 time=5.55 ms
15008 bytes from 192.168.68.1: icmp_seq=4 ttl=64 time=272 ms
15008 bytes from 192.168.68.1: icmp_seq=5 ttl=64 time=6.03 ms
15008 bytes from 192.168.68.1: icmp_seq=6 ttl=64 time=56.2 ms
15008 bytes from 192.168.68.1: icmp_seq=7 ttl=64 time=5.48 ms
15008 bytes from 192.168.68.1: icmp_seq=8 ttl=64 time=53.8 ms
15008 bytes from 192.168.68.1: icmp_seq=9 ttl=64 time=6.63 ms
15008 bytes from 192.168.68.1: icmp_seq=10 ttl=64 time=4.55 ms

--- 192.168.68.1 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9016ms
rtt min/avg/max/mdev = 4.554/59.393/272.486/80.267 ms
```

when i try Ubuntu live-cd, same driver has no delay
[image]

nudimannui4e, 2024-03-29:
dmesg
```
[Fri Mar 29 14:11:23 2024] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[Fri Mar 29 14:25:02 2024] wlan0: disconnect from AP c2:25:2f:74:67:2f for new auth to c2:25:2f:74:80:3f
[Fri Mar 29 14:25:02 2024] wlan0: authenticate with c2:25:2f:74:80:3f (local address=40:1a:58:65:ca:19)
[Fri Mar 29 14:25:02 2024] wlan0: send auth to c2:25:2f:74:80:3f (try 1/3)
[Fri Mar 29 14:25:02 2024] wlan0: authenticated
[Fri Mar 29 14:25:02 2024] wlan0: associate with c2:25:2f:74:80:3f (try 1/3)
[Fri Mar 29 14:25:02 2024] wlan0: RX ReassocResp from c2:25:2f:74:80:3f (capab=0x1011 status=0 aid=3)
[Fri Mar 29 14:25:03 2024] wlan0: associated
[Fri Mar 29 14:25:03 2024] wlan0: Limiting TX power to 35 (35 - 0) dBm as advertised by c2:25:2f:74:80:3f
[Fri Mar 29 14:27:31 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 14:27:31 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 14:55:08 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 14:55:08 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 14:55:36 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 14:55:36 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 14:55:54 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 14:55:54 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:00:28 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:00:28 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:00:51 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:00:51 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:01:19 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:01:19 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:08:39 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:08:39 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:14:58 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:14:58 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:15:11 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:15:11 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:33:45 2024] wlan0: deauthenticating from c2:25:2f:74:80:3f by local choice (Reason: 3=DEAUTH_LEAVING)
[Fri Mar 29 15:34:18 2024] rtw89_8852ce 0000:03:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[Fri Mar 29 15:34:18 2024] rtw89_8852ce 0000:03:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 1
[Fri Mar 29 15:34:18 2024] rtw89_8852ce 0000:03:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 3
[Fri Mar 29 15:34:18 2024] rtw89_8852ce 0000:03:00.0: chip rfe_type is 1
[Fri Mar 29 15:35:27 2024] rtw89_8852ce 0000:03:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[Fri Mar 29 15:35:27 2024] rtw89_8852ce 0000:03:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 1
[Fri Mar 29 15:35:27 2024] rtw89_8852ce 0000:03:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 3
[Fri Mar 29 15:35:27 2024] rtw89_8852ce 0000:03:00.0: chip rfe_type is 1
[Fri Mar 29 15:35:39 2024] wlan2: authenticate with c2:25:2f:74:80:3f (local address=40:1a:58:65:ca:19)
[Fri Mar 29 15:35:39 2024] wlan2: send auth to c2:25:2f:74:80:3f (try 1/3)
[Fri Mar 29 15:35:39 2024] wlan2: authenticated
[Fri Mar 29 15:35:39 2024] wlan2: associate with c2:25:2f:74:80:3f (try 1/3)
[Fri Mar 29 15:35:39 2024] wlan2: RX AssocResp from c2:25:2f:74:80:3f (capab=0x1011 status=0 aid=3)
[Fri Mar 29 15:35:39 2024] wlan2: associated
[Fri Mar 29 15:35:39 2024] wlan2: Limiting TX power to 35 (35 - 0) dBm as advertised by c2:25:2f:74:80:3f
[Fri Mar 29 15:39:57 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:39:57 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c
[Fri Mar 29 15:46:01 2024] rtw89_8852ce 0000:03:00.0: no tx fwcmd resource
[Fri Mar 29 15:46:01 2024] rtw89_8852ce 0000:03:00.0: failed to send h2c

```

lwfinger, 2024-03-30:
The only thing I see is that you have old firmware. My system shows:

[    6.722728] rtw89_8852be 0000:02:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    6.725835] rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[    6.725841] rtw89_8852be 0000:02:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3

If your distro does not provide the latest, the README.md tells you how to obtain it.

nudimannui4e, 2024-04-03:
doesn't work for me(
arch linux 6.8.2-arch2-1
tried: 
```bash
$ pacman -Qi rtw89-dkms-git                                                                                                                                                                       

Name            : rtw89-dkms-git
Version         : 1:r980.48680ab-1
Description     : Driver for Realtek 8852AE, an 802.11ax device
Architecture    : x86_64
URL             : https://github.com/lwfinger/rtw89
Licenses        : GPL2
Groups          : None
Provides        : rtw89
Depends On      : dkms
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : rtw89
Replaces        : None
Installed Size  : 8.15 MiB
Packager        : Unknown Packager
Build Date      : Wed 03 Apr 2024 04:40:48 PM +03
Install Date    : Wed 03 Apr 2024 04:41:03 PM +03
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None

Name            : rtw89-dkms-git
Version         : 1:r980.48680ab-1
Description     : Driver for Realtek 8852AE, an 802.11ax device
Architecture    : x86_64
URL             : https://github.com/lwfinger/rtw89
Licenses        : GPL2
Groups          : None
Provides        : rtw89
Depends On      : dkms
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : rtw89
Replaces        : None
Installed Size  : 8.15 MiB
Packager        : Unknown Packager
Build Date      : Wed 03 Apr 2024 04:40:48 PM +03
Install Date    : Wed 03 Apr 2024 04:41:03 PM +03
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None

```
and reboot.

and make install:
```
10180  git clone https://github.com/lwfinger/rtw89.git
10182  cd rtw89
10183  make
10184  sudo make sign-install
10186  sudo pacman -S mokutil
10187  sudo make sign-install
```
and reboot

After all this steps, every time i have thats old firmware =)
```
sudo dmesg  | grep rtw                                                                                                                                                                          

[    3.278341] rtw89_8852ce 0000:03:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[    3.278520] rtw89_8852ce 0000:03:00.0: enabling device (0000 -> 0003)
[    3.281378] rtw89_8852ce 0000:03:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 1
[    3.281382] rtw89_8852ce 0000:03:00.0: Firmware version 0.27.56.13 (ebbe528f), cmd version 0, type 3

```

lwfinger, 2024-04-03:
Because most distros supply the latest firmware, and there is a git repo for those that do not, I do not supply the firmware. The README.md file shows you where to get the latest, which you will need to copy to /lib/firmware/rtw89/.

FunkyloverOne, 2024-05-08:
Hey @nudimannui4e, I still can't get it working. Could you please share how you got it to work?

nudimannui4e, 2024-07-04:
Привет, увы - решения так и не нашел, просто в какой-то момент прилетела обнова очередная системы, и просадки стали уже гораздо меньше, но никуда не исчезли =(
[image]
[image]