lwfinger/rtw89 #326: rtw8852be: signature and/or required key missing - tainting kernel

TroyeJames9, 2024-03-30:
OS: ubuntu22.04.4 LTS
Kernel version: 5.15.0-101-generic
motherboard：[ASUS B760M AYW WIFI D4](https://www.asus.com/motherboards-components/motherboards/others/b760m-ayw-wifi-d4/)

- WLAN has been enabled in BIOS settings

- I use the `lsmod | grep rtw`, the output as follow:

```
rtw_8852be             16384  0
rtw_8852b             409600  1 rtw_8852be
rtw89pci               81920  1 rtw_8852be
rtw89core             770048  2 rtw89pci,rtw_8852b
mac80211             1249280  2 rtw89pci,rtw89core
cfg80211              974848  3 rtw89core,mac80211,rtw_8852b
```

- the dmesg.txt is follow:

[dmesg_log.txt](https://github.com/lwfinger/rtw89/files/14809639/dmesg_log.txt)

The error about rtw8852 in dmesg is as follows:
[image]

- The results of  `mokutil --sb-state` are as follows. I have disabled secure boot:

```
SecureBoot disabled
Platform is in Setup Mode
```

I'm just one step away from success. Please help me. I'm sorry to bother you.

TroyeJames9, 2024-03-30:
我使用了以下安装教程来进行安装的
[image]

lwfinger, 2024-03-30:
There is a known problem with Ubuntu 22.04. In the rtw89 directory,

git checkout 6dc944
make
sudo make

That should work for you.

closed 2024-03-30 by TroyeJames9