lwfinger/rtw89 #327: Show the wifi but nothing to connect

inuyasha-660, 2024-03-31:
I finished the install and use `sudo apt install firmware-realtk` .
After  reboot , i found the WiFi icon appears. But when I want to select and connect, I find that the WiFi list is empty, It didn't scan any WiFi.
I hava  try to reinstall remodprobe  it didn't work still

here is some information, what else do I need to provide?

```
$ lspci | grep -i net
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller
```

```
$ sudo ifconfig
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 41  bytes 3826 (3.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 41  bytes 3826 (3.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlp1s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether ba:fd:05:c1:b7:3c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions
```

lwfinger, 2024-03-31:
Are you running Ubuntu 20.04? If so, that is a known problem that does not affect any other distro. In the rtw89 directory, do the following:

git checkout 6dc944
make
sudo make

inuyasha-660, 2024-03-31:
> Are you running Ubuntu 20.04? If so, that is a known problem that does not affect any other distro. In the rtw89 directory, do the following:
> 
> git checkout 6dc944
> make
> sudo make
> 
> 

Thanks, I use Debian , after i Switch to sid, and update the kernel, It works properly.

lwfinger, 2024-03-31:
Glad to hear.

closed 2024-03-31 by lwfinger