lwfinger/rtw89 #333: rtw89_8852ce: probe of 0001:03:00.0 failed with error -22

razvanphp, 2024-04-30:
Not sure if related with the other issues added lately, but latest master failed today on a system that was working fine with a version from a few weeks ago (can't know for sure since I formatted the drive). No hardware changes.

Ubuntu 20.04.6 LTS (GNU/Linux 5.10.120-tegra aarch64)

```
$ sudo dmesg |grep rtw89
[   12.524217] rtw89_8852ce 0001:03:00.0: Adding to iommu group 7
[   12.532820] rtw89_8852ce 0001:03:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[   12.540450] rtw89_8852ce 0001:03:00.0: enabling device (0000 -> 0003)
[   12.548803] rtw89_8852ce 0001:03:00.0: Firmware version 0.27.56.14 (1942d927), cmd version 0, type 1
[   12.558231] rtw89_8852ce 0001:03:00.0: Firmware version 0.27.56.14 (1942d927), cmd version 0, type 3
[   12.833760] rtw89_8852ce 0001:03:00.0: chip rfe_type is 1
[   12.876162] Modules linked in: ip6table_filter ip6_tables xt_conntrack iptable_filter xt_MASQUERADE xt_nat xt_addrtype iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c rtw_8852ce(OE+) rtw_8852c(OE) rtw89pci(OE) rtw89core(OE) mac80211 aes_ce_blk crypto_simd cryptd aes_ce_cipher ghash_ce input_leds sha2_ce btusb sha256_arm64 sha1_ce btrtl nvgpu btbcm cfg80211 btintel pwm_fan nct1008 tegra_bpmp_thermal ina3221 r8168 nv_imx464 max96793 max96792 spi_tegra114 nvidia(OE) binfmt_misc nvmap ip_tables x_tables
[   12.876339]  rtw89_core_register+0x1f0/0x5e8 [rtw89core]
[   12.876344]  rtw89_pci_probe+0x6f8/0xa88 [rtw89pci]
[   12.876377]  rtw89_8852ce_driver_init+0x30/0x1000 [rtw_8852ce]
[   12.884231] rtw89_8852ce 0001:03:00.0: failed to register hw
[   12.892590] rtw89_8852ce 0001:03:00.0: failed to register core hw
[   12.898877] rtw89_8852ce 0001:03:00.0: failed to register core
[   12.944661] rtw89_8852ce: probe of 0001:03:00.0 failed with error -22
```

Way to reproduce:
```
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install
cd /lib/firmware/rtw89/
sudo sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw89/rtw8852c_fw.bin
sudo sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw89/rtw8852b_fw.bin
sudo sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw89/rtw8852b_fw-1.bin
```

Any hint? Thank you!

lwfinger, 2024-04-30:
There is a problem with Ubuntu 20.04 that I have not yet found a fix. As a workaround, in the rtw89 folder use terminal input 'git checkout 6dc944', then 'make' and 'sudo make install'. That will get you a version that works.

razvanphp, 2024-05-01:
Ok, thanks for the heads up, hope my findings help you narrow down the problem:

1. tried 6dc9441 and I got this compile error:
```
/home/jetson/rtw89/debug.c: In function ‘rtw89_debug_priv_txpwr_table_get’:
/home/jetson/rtw89/debug.c:846:2: error: implicit declaration of function ‘rtw89_print_tas’ [-Werror=implicit-function-declaration]
  846 |  rtw89_print_tas(m, rtwdev);
      |  ^~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /home/jetson/rtw89/debug.o] Error 1
make[1]: *** [Makefile:1854: /home/jetson/rtw89] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.120-tegra-ubuntu20.04_aarch64/kernel-5.10'
make: *** [Makefile:93: all] Error 2
```
2. then tried e39dd3b and it compiled, but same error like in OP.
3. then tried ebe87ac and it worked:
```
$ sudo dmesg -T |grep rtw89
[Wed May  1 07:02:30 2024] rtw89_8852ce 0001:03:00.0: Adding to iommu group 7
[Wed May  1 07:02:30 2024] rtw89_8852ce 0001:03:00.0: loaded firmware rtw89/rtw8852c_fw.bin
[Wed May  1 07:02:30 2024] rtw89_8852ce 0001:03:00.0: enabling device (0000 -> 0003)
[Wed May  1 07:02:30 2024] rtw89_8852ce 0001:03:00.0: Firmware version 0.27.56.14 (1942d927), cmd version 0, type 1
[Wed May  1 07:02:30 2024] rtw89_8852ce 0001:03:00.0: Firmware version 0.27.56.14 (1942d927), cmd version 0, type 3
[Wed May  1 07:02:30 2024] rtw89_8852ce 0001:03:00.0: chip rfe_type is 1
```

Jackfritt, 2024-08-23:
I can confirm https://github.com/lwfinger/rtw89/commit/ebe87ac234a72a63a1b2ede911a874f1b7a6fe43 also works for Debian Bookworm  6.1.0-23-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux

bigbearishappy, 2024-11-08:
I can confirm https://github.com/lwfinger/rtw89/commit/ebe87ac234a72a63a1b2ede911a874f1b7a6fe43 also works fine for JP5 on nvidia jetson xavier nx.
And the iperf test detail is here:
```
seeed@seeed-desktop:~/rtw89$ iperf3 -c 192.168.115.5 -p 1000
Connecting to host 192.168.115.5, port 1000
[  5] local 192.168.115.169 port 55220 connected to 192.168.115.5 port 1000
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  13.2 MBytes   111 Mbits/sec    0    413 KBytes
[  5]   1.00-2.00   sec  17.8 MBytes   149 Mbits/sec    0    693 KBytes
[  5]   2.00-3.00   sec  21.2 MBytes   178 Mbits/sec    0    950 KBytes
[  5]   3.00-4.00   sec  27.5 MBytes   231 Mbits/sec    0   1.26 MBytes
[  5]   4.00-5.00   sec  28.8 MBytes   241 Mbits/sec    0   1.33 MBytes
[  5]   5.00-6.00   sec  30.0 MBytes   252 Mbits/sec    0   1.66 MBytes
[  5]   6.00-7.00   sec  31.2 MBytes   262 Mbits/sec    0   1.94 MBytes
[  5]   7.00-8.00   sec  23.8 MBytes   199 Mbits/sec    0   1.94 MBytes
[  5]   8.00-9.00   sec  28.8 MBytes   241 Mbits/sec    0   2.05 MBytes
[  5]   9.00-10.00  sec  30.0 MBytes   252 Mbits/sec    0   2.05 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   252 MBytes   212 Mbits/sec    0             sender
[  5]   0.00-10.02  sec   250 MBytes   209 Mbits/sec                  receiver

iperf Done.

```

mmanto, 2024-11-24:
I can confirm ebe87ac also works fine for Lenovo R7 16Gb Ram GeForce RTX 4060 RTL8852CE wireless In Debian testing 12

Dhina17, 2024-12-01:
I can confirm [ebe87ac](https://github.com/lwfinger/rtw89/commit/ebe87ac234a72a63a1b2ede911a874f1b7a6fe43) also works fine and fixed my Wifi issue in my Linux debian 6.1.0-28-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64 GNU/Linux

a5a5aa555oo, 2025-01-20:
@razvanphp Do you wanna try my driver? It works on Debian 11 (kernel version: 5.15.0-124-generic).

https://github.com/a5a5aa555oo/rtw89/tree/6.6-lts