lwfinger/rtw89 #337: Installation causes system function to hang due to ksoftirqd 100% CPU usage

JorDank88, 2024-05-17:
Specs/Info:

Kernel: 5.15.0 (specifically 5.15.0-102, but the error seems to occur across all 5.15 iterations)

`cat /etc/hosts
127.0.0.1	localhost
127.0.1.1	GlaDos

# The following lines are desirable for IPv6 capable hosts
::1     ip6-localhost ip6-loopback
fe00::0 ip6-localnet
ff00::0 ip6-mcastprefix
ff02::1 ip6-allnodes
ff02::2 ip6-allrouters
`

`ls /etc/apt/sources.list.d
brave-browser-release.list                  openmw-openmw-jammy.list
github-cli.list                             signal-xenial.list
google-chrome.list                          teams.list
maarten-fonville-android-studio-jammy.list  virtualbox.list
nordvpn.list                                winehq-jammy.sources
official-package-repositories.list
`
Steps to reproduce:

sudo apt update 
sudo apt upgrade
sudo apt install make gcc build-essential git
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make

These warnings occur during the 'make' command:
[image]

Additionally, a large amount of 'Skipping BTF generation ... due to unavailability of vmlinux' not sure if that's a problem but figured I could include it

`Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw89core.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw89pci.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw89pci.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw89pci.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw89pci.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8851b.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8851b.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8851b.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8851b.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8851be.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8851be.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8851be.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8851be.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8852a.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8852a.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8852a.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8852a.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8852ae.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8852ae.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8852ae.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8852b.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8852b.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8852b.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8852b.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8852be.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8852be.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8852be.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8852be.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8852c.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8852c.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8852c.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8852c.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8852ce.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8852ce.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8852ce.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8922a.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8922a.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8922a.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8922a.ko due to unavailability of vmlinux
  CC [M]  /home/jordank/Desktop/rtw89/rtw_8922ae.mod.o
  LD [M]  /home/jordank/Desktop/rtw89/rtw_8922ae.ko
  BTF [M] /home/jordank/Desktop/rtw89/rtw_8922ae.ko
Skipping BTF generation for /home/jordank/Desktop/rtw89/rtw_8922ae.ko due to unavailability of vmlinux`

sudo make install

Installation appears to finish successfully
`sudo make install
make -C /lib/modules/5.15.0-101-generic/build M=/home/jordank/Desktop/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-101-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-101-generic'
Install rtw89 SUCCESS
`
On restart, Kernel is broken due to ksoftirqd using 100% of CPU

Please let me know if any additional information can help address this issue, and I apologize in advance i'm not that good at this sort of thing.

lwfinger, 2024-05-17:
The compile warnings and the BTF warnings are harmless. They are not causing your problem.

What distro are you using?

What wifi device do you have? Has it ever worked on your system?

Run the command 'sudo dmesg > dmesg.txt' and attach dmesg.txt to this issue.

delbonis, 2024-06-04:
Also having this issue on another machine with an onboard realtek b852 chipset on a ASUS motherboard.  NetworkManager wouldn't come up and anything that asked the kernel about network things like even `sudo` and `ip a` would hang.  ksoftirqd was pegged at 100% CPU so it feels like it's hogging a lock in a busy loop or something.  It's kinda annoying to get diagnostic data.

Distro is Linux mint 21.2.  Will get dmesg when I have a chance to.

dakejahl, 2024-07-19:
I am seeing the same issue on a jetson orin nano with jetpack 6 -- **5.15.136-tegra-ubuntu22.04_aarch64**. The ksoftirqd pegging cpu at 100% and dmesg hangs.

Trying to use a Telit WE310K6-P
https://www.telit.com/devices/we310k6/
```
0001:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device b852
	Subsystem: Realtek Semiconductor Co., Ltd. Device b852
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 236
	IOMMU group: 3
	Region 0: I/O ports at 1000 [size=256]
	Region 2: Memory at 20a8000000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel driver in use: rtw89_8852be
	Kernel modules: rtw_8852be
```