lwfinger/rtw89 #338: During installation, "make" command not successful, prompts "Skipping BTF generation for /rtw89/rtw_8852be.ko due to unavailability of vmlinux"

nikwou, 2024-05-25:
Currently running Debian 12 / bookworm / Linux 6.1.0-21-amd64 x86_64 . I believe starting with Linux 6.1.0-20 - several weeks ago -, the above notices appeared when trying to install the driver. As a result, the driver is no longer installed, resulting in unavailability of the network interface. Previously, everything worked fine; I am not aware of any other changes I might have done. Any ideas what could be wrong are very welcome.

lwfinger, 2024-05-25:
Does make report an error? If not, does the make operation produce *.ko files? If it does, the only problem is exactly what it says - vmlinux is not available. You need to look at the output os 'sudo dmesg | grep rtw' to see what it says. Otherwise, upgrade to sid with kernel 6.8.9 and use the in-kernel driver.

nikwou, 2024-05-26:
Thank you very much for the quick turnaround. I have posted the outputs of the respective steps (make, make install, dmesg). Looking at the directory after make and make install steps, I cannot see any *.ko files. /usr/lib/firmware/rtw89 contains *.bin files, but they bear older date stamps (May 1st, 2023).

I have in deed tried to use higher kernel versions, but that brings other problems which I am unable to solve.

[dmesg_output.log](https://github.com/lwfinger/rtw89/files/15447810/dmesg_output.log)
[make_install_output.log](https://github.com/lwfinger/rtw89/files/15447811/make_install_output.log)
[make_output_1.log.txt](https://github.com/lwfinger/rtw89/files/15447812/make_output_1.log.txt)
[make_output_2.log](https://github.com/lwfinger/rtw89/files/15447813/make_output_2.log)

lwfinger, 2024-05-26:
Your firmware is old, but it should work.
[   19.525472] rtw89_8852be 0000:2b:00.0: Firmware version 0.27.32.1 (ef7920a3), cmd version 0, type 1.

make_output_2.log shows the .ko files being made.

make -C /lib/modules/6.1.0-21-amd64/build
 ...
 CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/pci_be.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89pci.o
  MODPOST /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/Module.symvers
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89core.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89core.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89core.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89pci.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89pci.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw89pci.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8851b.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8851b.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8851b.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8851be.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8851be.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8851be.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852a.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852a.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852a.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852ae.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852ae.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852ae.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852b.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852b.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852b.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852be.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852be.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852be.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852c.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852c.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852c.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852ce.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852ce.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8852ce.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8922a.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8922a.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8922a.ko
  CC [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8922ae.mod.o
  LD [M]  /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8922ae.ko
  BTF [M] /home/nwwa/Dokumente/Misc/Computer/PC/23_Lenovo/Patch_WIFI/rtw89/rtw_8922ae.ko

I do not see anything wrong.

AndreyRz, 2024-07-11:
Is there any solution for systems with a core lower than 6.8.9? When making, Skipping BTF generation for is issued... for all files .ko and .oh, and the wi-fi module is not visible in the system, the bluetooth module is only running, I have no way to upgrade the kernel version. I try dkms version, but it doesn't work either.