lwfinger/rtw89 #340: Ubuntu 22.04 / rtw_8852be stuck authenticating

kfunigeneral, 2024-05-29:
## System:
* **Hardware**: Lenovo IdeaPad 5
* **Network Controller**: Realtek SemiConductor Co., Ltd Device b852
* **uname -r**: `6.5.0-35-generic`
* **OS**: `Ubuntu 22.04.2 LTS`
* **RTW89 Version**: Latest, pulled today

## Issue
The network discovery succeeds, but the connection is always hanging on connecting.

```
[    2.480886] rtw89core: loading out-of-tree module taints kernel.
[    2.480897] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    4.098530] rtw89_8852be 0000:01:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[    4.098714] rtw89_8852be 0000:01:00.0: enabling device (0000 -> 0003)
[    4.101476] rtw89_8852be 0000:01:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[    4.101484] rtw89_8852be 0000:01:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[    4.402503] rtw89_8852be 0000:01:00.0: chip rfe_type is 1
[    4.472565] rtw89_8852be 0000:01:00.0 wlp1s0: renamed from wlan0
[    8.368745] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[    8.981108] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   12.821808] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   13.117325] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   17.377408] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   17.688970] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   22.428901] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   22.717115] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   32.483146] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   32.789419] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   37.642607] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   37.937095] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   55.672943] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   55.969018] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   69.713973] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   70.009518] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   79.643603] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   79.933485] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   93.681525] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[   93.985081] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  107.746867] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  108.037211] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  121.782110] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  122.072734] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  240.113480] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  240.408808] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  254.153664] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  254.453171] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  436.665583] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  436.953138] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  450.693775] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  450.981025] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  461.640061] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  461.928981] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  475.681822] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  475.973167] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  489.727582] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  490.021063] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  503.765543] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  504.065187] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  517.823685] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  518.121407] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  531.865719] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  532.153498] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  639.273647] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  639.565109] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  653.305962] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  653.601144] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  844.603784] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  844.889290] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  858.637843] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  858.937122] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  872.696992] wlp1s0: authenticate with a4:98:13:bb:4b:9f
[  872.997265] wlp1s0: authenticate with a4:98:13:bb:4b:9f
```

I believe I previously used another version of RTW89, sadly I forgot which commit that was on :(, I tried updating to the latest because of frequent disconnect issues, and now the connection can't be made at all.

aGuyOverThere, 2024-06-03:
Since Pop_OS is 22.04, I am also having this issue.

nicball, 2024-06-09:
I have the same problem on kernel 6.6. However it works fine on 6.9.

dhanak, 2024-06-23:
I just ran into the same issue with Debian Bookworm (12.5), kernel `6.1.0-21-amd64`, with an RTL 8852BE device (`Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller`). I had to download the [rtw8852b_fw-1.bin](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89/rtw8852b_fw-1.bin) firmware manually. Before that, the device didn't show any SSIDs. Now it lists all the visible networks, but there are still no bars (all the networks show up with signal strength 0), and it can't connect, either.