lwfinger/rtw89 #344: Debian 12 kernel 6.12.0.21 rtw89 compiles ok, but..

kokup, 2024-06-10:
4.825430] r8169 0000:02:00.0: firmware: direct-loading firmware rtl_nic/rtl8125b-2.fw
[    4.852248] RTL8226B_RTL8221B 2.5Gbps PHY r8169-0-200:00: attached PHY driver (mii_bus:phy_addr=r8169-0-200:00, irq=MAC)
[    5.056293] r8169 0000:02:00.0 enp2s0: Link is Down
[    5.175962] halbb_dig_gain_para_init: skipped because ib_pbk not defined
[    5.175970] halbb_sdagc_follow_pagc_config: skipped because path0_p20_follow_by_pagcugc_en not defined

it's all G(r)eek to me; hope someone can gimme a fix

brianwitte, 2024-06-10:
Hello, that looks like it belongs to your Ethernet interface and not your wireless. 

The attached PHY driver and `enp2s0` interface are not for wireless.

 ```
 [ 4.852248] RTL8226B_RTL8221B 2.5Gbps PHY r8169-0-200:00: attached PHY driver (mii_bus:phy_addr=r8169-0-200:00, irq=MAC)
[ 5.056293] r8169 0000:02:00.0 enp2s0: Link is Down
```

closed 2024-06-10 by brianwitte

kokup, 2024-06-10:
thank you Brian; my eyes are dim; i had confused 
RTL8226B_RTL8221B 
with 8852BE
meantime, although rtw89 compiled ok, 8852BE doesn't show up as a wifi interface, so i'm temporising with a Ralink dongle.
here is (hopefully!) relevant dmesg:
[    3.944363] Bluetooth: hci0: RTL: examining hci_ver=0b hci_rev=000b lmp_ver=0b lmp_subver=8852
[    3.946357] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_fw.bin
[    3.946732] bluetooth hci0: firmware: direct-loading firmware rtl_bt/rtl8852bu_fw.bin
[    3.946744] Bluetooth: hci0: RTL: loading rtl_bt/rtl8852bu_config.bin
[    3.946836] bluetooth hci0: firmware: direct-loading firmware rtl_bt/rtl8852bu_config.bin
[    3.967369] 8852be: loading out-of-tree module taints kernel.
[    3.968524] 8852be: module verification failed: signature and/or required key missing - tainting kernel
[    4.183883] rtw8852be driver loaded
[    4.183896] rtl8852be 0000:03:00.0: enabling device (0000 -> 0003)
[    4.628256] rtl8852be 0000:03:00.0 wlp3s0: renamed from wlan0

STOP PRESS: great Scot!  After another restart, it **does** appear and does connect!! :)  Mine not to reason why...
It would be nice to know what "signature and/or required key missing - tainting kernel" is all about, but so long as it works, i no complain :)  bluetooth works too, so thankyou thankyou to lwfinger and friends