lwfinger/rtw89 #345: rtw89_8852ae: failed to allocate hw | probe failed with error -12

stormbringer83, 2024-06-12:
Hi,

rtw_8852ae doesn't detect the wifi card.

```
user@pc:~$ uname -r
6.8.11-300.fc40.x86_64

user@pc:~$ lspci -v
04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852AE 802.11ax PCIe Wireless Network Adapter
	DeviceName: RTL8111E Giga LAN
	Subsystem: Lenovo Device 4852
	Flags: fast devsel, IRQ 255, IOMMU group 15
	I/O ports at f000 [disabled] [size=256]
	Memory at fc500000 (64-bit, non-prefetchable) [disabled] [size=1M]
	Capabilities: <access denied>
	Kernel modules: rtw89_8852ae, rtw_8852ae

user@pc:~$ lsmod | grep rtw
rtw_8852ae             12288  0
rtw_8852a             729088  1 rtw_8852ae
rtw89pci              118784  1 rtw_8852ae
rtw89core            1097728  2 rtw89pci,rtw_8852a
mac80211             1724416  2 rtw89pci,rtw89core
cfg80211             1409024  3 rtw_8852a,rtw89core,mac80211

user@pc:~$ sudo dmesg | grep rtw
[sudo] password for user: 
[    5.929817] rtw89core: loading out-of-tree module taints kernel.
[    6.035403] rtw89_8852ae 0000:04:00.0: loaded firmware rtw89/rtw8852a_fw.bin
[    6.035447] Modules linked in: rtw_8852ae(O+) rtw_8852a(O) rtw89pci(O) rtw89core(O) intel_rapl_msr intel_rapl_common snd_hda_codec_hdmi edac_mce_amd kvm_amd snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec kvm snd_hda_core mac80211 snd_hwdep snd_seq btusb snd_seq_device irqbypass snd_pcm btrtl btintel libarc4 btbcm btmtk rapl cfg80211 snd_timer bluetooth think_lmi snd firmware_attributes_class wmi_bmof acpi_cpufreq pcspkr video soundcore k10temp rfkill i2c_piix4 hid_logitech_hidpp(+) gpio_amdpt gpio_generic loop nfnetlink zram crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni polyval_generic nvme ghash_clmulni_intel sha512_ssse3 sha256_ssse3 ccp sha1_ssse3 nvme_core sp5100_tco nvme_auth wmi hid_logitech_dj ip6_tables ip_tables fuse
[    6.035610]  rtw89_alloc_ieee80211_hw+0x94/0x160 [rtw89core]
[    6.035635]  rtw89_pci_probe+0x47/0xe70 [rtw89pci]
[    6.035676]  ? __pfx_rtw89_8852ae_driver_init+0x10/0x10 [rtw_8852ae]
[    6.035743] rtw89_8852ae 0000:04:00.0: failed to allocate hw
[    6.035745] rtw89_8852ae: probe of 0000:04:00.0 failed with error -12
```

stormbringer83, 2024-06-13:
Error stack:
```
[    6.085804] rtw89_8852ae 0000:04:00.0: loaded firmware rtw89/rtw8852a_fw.bin
[    6.085815] ------------[ cut here ]------------
[    6.085816] WARNING: CPU: 10 PID: 754 at net/mac80211/main.c:665 ieee80211_alloc_hw_nm+0x6a2/0x780 [mac80211]
[    6.085849] Modules linked in: rtw_8852ae(O+) rtw_8852a(O) rtw89pci(O) rtw89core(O) joydev snd_hda_codec_hdmi mac80211 snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec intel_rapl_msr intel_rapl_common edac_mce_amd snd_hda_core cdc_mbim cdc_wdm btusb btrtl kvm_amd snd_hwdep snd_seq btintel libarc4 btbcm snd_seq_device btmtk cdc_ncm cdc_ether kvm usbnet irqbypass cfg80211 snd_pcm mii bluetooth rapl snd_timer snd think_lmi firmware_attributes_class wmi_bmof k10temp pcspkr acpi_cpufreq rfkill video i2c_piix4 soundcore gpio_amdpt hid_logitech_hidpp gpio_generic loop nfnetlink zram crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni polyval_generic nvme ghash_clmulni_intel sha512_ssse3 sha256_ssse3 nvme_core ccp sha1_ssse3 sp5100_tco nvme_auth wmi hid_logitech_dj ip6_tables ip_tables fuse
[    6.085898] CPU: 10 PID: 754 Comm: (udev-worker) Tainted: G           O       6.8.11-300.fc40.x86_64 #1
[    6.085900] Hardware name: LENOVO 90RC01JVGE/3716, BIOS O4MKT2EA 03/01/2023
[    6.085901] RIP: 0010:ieee80211_alloc_hw_nm+0x6a2/0x780 [mac80211]
[    6.085924] Code: 54 00 01 00 00 e9 34 fb ff ff 89 c8 81 ca 63 83 01 08 83 c8 64 89 55 5c 88 45 63 e9 10 fb ff ff 48 89 ef e8 f0 66 bb ff eb a0 <0f> 0b eb 9c 4c 89 ee 48 c7 c7 20 f3 cd c0 e8 cb c5 c0 f2 e9 cb fe
[    6.085925] RSP: 0018:ffffb68381f57678 EFLAGS: 00010216
[    6.085927] RAX: 0000000000000021 RBX: ffff99304e184400 RCX: ffff99304e184400
[    6.085929] RDX: 0000000000000000 RSI: ffff99304e184400 RDI: 0000000000074308
[    6.085930] RBP: ffffffffc0f62240 R08: 0000000000000000 R09: ffff99304e184400
[    6.085931] R10: ffffb68381f576a0 R11: 0000000000000000 R12: ffff9930423220c0
[    6.085932] R13: 0000000000000003 R14: 00000000000699e0 R15: ffff99304138e4e8
[    6.085933] FS:  00007f5b12aac980(0000) GS:ffff99375e700000(0000) knlGS:0000000000000000
[    6.085935] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.085936] CR2: 00007f5b10c38000 CR3: 00000001047ce000 CR4: 0000000000f50ef0
[    6.085937] PKRU: 55555554
[    6.085938] Call Trace:
[    6.085940]  <TASK>
[    6.085941]  ? ieee80211_alloc_hw_nm+0x6a2/0x780 [mac80211]
[    6.085962]  ? __warn+0x81/0x130
[    6.085966]  ? ieee80211_alloc_hw_nm+0x6a2/0x780 [mac80211]
[    6.085987]  ? report_bug+0x16f/0x1a0
[    6.085990]  ? handle_bug+0x3c/0x80
[    6.085992]  ? exc_invalid_op+0x17/0x70
[    6.085994]  ? asm_exc_invalid_op+0x1a/0x20
[    6.085999]  ? ieee80211_alloc_hw_nm+0x6a2/0x780 [mac80211]
[    6.086020]  rtw89_alloc_ieee80211_hw+0x94/0x160 [rtw89core]
[    6.086045]  rtw89_pci_probe+0x47/0xe70 [rtw89pci]
[    6.086051]  ? create_pinctrl+0x61/0x430
[    6.086054]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086055]  ? down_write+0x12/0x60
[    6.086058]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086061]  local_pci_probe+0x45/0xa0
[    6.086063]  pci_device_probe+0xc1/0x2a0
[    6.086066]  really_probe+0x19e/0x3e0
[    6.086069]  ? __pfx___driver_attach+0x10/0x10
[    6.086071]  __driver_probe_device+0x78/0x160
[    6.086073]  driver_probe_device+0x1f/0xa0
[    6.086075]  __driver_attach+0xba/0x1c0
[    6.086077]  bus_for_each_dev+0x8f/0xe0
[    6.086080]  bus_add_driver+0x146/0x250
[    6.086083]  driver_register+0x5c/0x100
[    6.086085]  ? __pfx_rtw89_8852ae_driver_init+0x10/0x10 [rtw_8852ae]
[    6.086089]  do_one_initcall+0x5b/0x320
[    6.086093]  do_init_module+0x90/0x270
[    6.086096]  __do_sys_init_module+0x17a/0x1b0
[    6.086100]  do_syscall_64+0x83/0x170
[    6.086103]  ? touch_atime+0x1e/0x120
[    6.086105]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086107]  ? filemap_read+0x372/0x3e0
[    6.086110]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086114]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086116]  ? vfs_read+0x24f/0x380
[    6.086119]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086121]  ? syscall_exit_to_user_mode_prepare+0x178/0x1a0
[    6.086122]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086124]  ? syscall_exit_to_user_mode+0x83/0x230
[    6.086126]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086127]  ? do_syscall_64+0x8f/0x170
[    6.086129]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086131]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086132]  ? update_process_times+0x9c/0xb0
[    6.086135]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086136]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086138]  ? timerqueue_add+0x98/0xb0
[    6.086140]  ? srso_alias_return_thunk+0x5/0xfbef5
[... 5 lines cut ...]
[    6.086150]  ? ktime_get+0x47/0xb0
[    6.086152]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086154]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086155]  ? clockevents_program_event+0xa2/0x110
[    6.086158]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086159]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086160]  ? sched_clock_cpu+0x11/0x1f0
[    6.086163]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086164]  ? irqtime_account_irq+0x40/0xc0
[    6.086167]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086169]  ? srso_alias_return_thunk+0x5/0xfbef5
[    6.086171]  entry_SYSCALL_64_after_hwframe+0x78/0x80
[    6.086173] RIP: 0033:0x7f5b12d2c57e
[... cut ...]
[    6.086186] RDX: 00007f5b132c007d RSI: 00000000000ba008 RDI: 000055e1f96e7c00
[    6.086187] RBP: 00007ffc2a1b9c00 R08: 000055e1f797b010 R09: 0000000000000007
[    6.086188] R10: 0000000000000003 R11: 0000000000000246 R12: 00007f5b132c007d
[    6.086189] R13: 0000000000020000 R14: 000055e1f7b2d3f0 R15: 000055e1f81afa00
[    6.086192]  </TASK>
[    6.086193] ---[ end trace 0000000000000000 ]---
[    6.086215] rtw89_8852ae 0000:04:00.0: failed to allocate hw
[    6.086217] rtw89_8852ae: probe of 0000:04:00.0 failed with error -12
```

PixelatedVolume, 2024-06-13:
I’m having (maybe) the exact same problem.  Can I see the output of your `modprobe | grep rtw`?  I suspect in my case it’s being caused by improperly blocking one of the “core” modules, so I’m curious about what you have loaded.

PixelatedVolume, 2024-06-13:
My theory is that the problem has to do with an ambiguous definition for the function `rtw89_alloc_ieee80211_hw`.  The “failed to allocate hw” line is produced by a failed call to it.

In your lsmod results, you have `rtw89_8852ae` and also `rtw_8852a` (as do I).  Seems to me that this should be `rtw89_8852a`, but when I try to load it manually, I get a dmesg saying rtw89_core provides `rtw89_alloc_ieee80211_hw`, which is already provided by rtw89core.  I’ve tried blacklisting  rtw89core and then rtw89_core to stop this conflict, but it hasn’t had any effect.

brianwitte, 2024-06-14:
Hey, will take a look in the morning. I see you are running a new kernel. 

If you are running these drivers instead of the in-tree drivers, make sure you blacklist the in-tree modules. If you look at the Makefile in this repo and compare to the one's for rtw89 in the kernel source, you can see which to blacklist/keep.

I am on mobile and won't be back on computer until tmrw.

brianwitte, 2024-06-15:
Hi, any updates? @stormbringer83