lwfinger/rtw89 #346: kernel: BUG: kernel NULL pointe  kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)

azio7, 2024-06-16:
# system

archlinux 

Kernel: 6.6.33-1-lts

# issue

this happened after i update all my stuff by pacman, then i try old version of this driver and it works

```
6月 16 19:16:17 azio7 kernel: BUG: kernel NULL pointer dereference, address: 000000000000008c
6月 16 19:16:17 azio7 kernel: #PF: supervisor read access in kernel mode
6月 16 19:16:17 azio7 kernel: #PF: error_code(0x0000) - not-present page
6月 16 19:16:17 azio7 kernel: PGD 0 P4D 0 
6月 16 19:16:17 azio7 kernel: Oops: 0000 [#1] PREEMPT SMP NOPTI
6月 16 19:16:17 azio7 kernel: CPU: 9 PID: 487 Comm: (udev-worker) Tainted: P           OE      6.6.33-1-lts #1 099c530d40d937b727535a44e49902806a26db4a
6月 16 19:16:17 azio7 kernel: Hardware name: TIMI A7S/TM2037, BIOS RMGCZ6T0P1010 07/11/2022
6月 16 19:16:17 azio7 kernel: RIP: 0010:rtw89_pci_probe+0x77/0xbc0 [rtw89pci]
6月 16 19:16:17 azio7 kernel: Code: 0a 00 00 49 89 c5 48 8b 43 10 49 c7 85 18 05 00 00 a0 3e 27 c6 4c 89 ef 49 89 85 b8 00 00 00 41 c7 85 20 05 00 00 00 00 00 00 <8b> 90 8c 00 00 00 41 89 95 24 05 00 >
6月 16 19:16:17 azio7 kernel: RSP: 0018:ffffa2fdc16e3a60 EFLAGS: 00010286
6月 16 19:16:17 azio7 kernel: RAX: 0000000000000000 RBX: ffffffffc63ca080 RCX: ffffa2fdc16e38a8
6月 16 19:16:17 azio7 kernel: RDX: ffffa2fdc16e3868 RSI: 0000000000000000 RDI: ffff924621602060
6月 16 19:16:17 azio7 kernel: RBP: ffff924601adf000 R08: 0000000000000000 R09: ffff92471cf130ff
6月 16 19:16:17 azio7 kernel: R10: ffffffffc601aab6 R11: 0000000000000000 R12: ffffffffc63c8020
6月 16 19:16:17 azio7 kernel: R13: ffff924621602060 R14: ffff924601adf000 R15: ffff92460b2513d0
6月 16 19:16:17 azio7 kernel: FS:  000079d799a0d680(0000) GS:ffff9248fe840000(0000) knlGS:0000000000000000
6月 16 19:16:17 azio7 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
6月 16 19:16:17 azio7 kernel: CR2: 000000000000008c CR3: 0000000102e86000 CR4: 0000000000f50ee0
6月 16 19:16:17 azio7 kernel: PKRU: 55555554
6月 16 19:16:17 azio7 kernel: Call Trace:
6月 16 19:16:17 azio7 kernel:  <TASK>
6月 16 19:16:17 azio7 kernel:  ? __die+0x23/0x70
6月 16 19:16:17 azio7 kernel:  ? page_fault_oops+0x174/0x530
6月 16 19:16:17 azio7 kernel:  ? exc_page_fault+0x7f/0x180
6月 16 19:16:17 azio7 kernel:  ? asm_exc_page_fault+0x26/0x30
6月 16 19:16:17 azio7 kernel:  ? rtw89_pci_probe+0x77/0xbc0 [rtw89pci 36aad5b81bed9ac8ab5bb35fad28a3d171a5986b]
6月 16 19:16:17 azio7 kernel:  local_pci_probe+0x45/0xa0
6月 16 19:16:17 azio7 kernel:  pci_device_probe+0xbd/0x2a0
6月 16 19:16:17 azio7 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
6月 16 19:16:17 azio7 kernel:  ? sysfs_do_create_link_sd+0x6e/0xe0
6月 16 19:16:17 azio7 kernel:  really_probe+0x19e/0x3e0
6月 16 19:16:17 azio7 kernel:  ? __pfx___driver_attach+0x10/0x10
6月 16 19:16:17 azio7 kernel:  __driver_probe_device+0x78/0x160
6月 16 19:16:17 azio7 kernel:  driver_probe_device+0x1f/0xa0
6月 16 19:16:17 azio7 kernel:  __driver_attach+0xba/0x1c0
6月 16 19:16:17 azio7 kernel:  bus_for_each_dev+0x8f/0xe0
6月 16 19:16:17 azio7 kernel:  bus_add_driver+0x112/0x240
6月 16 19:16:17 azio7 kernel:  driver_register+0x5c/0x100
6月 16 19:16:17 azio7 kernel:  ? __pfx_rtw89_8852ae_driver_init+0x10/0x10 [rtw89_8852ae a391fb9a07b927c857717e4a1993b4ade0f509ef]
6月 16 19:16:17 azio7 kernel:  do_one_initcall+0x5d/0x320
6月 16 19:16:17 azio7 kernel:  do_init_module+0x60/0x240
6月 16 19:16:17 azio7 kernel:  init_module_from_file+0x89/0xe0
6月 16 19:16:17 azio7 kernel:  idempotent_init_module+0x121/0x2b0
6月 16 19:16:17 azio7 kernel:  __x64_sys_finit_module+0x5e/0xb0
6月 16 19:16:17 azio7 kernel:  do_syscall_64+0x5a/0x80
6月 16 19:16:17 azio7 kernel:  ? do_syscall_64+0x66/0x80
6月 16 19:16:17 azio7 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
6月 16 19:16:17 azio7 kernel:  ? syscall_exit_to_user_mode+0x22/0x40
6月 16 19:16:17 azio7 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
6月 16 19:16:17 azio7 kernel:  ? do_syscall_64+0x66/0x80
6月 16 19:16:17 azio7 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
6月 16 19:16:17 azio7 kernel:  ? __flush_smp_call_function_queue+0x9e/0x440
6月 16 19:16:17 azio7 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
6月 16 19:16:17 azio7 kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
6月 16 19:16:17 azio7 kernel:  ? __irq_exit_rcu+0x4b/0xc0
6月 16 19:16:17 azio7 kernel:  entry_SYSCALL_64_after_hwframe+0x78/0xe2
6月 16 19:16:17 azio7 kernel: RIP: 0033:0x79d799527e9d
6月 16 19:16:17 azio7 kernel: Code: ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 63 de 0c 00 f7 d8 64 89 01 48
6月 16 19:16:17 azio7 kernel: RSP: 002b:00007ffc09719da8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
6月 16 19:16:17 azio7 kernel: RAX: ffffffffffffffda RBX: 000056dd664d1d00 RCX: 000079d799527e9d
6月 16 19:16:17 azio7 kernel: RDX: 0000000000000004 RSI: 000079d799b4c376 RDI: 000000000000002e
6月 16 19:16:17 azio7 kernel: RBP: 000079d799b4c376 R08: 000079d7995f6b20 R09: 00007ffc09719df0
6月 16 19:16:17 azio7 kernel: R10: 000056dd665e4f10 R11: 0000000000000246 R12: 0000000000020000
6月 16 19:16:17 azio7 kernel: R13: 000056dd665e1a40 R14: 0000000000000000 R15: 000056dd665e4ed0
6月 16 19:16:17 azio7 kernel:  </TASK>
6月 16 19:16:17 azio7 kernel: Modules linked in: kvm(+) rtw89_8852ae(+) snd_compress ac97_bus snd_hda_intel(+) snd_pcm_dmaengine rtw_8852a(OE) irqbypass snd_pci_ps btusb snd_intel_dspcfg rtw89pci(OE)>
6月 16 19:16:17 azio7 kernel:  crypto_user dm_mod fuse loop nfnetlink bpf_preload ip_tables x_tables usbhid serio_raw atkbd amdgpu libps2 vivaldi_fmap nvme drm_exec amdxcp nvme_core sha256_ssse3 drm_>
6月 16 19:16:17 azio7 kernel: ucsi_acpi USBC000:00: UCSI_GET_PDOS failed (-95)
6月 16 19:16:17 azio7 kernel: CR2: 000000000000008c
6月 16 19:16:17 azio7 kernel: ---[ end trace 0000000000000000 ]---
6月 16 19:16:17 azio7 kernel: RIP: 0010:rtw89_pci_probe+0x77/0xbc0 [rtw89pci]
6月 16 19:16:17 azio7 kernel: Code: 0a 00 00 49 89 c5 48 8b 43 10 49 c7 85 18 05 00 00 a0 3e 27 c6 4c 89 ef 49 89 85 b8 00 00 00 41 c7 85 20 05 00 00 00 00 00 00 <8b> 90 8c 00 00 00 41 89 95 24 05 00>
6月 16 19:16:17 azio7 kernel: RSP: 0018:ffffa2fdc16e3a60 EFLAGS: 00010286
6月 16 19:16:17 azio7 kernel: RAX: 0000000000000000 RBX: ffffffffc63ca080 RCX: ffffa2fdc16e38a8
6月 16 19:16:17 azio7 kernel: RDX: ffffa2fdc16e3868 RSI: 0000000000000000 RDI: ffff924621602060
6月 16 19:16:17 azio7 kernel: RBP: ffff924601adf000 R08: 0000000000000000 R09: ffff92471cf130ff
6月 16 19:16:17 azio7 kernel: R10: ffffffffc601aab6 R11: 0000000000000000 R12: ffffffffc63c8020
6月 16 19:16:17 azio7 kernel: R13: ffff924621602060 R14: ffff924601adf000 R15: ffff92460b2513d0
6月 16 19:16:17 azio7 kernel: FS:  000079d799a0d680(0000) GS:ffff9248fe840000(0000) knlGS:0000000000000000
6月 16 19:16:17 azio7 kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
6月 16 19:16:17 azio7 kernel: CR2: 000000000000008c CR3: 0000000102e86000 CR4: 0000000000f50ee0
6月 16 19:16:17 azio7 kernel: PKRU: 55555554
6月 16 19:16:17 azio7 kernel: note: (udev-worker)[487] exited with irqs disabled
```

pkshih, 2024-06-25:
Between 23547bb and f0d6e99, only these five commits:

```
f0d6e99a707f wifi: rtw89: pci: correct TX resource checking for PCI DMA channel of firmware command
1a55db023888 rtw89: Fix build errors on older kernels
fba4f52b56f4 wifi: rtw89: 8922a: configure UL MU/OFDMA power setting
08f4ca25ec25 rtw89: Bring this repo up to the level of rtw-next as of April 3, 2024.
23547bbbd06c wifi: rtw89: 8852c: disable PCI PHY EQ to improve compatibility
```

I feel these should not affect pci_probe.
Could you find out the commit causing your problem?

azio7, 2024-06-25:
> Between [23547bb](https://github.com/lwfinger/rtw89/commit/23547bbbd06c1d8eac792e94875b357101d4d82e) and [f0d6e99](https://github.com/lwfinger/rtw89/commit/f0d6e99a707f53aa4fa71e5c3c4141b3cb44fb67), [...]

i try addr2line
```
azio7@azio7 /d/p/r/rtw89 ((f0d6e99a))> objdump -t rtw89pci.ko | grep rtw89_pci_probe
0000000000000259 l       __ksymtab_strings      0000000000000000 __kstrtab_rtw89_pci_probe
0000000000000269 l       __ksymtab_strings      0000000000000000 __kstrtabns_rtw89_pci_probe
00000000000000fc l       __ksymtab      0000000000000000 __ksymtab_rtw89_pci_probe
0000000000003840 g     F .text  0000000000000010 __pfx_rtw89_pci_probe
0000000000003850 g     F .text  0000000000000bb5 rtw89_pci_probe
azio7@azio7 /d/p/r/rtw89 ((f0d6e99a))> addr2line -Cfie rtw89pci.ko 0x38c7
rtw89_pci_probe
/home/azio7/project/rtw89-dkms-git/rtw89/pci.c:4261
```
the line is
```
	rtwdev->hci.rpwm_addr = pci_info->rpwm_addr;
```

pkshih, 2024-06-26:
Backward trace code from this point: 

```
rtwdev->hci.rpwm_addr = pci_info->rpwm_addr;
```

Then,

```
static const struct rtw89_driver_info rtw89_8852ae_info = {
	.chip = &rtw8852a_chip_info,
	.quirks = NULL,
	.bus = {
		.pci = &rtw8852a_pci_info,
	},
};

static const struct pci_device_id rtw89_8852ae_id_table[] = {
	{
		PCI_DEVICE(PCI_VENDOR_ID_REALTEK, 0x8852),
		.driver_data = (kernel_ulong_t)&rtw89_8852ae_info,
	},

info = (const struct rtw89_driver_info *)id->driver_data;

pci_info = info->bus.pci;
```

Trace looks well. 

I build [f0d6e99](https://github.com/lwfinger/rtw89/commit/f0d6e99a707f53aa4fa71e5c3c4141b3cb44fb67) and run on kernel 6.6.8 and 6.9.2, both are fine. 

No good idea how it fails on your platform. Maybe add more debug message yourself to dig the cause. 

By the way, on kernel 6.6.33 you can use built-in driver for RTL8852AE, which has been not changed for a long time.

azio7, 2024-06-26:
I use this driver because the built-in driver may disconnect sometime, so i turn to this driver.  Im noob on kernel, i cannot dig more message.
i guess the problem comes from the kernel, i didnt meet the bug until i update

pkshih, 2024-06-27:
I suggest you can use built-in driver, because I feel driver in this repository has the same problem even if it can run on your platform.

Please follow the main page of this repository to add option below to 70-rtw89.conf: 
```
options rtw89_core disable_ps_mode=y
```

Then, copy it to proper place
```
sudo cp 70-rtw89.conf /etc/modprobe.d/
```

Edit /etc/NetworkManager/conf.d/default-wifi-powersave-on.conf as below. 
(If file is not existing, just create it.)
```
[connection]
wifi.powersave = 2
```

azio7, 2024-06-27:
i see, thanks for your suggestion

azio7, 2025-04-13:
today, i update this driver and the error gone

closed 2025-04-13 by azio7