lwfinger/rtw89 #348: BT coexistence not working on 2,4 Ghz Wifi

Feuerhamster, 2024-06-29:
Wireless card: RTL8852BE
Tried on Debian 12 (Testing), Pop OS and Fedora Workstation 40

I have a very strange issue with coexisting Bluetooth audio and Wi-Fi connection.
After a few seconds, the audio on the Bluetooth headphones stutters.
After about 2 minutes, there is no internet connection anymore. 
It says Wi-Fi is still connected, but no traffic gets through.

This issue only appears if I am on a 2,4 Ghz Wi-Fi Network.
Some people in forums suggests to just use 5 Ghz Wi-Fi but except for my home, I cannot control what frequencies are provided in, for example, public networks.

It's really annoying because the basic use case of watching YouTube on the train is completely broken.

I tried to install the driver manually from this repo, but that didn't help and since it is now included in the kernel, I guess it wouldn't change anything. 

Is there something I can do to get BT coex working on 2,4 Ghz Wi-Fi networks?

brianwitte, 2024-07-03:
hello, @Feuerhamster. this is actually a backport repo you would want to use this only if you have older kernel version that is not supporting your needs for your wireless driver. 

I would debian testing's in-tree driver for rtw89 would support 8852BE properly.

Here are some examples for giving me good debug output to see what you are working with.

You can just adjust commands below to for `rtw89` specific modules/names/files:

https://wireless.wiki.kernel.org/en/users/documentation/reporting_bugs

https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi/debugging

Feuerhamster, 2024-07-03:
I know that the driver is included in more recent kernels. On the distros I've tried I used a clean install with the included driver and did not installed this repo on top. But I didn't know where else to ask for help for this issue. 

I currently swapped my wifi card so for everyday use, but I will post the debug logs here as soon as I swapped back for debugging.

AqilbekAbilaev, 2024-10-07:
@Feuerhamster Hello!
I have the same Realtek card. I am facing the symptoms you described above. It is so disappointing to turn off bluetooth in order to use wifi(I cannot use bluetooth earphones). Have you found the solution ? If you have, please let us know.
Thanks!

Feuerhamster, 2024-10-12:
@AqilbekAbilaev I got so frustrated with it that I ended up buying an intel network card and never bothered myself with it.

n6v26r, 2024-11-30:
Hey, I have a different problem with coexistance.
When I am on a 2.4 Ghz network, bluetooth will stop working with no software indication.

It might hapen after 12 minutes or 2 hours, but is always dies when wifi is on 2.4

FarzadMohtasham, 2024-12-07:
> Hey, I have a different problem with coexistance, on the same card (RTL8852BE) When I am on a 2.4 Ghz network, bluetooth will stop working with no software indication.
> 
> It might hapen after 12 minutes or 2 hours, but is always dies when wifi is on 2.4

I have the same problem, Could you solve it?

FarzadMohtasham, 2024-12-08:
> @FarzadMohtasham I am curious about which driver you are using? the built-in rtw89 driver in your kernel? the rtw89 driver in this repo? Or???

Nothing, Not installed rtw89 driver, The mouse was connected to my laptop with Bluetooth, but it just got disconnected after 1 hour of use, and Bluetooth no longer works.

Laptop model: Lenovo Ideapad slim 3 15IRH8
OS: Kubuntu 24.10 with latest updates
M.2 Network Interface Card Model: Realtek RTL8852BE

Any solution?

n6v26r, 2024-12-09:
> @FarzadMohtasham 
> I am curious about which driver you are using? the built-in rtw89 driver in your kernel? the rtw89 driver in this repo? Or???

I have an asus tuf A15 (2021 model)
I'm using the built-in driver.

Is there any difference between them?
Has anyone fixed the issue by changing the dirver?

a5a5aa555oo, 2024-12-09:
> > @FarzadMohtasham I am curious about which driver you are using? the built-in rtw89 driver in your kernel? the rtw89 driver in this repo? Or???
> 
> Nothing, built-in rtw89 driver in my kernel, The mouse was connected to my laptop with Bluetooth, but it just got disconnected after 1 hour of use, and Bluetooth no longer works.
> 
> Laptop model: Lenovo Ideapad slim 3 15IRH8 OS: Kubuntu 24.10 with latest updates M.2 Network Interface Card Model: Realtek RTL8852BE
> 
> Any solution?

Maybe no, I tried the latest wifi/bt driver for 8852BE in linux mainline but they still have the same problem.

FarzadMohtasham, 2024-12-09:
> Maybe no, I tried the latest wifi/bt driver for 8852BE in linux mainline but they still have the same problem.

Tried all versions of kernel with mainline, latest kernel is 6.12.3 but problem not fixed, I think should I buy brand-new Intel Wi-Fi/Bluetooth card to replace with Realtek RTL8852BE.

a5a5aa555oo, 2024-12-09:
> > @FarzadMohtasham
> > I am curious about which driver you are using? the built-in rtw89 driver in your kernel? the rtw89 driver in this repo? Or???
> 
> I have an asus tuf A15 (2021 model) I'm using the built-in driver.
> 
> Is there any difference between them? 

If you are using a kernel 6.11 or later,  It's recommended to use the built-in rtw89 driver becasue it's newer than the one here.

> Has anyone fixed the issue by changing the dirver?

I tried the latest wifi/bt driver (and firmware) in linux 6.13-rc2 but still no luck. :(

FarzadMohtasham, 2024-12-10:
> If you are using a kernel 6.11 or later, It's recommended to use the built-in rtw89 driver becasue it's newer than the one here.
>
>
> I tried the latest wifi/bt driver (and firmware) in linux 6.13-rc2 [...]

Try Kernel 6.12.2, It works well in my machine

Lenovo IdeaPad slim 3 15IRH8 I7 13620H
Wi-Fi/Bluetooth: Realtek RTL8852BE
OS: Kubuntu 24.10
Kernel: Mainline 6.12.2
Wi-Fi Power Management: 2 (No Power Saving on Wi-Fi Chip)

If you tested, Let me know

a5a5aa555oo, 2024-12-12:
@FarzadMohtasham

How is your wifi speed when your bluetooth mouse is in use?
my wifi speed is pretty slow when 2.4g wifi and bt are working at the same time.

[image]

FarzadMohtasham, 2024-12-12:
@a5a5aa555oo 

[image]
It's good to me with a bt5.2 mouse and headphone, I don't have any problem with RTL8852BE on kernel 6.12.2

Semulia, 2025-01-16:
Any thoughts how to use WiFi 2.4 and Bluetooth together? May be some how use different antennas for each or something like this?

a5a5aa555oo, 2025-02-05:
I bought a Mediatek MT7922A Wi-Fi 6E card and it works much better than 8852BE.

OS: Arch Linux
kernel version: 6.6.72-1-lts
2.4G Wi-Fi speed when a mobile phone is transfering a file to the PC at the same time.
[image]