lwfinger/rtw89 #349: Don't get it to work in openSUSE 15.6

joesse, 2024-06-30:
I had the driver working some months ago, but failed to do the kernel update right.
Now, after updating to openSUSE LEAP 15.6, I could re-make and install the driver. I can see the available networks, select my wireless network and even enter the password, but then after (automatically) reconnecting a few times I am stuck with Ethernet.
My laptop is a HP, so i also did `sudo cp 70-rtw89.conf /etc/modprobe.d/` which removed one error from dmesg.
When I try to unload the module, `sudo modprobe -rv rtw_8852ae` works fine, but `sudo modprobe -rv rtw89core` results in `modprobe: FATAL: Module rtw89core is in use.`
I put my dmesg output in a pastebin:
https://bin.disroot.org/?45d2dcb1eab4df12#FgKFYW7wCgNDKbAAeLWSJVJKktdtBLDk2ZDUQgb9FC28

brianwitte, 2024-07-03:
```
[   13.172773] rtw89core: loading out-of-tree module taints kernel.
[   13.172779] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[   13.173233] input: HD-Audio Generic Mic as /devices/pci0000:00/0000:00:08.1/0000:04:00.6/sound/card1/input16
[   13.204471] input: HD-Audio Generic HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:08.1/0000:04:00.1/sound/card0/input13
[   13.204542] input: HD-Audio Generic HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:08.1/0000:04:00.1/sound/card0/input14
[   13.204607] input: HD-Audio Generic Headphone as /devices/pci0000:00/0000:00:08.1/0000:04:00.6/sound/card1/input17
[   13.204748] input: HD-Audio Generic HDMI/DP,pcm=9 as /devices/pci0000:00/0000:00:08.1/0000:04:00.1/sound/card0/input15
[   13.206205] rtw89_8852be: disagrees about version of symbol rtw89_pci_fill_txaddr_info
[   13.206208] rtw89_8852be: Unknown symbol rtw89_pci_fill_txaddr_info (err -22)
[   13.206212] rtw89_8852be: disagrees about version of symbol rtw89_pci_recognize_intrs
[   13.206213] rtw89_8852be: Unknown symbol rtw89_pci_recognize_intrs (err -22)
[   13.206216] rtw89_8852be: disagrees about version of symbol rtw89_pci_config_intr_mask
[   13.206218] rtw89_8852be: Unknown symbol rtw89_pci_config_intr_mask (err -22)
[   13.206220] rtw89_8852be: disagrees about version of symbol rtw89_pci_ltr_set
[   13.206221] rtw89_8852be: Unknown symbol rtw89_pci_ltr_set (err -22)
[   13.206225] rtw89_8852be: disagrees about version of symbol rtw8852b_chip_info
[   13.206226] rtw89_8852be: Unknown symbol rtw8852b_chip_info (err -22)
[   13.206229] rtw89_8852be: disagrees about version of symbol rtw89_pci_enable_intr
[   13.206230] rtw89_8852be: Unknown symbol rtw89_pci_enable_intr (err -22)
[   13.206235] rtw89_8852be: disagrees about version of symbol rtw89_pci_disable_intr
[   13.206236] rtw89_8852be: Unknown symbol rtw89_pci_disable_intr (err -22)
```

here from the looks of it, you have an `8852be`, not `8852ae`. 

Can you adjust your commands and see if that helps?

joesse, 2024-07-06:
Adjusting  the commands seems to work in that way, that I can actually can unload and load the modules. I corrected /usr/lib/systemd/system-sleep/ accordingly.
Still I get no connection. I can see the available wireless networks and am asked for a password. But don't get no connection.

joesse, 2024-07-31:
Recently I installed a kernel update to version 6.4.0-150600.23.14-default x86_64. This included kernel-firmware-realtek version 20240712. Now my wireless connection is working flawlessly.
Also `sudo modprobe -rv rtw_8852be` results in `modprobe: FATAL: Module rtw_8852be not found.`
Does that mean that my 8852be is now supported 'officially'?