lwfinger/rtw89 #352: rtw8852be: Beacon set failed: -2 (No such file or directory)

sohorx, 2024-07-18:
I'm currently running on debian bookworm (linux 6.1.94) and I'm trying to use interface in AP mode through hostapd.

Yet hostapd fail trying to set beacon and stop:

```
> hostapd
...
nl80211: ssid=test
  * beacon_int=100
  * dtim_period=2
nl80211: hidden SSID not in use
nl80211: privacy=2
nl80211: auth_algs=0x1
nl80211: wpa_version=0x2
nl80211: key_mgmt_suites=0x2
nl80211: pairwise_ciphers=0x10
nl80211: group_cipher=0x10
nl80211: beacon_ies - hexdump(len=10): 7f 08 00 00 40 02 00 00 00 40
nl80211: proberesp_ies - hexdump(len=10): 7f 08 00 00 40 02 00 00 00 40
nl80211: assocresp_ies - hexdump(len=10): 7f 08 00 00 40 02 00 00 00 40
nl80211: Beacon set failed: -2 (No such file or directory)
Failed to set beacon parameters
...
```

I installed the firmware (from the link provided in the readme) and the driver (from the main branch)

```
> ethtool -i wlp9s0
driver: rtw89_8852be
version: 6.1.94-1-bb-amd64
firmware-version: N/A
expansion-rom-version:
bus-info: 0000:09:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

```
> dmesg
...
[   14.166093] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   14.796072] cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[   14.840512] cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[   14.951981] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   14.985102] intel_rapl_common: Found RAPL domain package
[   14.991232] intel_rapl_common: Found RAPL domain core
[   14.997051] intel_rapl_common: Found RAPL domain uncore
[   15.003022] intel_rapl_common: Found RAPL domain dram
[   15.050087] cfg80211: Loaded X.509 cert 'wens: 61c038651aabdcf94bd0ac7ff06c7248db18c600'
[   15.067602] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[   15.127781] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
[   16.460782] rtw89_8852be 0000:09:00.0: firmware: direct-loading firmware rtw89/rtw8852b_fw-1.bin
[   16.470373] rtw89_8852be 0000:09:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[   16.478473] ACPI: \: failed to evaluate _DSM e8c3a8d2-694b-004f-82bd-fe8607803aa7 (0x1001)
[   16.487480] rtw89_8852be 0000:09:00.0: enabling device (0000 -> 0003)
[   16.801987] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[   16.812518] rtw89_8852be 0000:09:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[   18.128827] rtw89_8852be 0000:09:00.0: chip rfe_type is 1
[   18.400273] ACPI: \: failed to evaluate _DSM e8c3a8d2-694b-004f-82bd-fe8607803aa7 (0x1001)
[   18.548931] rtw89_8852be 0000:09:00.0 wlp9s0: renamed from wlan0
...
```

The only warning I get from those logs is the ACPI failed evaluation (`[   16.478473] ACPI: \: failed to evaluate _DSM e8c3a8d2-694b-004f-82bd-fe8607803aa7 (0x1001)`) but I do not know if it's relevant. 

I'm wondering if i'm doing something wrong or if this issue is known and/or already resolved ?

pkshih, 2024-07-30:
I can start AP mode with the latest upstream driver. Can you share your hostapd.conf?

The ACPI evaluation is used to determine if platform can support some features, but platforms might not define the fields causing message of "failed to evaluate ...". The message can be ignored.

sha534, 2024-08-17:
I have similar problem
Config:
`
# 802.11b/g/n with WPA2-PSK

# the interface used by the AP
interface=wlan0
# "g" simply means 2.4GHz band
hw_mode=g
# the channel to use
channel=10
# limit the frequencies used to those allowed in the country
ieee80211d=1
# the country code
country_code=US
# 802.11n support
ieee80211n=1
# QoS support, also required for full speed on 802.11n/ac/ax
wmm_enabled=1

# the name of the AP
ssid=InfoSec-Test
# 1=wpa, 2=wep, 3=both
auth_algs=1
# WPA2 only
wpa=2
wpa_key_mgmt=WPA-PSK
rsn_pairwise=CCMP
wpa_passphrase=ComplexPassword123

#1 - only open system authentication
#2 - both open system authentication and shared key authentication
auth_algs=2
`

Debian 12

pkshih, 2024-08-19:
Using upstream driver and Larry's driver (main branch with some fixes), I run hostap with your config under kernel  6.10.2, it can start without "nl80211: Beacon set failed: -2 (No such file or directory)".
Could you try it on the new kernel?

rm7rzedg, 2024-09-17:
> Using upstream driver and Larry's driver (main branch with some fixes [1]), I run hostap with your config under kernel 6.10.2, it can start without "nl80211: Beacon set failed: -2 (No such file or directory)". Could you try it on the new kernel?
> 
> [1] [#356 (comment)](https://github.com/lwfinger/rtw89/issues/356#issuecomment-2295542386)

Having the same problem.
Unfortunately the latest kernel for proxmox ve is currently 6.8.