lwfinger/rtw89 #353: Incompatibility with Thinkpad E16 RTL8852BE PCIe 802.11ax Wireless Network Controller

PickleGaming32, 2024-07-23:
Facing issue on multiple linux distros (Fedora, openSUSE, debian, archlinux) where Wireless Network Controller shuts down completely after unsepcified amount of time. It shows up in lspci, but after it shuts down, neither lspci or any other form of hardware monitoring manages to show it. Even after having installed these drivers correctly, with the help of other archlinux users, I am facing the same issue, such that I have been advised to submit a ticket.

debuggerboy, 2024-08-03:
I believe may be  I am suffering from the same problem, with my Lenovo Thinkpad E16.

In my case I am using Debian GNU/Linux 12, as the official debian kernels do not support my wireless NIC, I was using `lwfinger/rtw89` .

The drivers were working fine on linux kernel images:

- linux-image-6.1.0-13-amd64
- linux-image-6.1.0-16-amd64
- linux-image-6.1.0-17-amd64
- linux-image-6.1.0-18-amd64

but from the next releases onwards  the driver builds are successful, but the the new driver is not working. 

So I have went back to linux kernel image `linux-image-6.1.0-18-amd64` from debian repo. That is how I continue to work from my E16.

Thanks

PickleGaming32, 2024-08-03:
Thanks for your input.

While I am thankful that there are some linux (especially debian) images that work with this NIC, do you have any idea if some of the older archlinux images may solve this issue? Debian's awesome, but I'd love to stick to arch if at all possible.

Thanks.

debuggerboy, 2024-09-28:
@PickleGaming32 

I was reading this issue reported:

> https://github.com/lwfinger/rtw89/issues/364

In this issue @prodrone-liu got the WiFi driver working with commit [#ebe87ac](https://github.com/lwfinger/rtw89/commit/ebe87ac234a72a63a1b2ede911a874f1b7a6fe43) in his Ubuntu 20.04

I tried checking out to this commit [#ebe87ac](https://github.com/lwfinger/rtw89/commit/ebe87ac234a72a63a1b2ede911a874f1b7a6fe43) in my Thinkpad E16 with Debian 12 and the driver builds are successful, also the new WiFi driver is not working on my Debian.

May be this commit [#ebe87ac](https://github.com/lwfinger/rtw89/commit/ebe87ac234a72a63a1b2ede911a874f1b7a6fe43) will help you. Give it a try.

Thanks.

Sam2much96, 2025-05-20:
Mine was missing a particular linux firmware and I Fixed it by

1. Debugging my wifi firmmware with `sudo dmesg | grep -i firmware`. It showed that one of my firmware binaries had failed to load `rtw89_8852be 0000:03:00.0: firmware: failed to load rtw89/rtw8852b_fw-1.bin (-2)`  
2. I downloaded the missing binary from linux repository and saved it as `/lib/firmware/rtw89/rtw8852b_fw-1.bin`
3. i ran `dracut regenerate --all` to regenerate initramfs
4. It worked after rebooting