lwfinger/rtw89 #356: Cannot build on Linux 6.10 Kernel

LFRon, 2024-08-05:
I use Linux 6.10 Kernel and use make command,it reports:
/home/mozixun/Cache/rtw89/phy.c: In function ‘rtw89_phy_ra_sta_update’:
/home/mozixun/Cache/rtw89/phy.c:397:21: warning: unused variable ‘mcs_map’ [-Wunused-variable]
  397 |                 u16 mcs_map = le16_to_cpu(sta->deflink.vht_cap.vht_mcs.rx_mcs_map);
      |                     ^~~~~~~
/home/mozixun/Cache/rtw89/phy.c:399:23: error: ‘struct ieee80211_sta’ has no member named ‘vht_cap’
  399 |         } else if (sta->vht_cap.vht_supported) {
      |                       ^~
In file included from ./include/linux/byteorder/little_endian.h:5,
                 from ./arch/x86/include/uapi/asm/byteorder.h:5,
                 from ./include/asm-generic/qrwlock_types.h:6,
                 from ./arch/x86/include/asm/spinlock_types.h:7,
                 from ./include/linux/spinlock_types_raw.h:7,
                 from ./include/linux/ratelimit_types.h:7,
                 from ./include/linux/printk.h:9,
                 from ./include/asm-generic/bug.h:22,
                 from ./arch/x86/include/asm/bug.h:87,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/average.h:5,
                 from /home/mozixun/Cache/rtw89/core.h:8,
                 from /home/mozixun/Cache/rtw89/coex.h:8,
                 from /home/mozixun/Cache/rtw89/phy.c:5:
/home/mozixun/Cache/rtw89/phy.c:400:46: error: ‘struct ieee80211_sta’ has no member named ‘vht_cap’
  400 |                 u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
      |                                              ^~
./include/uapi/linux/byteorder/little_endian.h:37:51: note: in definition of macro ‘__le16_to_cpu’
   37 | #define __le16_to_cpu(x) ((__force __u16)(__le16)(x))
      |                                                   ^
/home/mozixun/Cache/rtw89/phy.c:400:31: note: in expansion of macro ‘le16_to_cpu’
  400 |                 u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
      |                               ^~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /home/mozixun/Cache/rtw89/phy.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.10.0-dppve/Makefile:1934: /home/mozixun/Cache/rtw89] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.10.0-dppve'
make: *** [Makefile:104: all] Error 2

Could developer improve the support of Linux higher kernel version?Thanks!

gustavoflw, 2024-08-05:
I managed to get my WiFi working again in Arch Linux by uninstalling this driver and unblacklisting the kernel modules in ``/etc/modprobe.d/``.

LFRon, 2024-08-06:
> I managed to get my WiFi working again in Arch Linux by uninstalling this driver and unblacklisting the kernel modules in `/etc/modprobe.d/`.

Sounds good!How to do it?

gustavoflw, 2024-08-06:
> > I managed to get my WiFi working again in Arch Linux by uninstalling this driver and unblacklisting the kernel modules in `/etc/modprobe.d/`.
> 
> Sounds good!How to do it?
```sh
# Uninstall the driver. In my case, it had been installed with yay
yay -R rtw89-dkms-git

# Unblacklist the module. In my case there was a file that I had placed for blacklisting kernel modules. Find the file in:
ls /etc/modprobe.d/

# Rebuild initramfs
sudo mkinitcpio -P

# Mine was fine after rebooting
reboot
```

CharSnipeur, 2024-08-14:
my modifs to compil under 6.10.4 :
```
diff --git a/pci.c b/pci.c
index 7b00476..3817ae6 100644
--- a/pci.c
+++ b/pci.c
@@ -3637,7 +3637,7 @@ static int rtw89_pci_request_irq(struct rtw89_dev *rtwdev,
        unsigned long flags = 0;
        int ret;
 
-       flags |= PCI_IRQ_LEGACY | PCI_IRQ_MSI;
+       flags |=  PCI_IRQ_MSI;
        ret = pci_alloc_irq_vectors(pdev, 1, 1, flags);
        if (ret < 0) {
                rtw89_err(rtwdev, "failed to alloc irq vectors, ret %d\n", ret);
diff --git a/phy.c b/phy.c
index 8427b32..789d91a 100644
--- a/phy.c
+++ b/phy.c
@@ -395,7 +395,7 @@ static void rtw89_phy_ra_sta_update(struct rtw89_dev *rtwdev,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 0) || (RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(9, 0)))
        } else if (sta->deflink.vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->deflink.vht_cap.vht_mcs.rx_mcs_map);
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 11, 0)
        } else if (sta->vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
 #endif

```
But modules don't insert good => Invalid Aggument... :-(

CharSnipeur, 2024-08-18:
According to https://lore.kernel.org/linux-rdma/37fcc27a-7c6b-4aed-88be-92aadfaa67fe@intel.com/t/ I replace **PCI_IRQ_LEGACY** by **PCI_IRQ_INTX**
I think is better.
I have the "warning" _Skipping BTF generation [my module name] due to unavailabilty of vmlinux_ And use https://askubuntu.com/questions/1348250/skipping-btf-generation-xxx-due-to-unavailability-of-vmlinux-on-ubuntu-21-04 to correct this.
But Always : `# modprobe -v rtw_8851b
insmod /lib/modules/6.10.4-custom/kernel/drivers/net/wireless/realtek/rtw89/rtw89core.ko.xz 
modprobe: ERROR: could not insert 'rtw_8851b': Invalid argument
`

pkshih, 2024-08-19:
My changes for kernel 6.10.2. 
I think it can be also compatible to older kernels. 

@brianwitte 
Could you merge this change below? 

```
diff --git a/pci.c b/pci.c
index 7b00476a5dee..e7ee2729179f 100644
--- a/pci.c
+++ b/pci.c
@@ -3637,7 +3637,11 @@ static int rtw89_pci_request_irq(struct rtw89_dev *rtwdev,
        unsigned long flags = 0;
        int ret;

-       flags |= PCI_IRQ_LEGACY | PCI_IRQ_MSI;
+#ifndef PCI_IRQ_INTX
+#define PCI_IRQ_INTX PCI_IRQ_LEGACY
+#endif
+
+       flags |= PCI_IRQ_INTX | PCI_IRQ_MSI;
        ret = pci_alloc_irq_vectors(pdev, 1, 1, flags);
        if (ret < 0) {
                rtw89_err(rtwdev, "failed to alloc irq vectors, ret %d\n", ret);
diff --git a/phy.c b/phy.c
index 8427b3231560..bb00905287bc 100644
--- a/phy.c
+++ b/phy.c
@@ -395,10 +395,9 @@ static void rtw89_phy_ra_sta_update(struct rtw89_dev *rtwdev,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 0) || (RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(9, 0)))
        } else if (sta->deflink.vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->deflink.vht_cap.vht_mcs.rx_mcs_map);
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)
+#elif LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)
        } else if (sta->vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
-#endif
 #else
                u16 mcs_map = 0;
 #endif

```

brianwitte, 2024-08-19:
hey, @pkshih. great to see you :sunglasses:  and thank you for the patch!

I just applied it here. Patch is applied to `main`.

https://github.com/lwfinger/rtw89/commit/d823c59fb32bd16f080f9a3c4ffb971b0e7f6c79

Just a heads up, I have tried emailing you a few times, but I get blocked by the mail server. I have some updates on backports related progress and some other rtw89 related things (patches for usb support that I was working on with Larry that I have made progress on).

Here are my two emails:

brianwitte@mailfence.com and brian.witte@mailbox.org

I used to use the mailfence.com one, but that one is not used for most of my open source work so the mailbox.org one is the only one I really need you to whitelist.

pkshih, 2024-08-19:
Hi @brianwitte ,

Thanks for the help.

By the way, I failed to mail you brianwitte@mailfence.com and brian.witte@mailbox.org via my company and gmail.

```
<brianwitte@mailfence.com>... Deferred: Connection timed out with
smtp2.mailfence.com.
reason: 550 5.1.1 <brian.witte@mailbox.org>: Recipient address
rejected: User unknown in relay recipient table
```

Not sure what's wrong. But it seems not suitable to discuss it in this thread.

CharSnipeur, 2024-08-19:
Hum... 
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 0) || (RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(9, 0)))
#elif LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)
#endif
```
Maybe I'm wrong, but you never fall in "LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)", because LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0) **implie** LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 0).
No ?

pkshih, 2024-08-20:
You are correct.

The commit 6cc0f850ddb61 messed up the code that we should revert: 
```
@@ -401,12 +393,15 @@ static void rtw89_phy_ra_sta_update(struct rtw89_dev *rtwdev,
                rtw89_phy_ra_gi_ltf(rtwdev, rtwsta, chan, &fix_giltf_en, &fix_giltf);
 #endif
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 0) || (RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(9, 0)))
-        } else if (sta->deflink.vht_cap.vht_supported) {
-                u16 mcs_map = le16_to_cpu(sta->deflink.vht_cap.vht_mcs.rx_mcs_map);
-#else
+       } else if (sta->deflink.vht_cap.vht_supported) {
+               u16 mcs_map = le16_to_cpu(sta->deflink.vht_cap.vht_mcs.rx_mcs_map);
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)
        } else if (sta->vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
 #endif
+#else
+               u16 mcs_map = 0;
+#endif

                mode |= RTW89_RA_MODE_VHT;
                csi_mode = RTW89_RA_RPT_MODE_VHT;

```

In fact, there is no change after kernel 6.10, so we need below change to correct it.
Tested on kernel 6.10.2. 
(@brianwitte Please help to apply below change. Sorry for the mistakes in https://github.com/lwfinger/rtw89/issues/356#issuecomment-2295542386)

```
diff --git a/phy.c b/phy.c
index bb00905287bc..8583af51991c 100644
--- a/phy.c
+++ b/phy.c
@@ -395,11 +395,9 @@ static void rtw89_phy_ra_sta_update(struct rtw89_dev *rtwdev,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 19, 0) || (RHEL_RELEASE_CODE >= RHEL_RELEASE_VERSION(9, 0)))
        } else if (sta->deflink.vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->deflink.vht_cap.vht_mcs.rx_mcs_map);
-#elif LINUX_VERSION_CODE >= KERNEL_VERSION(6, 10, 0)
+#else
        } else if (sta->vht_cap.vht_supported) {
                u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
-#else
-               u16 mcs_map = 0;
 #endif

                mode |= RTW89_RA_MODE_VHT;
```

brianwitte, 2024-08-23:
hey, @pkshih. reverting tonight, sorry I somehow missed these notifications.

thank you, @CharSnipeur :+1:

brianwitte, 2024-08-25:
@pkshih -- applied here -> https://github.com/lwfinger/rtw89/commit/d1fced1b8a741dc9f92b47c69489c24385945f6e