lwfinger/rtw89 #357: firmware package version

jaddle, 2024-08-20:
In the readme, the "note regarding firmware" says that firmware-realtek >= 20230117-1 should work, so the current debian stable package (20230210-5) should be fine. But this package doesn't contain the required rtw8852b_fw-1.bin so I had errors about failed firmware loading. Upgrading to the version of the package from debian unstable (20240709-1) corrected the problem. I'm not sure which version of the package introduced the missing firmware, but it was certainly later than stated in the README!

The connection still isn't working (failing to authenticate, and all wireless networks are showing in network manager with zero bars of signal strength, but with the firmware properly loaded, at least it's getting that far!

This is on a lenovo yoga 6, by the way, with the RTK8852BE inside.

jaddle, 2024-08-20:
For the record, the fix mentioned in https://github.com/lwfinger/rtw89/issues/340 (revert back to https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61) fixed everything, and it seems to work now just fine.

TheMostOGName, 2024-08-21:
> For the record, the fix mentioned in #340 (revert back to [6dc9441](https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61)) fixed everything, and it seems to work now just fine.

Thank you very, very much for this. I've been experiencing this issue as well for about a month I believe, and I've just been not dealing with it because a) I've been busy, b) I haven't really needed to use wifi recently, and c) I dual boot and my other operating systems support wifi without this driver. But, it's still been driving me insane, so I've been trying to fix it for the past few days. I got about as far as you did in your first post, I found out that rtw8852b_fw-1.bin was missing and when I tried to manually re-add it I was able to search for wifi networks but entirely unable to connect to them (0% signal strength, just like you described). But yeah, I've been driving myself insane for this, so I was thrilled to see a new issue here describing my exact problem _and_ providing a possible solution I haven't tried yet. Reverting back to 6dc9441 worked perfectly for me as well! Tyvm for restoring the wifi for my machine, right when I'll be needing to use it too!

If it helps, I'm using Debian 12, with an HP Pavilion 15t-eg300 and a RTL8852BE PCIe network controller. I'm also definitely happy to answer any other questions or try different stuff if it helps get to the bottom of the issue too, so if I can be any help let me know.