lwfinger/rtw89 #359: Debian 12, signals are 0% strength & cannot connect

OzSecurityzugo, 2024-08-28:
Hi there, been trying for a few days getting the driver to work.

eventually got it "loading" having secure boot enabled, enrolled dkms key and yea installed with DKMS.
I also had to manually add some firmwire namely rtw8852b_fw-1.bin from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89

I have a Lenovo and so i also followed the guidance to add the config files that are needed for the options of the driver.

lsmod | grep rtw

```
rtw_8852be             16384  0
rtw_8852b             372736  1 rtw_8852be
rtw89pci               90112  1 rtw_8852be
rtw89core             741376  2 rtw89pci,rtw_8852b
mac80211             1175552  2 rtw89pci,rtw89core
cfg80211             1142784  3 rtw89core,mac80211,rtw_8852
```
sudo dmesg | grep rtw

```
[   20.184690] rtw89core: loading out-of-tree module taints kernel.
[   20.223844] rtw89_8852be 0000:2c:00.0: firmware: direct-loading firmware rtw89/rtw8852b_fw-1.bin
[   20.223848] rtw89_8852be 0000:2c:00.0: loaded firmware rtw89/rtw8852b_fw-1.bin
[   20.223914] rtw89_8852be 0000:2c:00.0: enabling device (0000 -> 0003)
[   20.225136] rtw89_8852be 0000:2c:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 5
[   20.225138] rtw89_8852be 0000:2c:00.0: Firmware version 0.29.29.5 (da87cccd), cmd version 0, type 3
[   20.525761] rtw89_8852be 0000:2c:00.0: chip rfe_type is 1
[   20.594616] rtw89_8852be 0000:2c:00.0 wlp44s0: renamed from wlan0
```
ip link | grep wlp44

`
3: wlp44s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DORMANT group default qlen 1000
`

Does anyone have any idea? I've read that others got it working switching to unstable debian but I do not wanna do that as I am actually quite new to it ( few months )

OzSecurityzugo, 2024-08-28:
Another useful output maybe?

sudo dkms status
`
rtw89/1.0.2, 6.1.0-23-amd64, x86_64: installed`

TheMostOGName, 2024-08-30:
Does the solution in #357 and/or #340 fix this issue (revert to [6dc9441](https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61) ) do anything? This seems almost identical to what I was experiencing, that solution worked for me.

OzSecurityzugo, 2024-09-02:
> Does the solution in #357 and/or #340 fix this issue (revert to [6dc9441](https://github.com/lwfinger/rtw89/commit/6dc9441698a7f2f79ff8f74c6ea3704c0c8feb61)) do anything? This seems almost identical to what I was experiencing, that solution worked for me.

This solved it, thanks!

closed 2024-09-02 by OzSecurityzugo