lwfinger/rtw89 #361: rtw89_8851be: probe of 0000:01:00.0 failed with error -22

rautyrauty, 2024-08-30:
Initially, the Internet worked. I tried to solve the problem with sleep mode, went into the BIOS, turned fastboot on and off. Then I noticed that the Internet was gone. I did not update the firmware. No packages either. Reinstalled Linux - the problem remains.

0000:01:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. Device [10ec:b851]
<details>
<summary>cat sos_commands/kernel/dmesg | grep rtw89</summary>
<pre>
[    2.326809] rtw89core: loading out-of-tree module taints kernel.
[    2.327013] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[    2.375114] rtw89_8851be 0000:01:00.0: loaded firmware rtw89/rtw8851b_fw.bin
[    2.380651] rtw89_8851be 0000:01:00.0: Firmware version 0.29.41.3 (65cefb31), cmd version 0, type 5
[    2.380663] rtw89_8851be 0000:01:00.0: Firmware version 0.29.41.3 (65cefb31), cmd version 0, type 3
[    2.800671] rtw89_8851be 0000:01:00.0: fw security fail
[    2.800675] rtw89_8851be 0000:01:00.0: download firmware fail
[    2.800676] rtw89_8851be 0000:01:00.0: failed to setup chip information
[    2.801781] rtw89_8851be: probe of 0000:01:00.0 failed with error -22
</pre>
</details>

[lspci.txt](https://github.com/user-attachments/files/16818323/lspci.txt)

rautyrauty, 2024-09-02:
Even more logs with the Internet over the wire:
[dmesg-grep-rtw-bad.log](https://github.com/user-attachments/files/16836494/dmesg-grep-rtw-bad.log)

I found logs with working wifi and there is not a line with rtw:
[dmesg-good.log](https://github.com/user-attachments/files/16836571/dmesg-good.log)

I found someone else who is facing this problem:
https://bbs.archlinux.org/viewtopic.php?id=298678
I cleared the CMOS according to this guide:
https://www.youtube.com/watch?v=l9p0sAZ4Jno
And the wifi is working again! I broke it again as follows: Go to BIOS -> Exit without saving changes.

rautyrauty, 2024-09-02:
I also installed Windows near Linux and wifi doesn't work there either :)

Narsimha96, 2024-09-09:
@rautyrauty I'm facing the same exact issue. Did you solve the issue? Os detects realtek wifi card but it doesnt show up in settings.

rautyrauty, 2024-09-09:
@Narsimha96 No, I haven't solved the problem. This is a laptop from ASUS - I need to write to them, since the problem is with Windows too (They don't support Linux).

Narsimha96, 2024-09-09:
I have ASUS Z790-V AX mobo. I installed windows, but drivers seem to be working fine. I'm facing issues with just ubuntu

pkshih, 2024-11-06:
A patch [1] might be related to this issue. The upcoming 6.12 will contain this patch. 

[1] https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/drivers/net/wireless/realtek/rtw89?id=80fb81bb46a57daedd5decbcc253ea48428a254e

rautyrauty, 2024-11-07:
> A patch [1] might be related to this issue. The upcoming 6.12 will contain this patch.
> 
> [1] https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/drivers/net/wireless/realtek/rtw89?id=80fb81bb46a57daedd5decbcc253ea48428a254e

Great! I will definitely check the fix this week or next!

rautyrauty, 2024-11-08:
Tested, it works. Thank you for your work, @pkshih.

closed 2024-11-08 by rautyrauty