lwfinger/rtw89 #365: Realtek 8852CE Bluetooth and Wi-Fi not working on Ubuntu 20.04 (Kernel 5.15.0-122)

Leo5050xvjf, 2024-10-04:
### System Information:
- Operating System: Ubuntu 20.04
- Kernel: 5.15.0-122-generic
- Motherboard: Z790 AORUS ELITE AX-W
- Wireless Network Adapter: Realtek 8852CE
- Driver Version: rtw89 (commit 6dc944)
- Firmware Loaded: rtw8852c_fw.bin

### Problem Description:
Both Bluetooth and Wi-Fi are not working properly on the system. Bluetooth cannot detect devices, and dmesg shows the message: Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000c, hci ver 000c, assuming no firmware upload is required. The Wi-Fi adapter can scan and find available networks, but it cannot connect to any of them. Attempts to connect result in failures, and further investigation may be required.

### Steps Taken:

1. Installed and compiled the latest rtw89 driver, and switched to a specific commit (6dc944).
2. Verified that the firmware exists and loaded the correct rtw8852c_fw.bin.
3. Driver modules are loaded correctly (lsmod shows rtw_8852ce and rtw89core modules).

### Logs and Related Outputs:

- `dmesg | grep -i bluetooth`:
[    5.873876] usb 1-14: Product: Bluetooth Radio
[    6.360216] Bluetooth: Core ver 2.22
[    6.360338] NET: Registered PF_BLUETOOTH protocol family
[    6.360339] Bluetooth: HCI device and connection manager initialized
[    6.360342] Bluetooth: HCI socket layer initialized
[    6.360343] Bluetooth: L2CAP socket layer initialized
[    6.360345] Bluetooth: SCO socket layer initialized
[    6.390009] Bluetooth: hci0: RTL: examining hci_ver=0c hci_rev=000c lmp_ver=0c lmp_subver=8852
[    6.601999] Bluetooth: hci0: RTL: examining hci_ver=0c hci_rev=000c lmp_ver=0c lmp_subver=8852
[    6.602001] Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000c, hci ver 000c
[    6.602002] Bluetooth: hci0: RTL: assuming no firmware upload needed
[    6.897408] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.897409] Bluetooth: BNEP filters: protocol multicast
[    6.897411] Bluetooth: BNEP socket layer initialized
[  102.806725] Bluetooth: RFCOMM TTY layer initialized
[  102.806729] Bluetooth: RFCOMM socket layer initialized
[  102.806731] Bluetooth: RFCOMM ver 1.11
[ 1093.331884] Bluetooth: hci0: urb 00000000ffdf91f6 failed to resubmit (2)
[ 1093.496013] Bluetooth: hci0: RTL: examining hci_ver=0c hci_rev=000c lmp_ver=0c lmp_subver=8852
[ 1093.704145] Bluetooth: hci0: RTL: examining hci_ver=0c hci_rev=000c lmp_ver=0c lmp_subver=8852
[ 1093.704151] Bluetooth: hci0: RTL: unknown IC info, lmp subver 8852, hci rev 000c, hci ver 000c
[ 1093.704152] Bluetooth: hci0: RTL: assuming no firmware upload needed

- run `bluetoothctl`
[bluetooth]# scan on
Discovery started
[CHG] Controller 7C:66:EF:F1:06:95 Discovering: yes
[bluetooth]#

- hciconfig
hci0:	Type: Primary  Bus: USB
	BD Address: 7C:66:EF:F1:06:95  ACL MTU: 1021:8  SCO MTU: 255:12
	UP RUNNING PSCAN ISCAN 
	RX bytes:4009 acl:0 sco:0 events:390 errors:0
	TX bytes:12910 acl:0 sco:0 commands:370 errors:0

Please help confirm if further driver updates are needed or if there are any other possible solutions.

Thank you!

tac-dog, 2024-11-20:
I have the same issue as you

tac-dog, 2024-11-20:
I have the same issue as you.

tac-dog, 2024-11-21:
Driver Version: rtw89 (commit 6dc944)

tac-dog, 2024-11-21:
Driver Version: rtw89 (commit 6dc944),when I return to this version，my Wi-Fi  can work.

MiangChen, 2024-11-21:
I have the same problem

MiangChen, 2024-11-21:
> Driver Version: rtw89 (commit 6dc944),when I return to this version，my Wi-Fi can work.

`git checkout 6dc944` works

nuricanozturk01, 2025-01-12:
> > Driver Version: rtw89 (commit 6dc944),when I return to this version，my Wi-Fi can work.
> 
> `git checkout 6dc944` works

wifi connection is gone after a while in `6dc944`

MiangChen, 2025-01-13:
> > > Driver Version: rtw89 (commit 6dc944),when I return to this version，my Wi-Fi can work.
> > 
> > 
> > `git checkout 6dc944` works
> 
> wifi connection is gone after a while in `6dc944`. In my case, `ebe87ac` is working

I also encountered this problem after I recently updated Ubuntu 20.04. This time I needed to use `ebe87ac` to successfully install the WIFI driver

closed 2025-04-04 by Leo5050xvjf