lwfinger/rtw89 #367: Cannot build on Linux Kernel 6.11.3-3-cachyos

mxaban, 2024-10-13:
When trying to build the module it gives me the following error:

Install DKMS modules
==> dkms install --no-depmod rtw89/r1090.d1fced1 -k 6.11.3-3-cachyos
Error! Bad return status for module build on kernel: 6.11.3-3-cachyos (x86_64)
Consult /var/lib/dkms/rtw89/r1090.d1fced1/build/make.log for more information.
==> WARNING: `dkms install --no-depmod rtw89/r1090.d1fced1 -k 6.11.3-3-cachyos' exited 10

Content of /var/lib/dkms/rtw89/r1090.d1fced1/build/make.log

DKMS make.log for rtw89-r1090.d1fced1 for kernel 6.11.3-3-cachyos (x86_64)
dom 13 oct 2024 06:36:03 CST
make: se entra en el directorio '/usr/lib/modules/6.11.3-3-cachyos/build'
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/core.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/mac80211.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/mac.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/mac_be.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/phy.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/phy_be.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/fw.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/cam.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/efuse.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/efuse_be.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/regd.o
  CC [M]  /var/lib/dkms/rtw89/r1090.d1fced1/build/sar.o
/var/lib/dkms/rtw89/r1090.d1fced1/build/mac80211.c:1177:12: error: incompatible function pointer types initializing 'void (*)(struct ieee80211_hw *, bool)' (aka 'void (*)(struct ieee80211_hw *, _Bool)') with an expression of type 'void (struct ieee80211_hw *)' [-Wincompatible-function-pointer-types]
 1177 |         .stop                   = rtw89_ops_stop,
      |                                   ^~~~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/Makefile.build:244: /var/lib/dkms/rtw89/r1090.d1fced1/build/mac80211.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
make[1]: *** [/usr/lib/modules/6.11.3-3-cachyos/build/Makefile:1934: /var/lib/dkms/rtw89/r1090.d1fced1/build] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: se sale del directorio '/usr/lib/modules/6.11.3-3-cachyos/build'

A similar error appears with rtw89-BT

Any help is appreciated, thanks

pkshih, 2024-11-06:
I think you can just use built-in driver of kernel 6.11. 
Did you encounter problems? Which WiFi chip you are using?

pkshih, 2024-11-07:
We found interoperability problem if platforms are not Intel, and the patch is draining to stable 6.11. 
Please try kernel 6.10 with built-in driver to see if it works to you. 

By the way, attaching log will be helpful to understand the case you met.

ppogorze, 2024-11-21:
I can easily build module on latest 6.12 cachyos kernel, but it think my install still uses old one.
Also there could be some regression - on 6.6 LTS kernel when downloading large files via wifi (5GHZ) my BT audio and mouse ok, on 6.11+ its choppy and unresponsive.

a5a5aa555oo, 2024-11-21:
The rtw89 driver here is older than the one in kernel 6.11 and 6.12, why do you still insist on using this driver?

a5a5aa555oo, 2024-11-22:
@ppogorze 

Do you wanna try [my rtw89 driver](https://github.com/a5a5aa555oo/rtw89)? kernel 6.6 ~ 6.13 are supported.
You can simply install [rtw89-dkms-git](https://aur.archlinux.org/packages/rtw89-dkms-git) if you are using a Arch-based Linux distro.

ppogorze, 2024-11-23:
Thanks, great.
Weird thing is when I connect AirPods (even for a few seconds) my all issues is gone until reboot.

I will test your driver and come back with more info.

btw nothing weird in btmon, journal and dmesg

a5a5aa555oo, 2024-11-23:
> Thanks, great. Weird thing is when I connect AirPods (even for a few seconds) my all issues is gone until reboot.
> 
> I will test your driver and come back with more info.
> 
> btw nothing weird in btmon, journal and dmesg
> 
> edit; same issue with dkms from arch. maybe its related to usb power management in newer kernels, because bluetooth is using btusb?

I'm not a pro driver programer, so the only thing I can do  is backporting the latest stable rtw89 driver to older kernels.
if you found the latest stable driver still has some issues, it is recommended to email [linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org) as mentioned in README.md
sorry that there is nothing I can do. :)

`
When you have problems where the driver builds and loads correctly, but fails to work, a GitHub issue is NOT the best place to report it. I have no idea of the internal workings of any of the chips, and the Realtek engineers who do will not read these issues. To reach them, send E-mail to [linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org). Include a detailed description of any messages in the kernel logs and any steps that you have taken to analyze or fix the problem. If your description is not complete, you are unlikely to get any satisfaction. One other thing - your mail MUST be plain test. HTML mail is rejected.
`

a5a5aa555oo, 2025-07-04:
@eyalroz 

The kernel provided by your distro is pretty new, does your Wi-Fi device not work with that kernel? any error messages in kernel log?

> You write that the repo you linked to supports kernel versions 6.6 - 6.13; but the repo itself says:
> so... which is it? :-(

At the moment I wrote that reply, my repo contains two branches: 
1. main branch: this branch contains a rtw89 driver that support kernel 6.6+
2. 6.6-lts branch: this branch contains a rtw89 driver that support kernel 5.15 ~ 6.5

On 20 May 2025, I transferred my main branch to morrownr because they needed a rtw89 repo to test the new drivers for USB Wi-Fi dongles, so the main branch is now at https://github.com/morrownr/rtw89/

Now try the rtw89 driver at https://github.com/morrownr/rtw89/ instead.