lwfinger/rtw89 #369: Rtw 89 issue

MaxHumeniuk, 2024-11-23:
6.451940] rtw89core: loading out-of-tree module taints kernel.
[ 6.452415] rtw89core: module verification failed: signature and/or required key missing tainting kernel 6.545867] rtw89_8852be 0000:02:00.0: firmware: direct-loading firmware rtw89/rtw8852b_fw-1.bin
This error is in logs. Do you know how to fix it?
[image]

a5a5aa555oo, 2024-11-23:
Those warnings messages are shown in my system too, you can ignore them if your driver works fine.

MaxHumeniuk, 2024-11-23:
> Those warnings messages are shown in my system too, you can ignore them if your driver works fine.

He is actually not working fine. I can see other wi-fi but to connect to them i can't, waiting 2-4 minutes while conneting and then i got disconnected(

a5a5aa555oo, 2024-11-23:
What distro do you use? kernel version?

MaxHumeniuk, 2024-11-23:
> What distro do you use? kernel version?

debian 12 
kernel 6.1.0-27-amd64

a5a5aa555oo, 2024-11-23:
The driver in this repo only works with kernel 6.9 and 6.10.
You can upgrade your kernel to v6.11, the built-in rtw89 driver in kernel 6.11 works fine with 8852BE (for most 8852BE users).

First, add this line to /etc/apt/sources.list
```deb http://deb.debian.org/debian bookworm-backports main```

and then install kernel v6.11 from bookworm-backports repo
```
sudo apt update
sudo apt install -t bookworm-backports linux-image-amd64 linux-headers-amd64
sudo reboot
```

MaxHumeniuk, 2024-11-23:
Does this driver also works with RTL8852BE?

a5a5aa555oo, 2024-11-23:
> Does this driver also works with RTL8852BE?

"this driver"? Do you mean the driver in this repo?
Yes, it supports RTL8852BE, but unfortunately it only works with kernel 6.9 and 6.10 currently.

if you don't want to upgrade your kernel, you can try the [driver in my repo](https://github.com/a5a5aa555oo/rtw89/tree/6.6-lts), it supports kernel 5.15 ~ 6.6.
i'm testing it on Debian 11.11 with kernel 5.15 and it works fine.
```
# remove the rtw89 driver which doesn't work for you
sudo rm -rf /lib/modules/$(KVER)/kernel/drivers/net/wireless/realtek/rtw89

# download, build and then install
git clone -b 6.6-lts https://github.com/a5a5aa555oo/rtw89
cd rtw89
make
sudo make install
```

MaxHumeniuk, 2024-11-23:
> > Does this driver also works with RTL8852BE?
> 
> "this driver"? Do you mean the driver in this repo? Yes, it supports RTL8852BE, but unfortunately it only works with kernel 6.9 and 6.10 currently.
> 
> if you don't want to upgrade your kernel, you can try the [driver in my repo](https://github.com/a5a5aa555oo/rtw89/tree/6.6-lts), it supports kernel 5.15 ~ 6.6. i'm testing it on Debian 11.11 with kernel 5.15 and it works fine.
> 
> ```
> # remove the rtw89 driver which doesn't work for you
> sudo rm -rf /lib/modules/$(uname -r)/kernel/drivers/net/wireless/realtek/rtw89
> 
> # download, build and then install
> git clone -b 6.6-lts https://github.com/a5a5aa555oo/rtw89
> cd rtw89
> make
> sudo make install
> ```

Ok, thanks i will try it too.

MaxHumeniuk, 2024-11-25:
Yes it worked, thanks

closed 2024-11-25 by MaxHumeniuk