lwfinger/rtw89 #370: rtw 89 kernal 6.11.2 issue  RTL8852BE

sahl36, 2024-11-24:
make -C /lib/modules/6.11.5-amd64/build M=/home/kkiragu/Documents/git/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-6.11.5-amd64'
  CC [M]  /home/kkiragu/Documents/git/rtw89/core.o
  CC [M]  /home/kkiragu/Documents/git/rtw89/mac80211.o
/home/kkiragu/Documents/git/rtw89/mac80211.c:1177:35: error: initialization of ‘void (*)(struct ieee80211_hw *, bool)’ {aka ‘void (*)(struct ieee80211_hw *, _Bool)’} from incompatible pointer type ‘void (*)(struct ieee80211_hw *)’ [-Wincompatible-pointer-types]
 1177 |         .stop                   = rtw89_ops_stop,
      |                                   ^~~~~~~~~~~~~~
/home/kkiragu/Documents/git/rtw89/mac80211.c:1177:35: note: (near initialization for ‘rtw89_ops.stop’)
make[3]: *** [/usr/src/linux-headers-6.11.5-common/scripts/Makefile.build:249: /home/kkiragu/Documents/git/rtw89/mac80211.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.11.5-common/Makefile:1951: /home/kkiragu/Documents/git/rtw89] Error 2
make[1]: *** [/usr/src/linux-headers-6.11.5-common/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.11.5-amd64'
make: *** [Makefile:104: all] Error 2``

a5a5aa555oo, 2024-11-24:
Larry passed away a few months ago, so this repo is no longer maintained.
Doesn't the rtw89 driver in kernel 6.11 work for you?

closed 2024-12-03 by sahl36

iiivwviii, 2024-12-25:
This error appeared on Kali Linux kernel 6.11.2. Is there a solution?

a5a5aa555oo, 2024-12-25:
Hi @iiivwviii,

There are two solution available for you.
1. Use the built-in rtw89 in your kernel which is even newer than the one here.
2. Use my rtw89 driver which is backported from kernel 6.13.

iiivwviii, 2024-12-26:
unfortunately, these methods do not help (
Yes, all drivers are installed correctly, there are no errors during the installation process, but after restarting the PC, the Internet still works unstable, it may appear for a while, and then disappears again.  There is a wi-fi connection, and when you try to execute a ping command to any site, there are long delays (1000ms) and severe packet loss, although there is a connection. Something is not working as it should.
Any idea why this might be?

a5a5aa555oo, 2024-12-26:
@iiivwviii
 Is your chip a RTL8852BE, too? Is there any error message in your dmesg? Could you  post the full dmesg?
btw, the rtw89 driver here will work on kernel 6.11 if [this patch](https://github.com/lwfinger/rtw89/pull/366) is applied.

iiivwviii, 2024-12-26:
@a5a5aa555oo
[dmesg.txt](https://github.com/user-attachments/files/18252486/dmesg.txt)
Everything seems to be fine in dmesg. There are no errors related to rtw89.

a5a5aa555oo, 2024-12-26:
> @a5a5aa555oo [dmesg.txt](https://github.com/user-attachments/files/18252486/dmesg.txt) Yes, my chip is RTL8852BE. Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller Everything seems to be fine in dmesg. There are no errors related to rtw89. [patch for mac80211.c](https://github.com/lwfinger/rtw89/pull/366) did not help me.

I tested my rtw89 driver on Kali LiveCD 2024.4 and no problem found, so...no idea what is wrong.

[image]

iiivwviii, 2024-12-27:
@a5a5aa555oo 
I tried to reinstall the system completely, but the problem remains(.
Both the built-in kernel and your driver refuse to work((. 
I guess I'll just have to change the adapter.

a5a5aa555oo, 2024-12-27:
@iiivwviii 
I am curious about what is the signal strength shown in [wavemon](https://pkg.kali.org/pkg/wavemon)? 
Did you ever try the older Kali LiveCD which has kernel 6.8 or 6.10? does your 8852BE work fine in these older release?

iiivwviii, 2024-12-27:
@a5a5aa555oo 
I'm currently using version 6.8.11. Everything is working flawlessly.
[image]

a5a5aa555oo, 2024-12-27:
It looks like your 8852BE card is not broken, maybe you can email [linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org) to ask Realtek engineers for help as mentioned in [README.md](https://github.com/lwfinger/rtw89/blob/main/README.md), that is the right place for you.

iiivwviii, 2024-12-27:
@a5a5aa555oo Thanks for the help! I'll try to turn to linux-wireless@vger.kernel.org.