lwfinger/rtw89 #372: Computer restarts on shutdown with PCI (TP Link AX1800, Archer TX20E)

soypat, 2025-01-04:
I have a  TP Link AX1800 (Archer TX20E, running `rtw89_8852be`. 
When I go to shutdown my computer and this device is plugged into PCIe port the computer restarts.

This happens with linux kernel version 6. On Kernel 5 version computers this does not happen. 

This happens with Endeavour OS and Linux Mint latest distros at the time of writing this.

a5a5aa555oo, 2025-01-04:
Hi,

This problem will be fixed in kernel 6.14. Before 6.14 is released, you can use [my driver](https://github.com/a5a5aa555oo/rtw89) which already fixes it temporarily.
If you are on Endeavour OS, you can simply install the package [rtw89-dkms-git](https://aur.archlinux.org/packages/rtw89-dkms-git) in AUR.

a5a5aa555oo, 2025-01-04:
[This commit](https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/?id=9c1df813e08832c3836c254bc8a2f83ff22dbc06) fixes it and will be in kernel 6.14.