lwfinger/rtw89 #373: Make issue in Fedora 41

jason0848, 2025-01-06:
Using gcc-0:14.2.1-3.fc41.x86_64 and kernel 6.12.7-200.fc41.x86_64 and I'm getting the following when I use the make command:
'''
make -C /lib/modules/6.12.7-200.fc41.x86_64/build M=/home/jason0848/Downloads/rtw89 modules
make[1]: Entering directory '/usr/src/kernels/6.12.7-200.fc41.x86_64'
  CC [M]  /home/jason0848/Downloads/rtw89/core.o
  CC [M]  /home/jason0848/Downloads/rtw89/mac80211.o
/home/jason0848/Downloads/rtw89/mac80211.c:1177:35: error: initialization of ‘void (*)(struct ieee80211_hw *, bool)’ {aka ‘void (*)(struct ieee80211_hw *, _Bool)’} from incompatible pointer type ‘void (*)(struct ieee80211_hw *)’ [-Wincompatible-pointer-types]
 1177 |         .stop                   = rtw89_ops_stop,
      |                                   ^~~~~~~~~~~~~~
/home/jason0848/Downloads/rtw89/mac80211.c:1177:35: note: (near initialization for ‘rtw89_ops.stop’)
make[3]: *** [scripts/Makefile.build:229: /home/jason0848/Downloads/rtw89/mac80211.o] Error 1
make[2]: *** [/usr/src/kernels/6.12.7-200.fc41.x86_64/Makefile:1977: /home/jason0848/Downloads/rtw89] Error 2
make[1]: *** [Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.12.7-200.fc41.x86_64'
make: *** [Makefile:104: all] Error 2
'''

a5a5aa555oo, 2025-01-07:
The built-in rtw89 driver in the kernel 6.12 is newer than the one in this repo, doesn't it work for you?

bpoteet, 2025-01-22:
I can confirm this is happening Debian 12 on 6.11.10+bpo-rt-amd64 from backports as well. For me I am unable to use the 6.12 backports version as it is failing for another driver and I can't clear that bug so I backed down to 6.11.

```
Loading new rtw89-1.0.2 DKMS files...
Building for 6.11.10+bpo-rt-amd64
Building initial module for 6.11.10+bpo-rt-amd64
Error! Bad return status for module build on kernel: 6.11.10+bpo-rt-amd64 (x86_64)
Consult /var/lib/dkms/rtw89/1.0.2/build/make.log for more information.
dpkg: error processing package rtw89-dkms (--configure):
 installed rtw89-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 rtw89-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
```
DKMS make.log for rtw89-1.0.2 for kernel 6.11.10+bpo-rt-amd64 (x86_64)
Tue Jan 21 10:31:51 PM EST 2025
make: Entering directory '/usr/src/linux-headers-6.11.10+bpo-rt-amd64'
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/core.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/mac80211.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/mac.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/mac_be.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/phy.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/phy_be.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/fw.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/cam.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/efuse.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/efuse_be.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/regd.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/sar.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/coex.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/ps.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/chan.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/debug.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/ser.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/wow.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/acpi.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8851b.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8851b_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8851b_rfk.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8851b_rfk_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8851be.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852a.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852a_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852a_rfk.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852a_rfk_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852ae.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852b.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852b_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852b_rfk.o
/var/lib/dkms/rtw89/1.0.2/build/mac80211.c:1177:35: error: initialization of ‘void (*)(struct ieee80211_hw *, bool)’ {aka ‘void (*)(struct ieee80211_hw *, _Bool)’} from incompatible pointer type ‘void (*)(struct ieee80211_hw *)’ [-Werror=incompatible-pointer-types]
 1177 |         .stop                   = rtw89_ops_stop,
      |                                   ^~~~~~~~~~~~~~
/var/lib/dkms/rtw89/1.0.2/build/mac80211.c:1177:35: note: (near initialization for ‘rtw89_ops.stop’)
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852b_rfk_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852be.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852c.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852c_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852c_rfk.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852c_rfk_table.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8852ce.o
  CC [M]  /var/lib/dkms/rtw89/1.0.2/build/rtw8922a.o
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.11.10+bpo-common-rt/scripts/Makefile.build:249: /var/lib/dkms/rtw89/1.0.2/build/mac80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/usr/src/linux-headers-6.11.10+bpo-common-rt/Makefile:1951: /var/lib/dkms/rtw89/1.0.2/build] Error 2
make: *** [/usr/src/linux-headers-6.11.10+bpo-common-rt/Makefile:236: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.11.10+bpo-rt-amd64
```

a5a5aa555oo, 2025-01-22:
@bpoteet 

Larry passed away last year, so the driver in this repo is no longer maintained.

I am curious about what chip do you have?

If you want to use the driver in this repo, you need this patch.

```patch
diff --git a/mac80211.c b/mac80211.c
index 0330cfd..30b9983 100644
--- a/mac80211.c
+++ b/mac80211.c
@@ -66,7 +66,11 @@ static int rtw89_ops_start(struct ieee80211_hw *hw)
 	return ret;
 }
 
+#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 11, 0)
 static void rtw89_ops_stop(struct ieee80211_hw *hw)
+#else
+static void rtw89_ops_stop(struct ieee80211_hw *hw, bool suspend)
+#endif
 {
 	struct rtw89_dev *rtwdev = hw->priv;
 
-- 
2.48.1

```