lwfinger/rtw89 #375: Cannot install on Debian GNU/Linux 12 (bookworm) with device Realtek RTL8852CE PCIe 802.11ax Wireless Network

jack5github, 2025-01-25:
Steps followed:

```
sudo apt-get update
sudo apt-get install make gcc linux-headers-$(uname -r) build-essential git
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
make
sudo make install
sudo update-initramfs -u
sudo reboot
```

Output of `inxi -Fxz` before and after performing the above steps:

```
System:
  Kernel: 6.1.0-30-amd64 arch: x86_64 bits: 64 compiler: gcc v: 12.2.0
    Desktop: Xfce v: 4.18.1 Distro: Debian GNU/Linux 12 (bookworm)
Machine:
  Type: Desktop Mobo: Gigabyte model: B650 GAMING X AX V2 v: x.x
    serial: <superuser required> UEFI: American Megatrends LLC. v: F31
    date: 08/14/2024
CPU:
  Info: 8-core model: AMD Ryzen 7 7700 bits: 64 type: MT MCP arch: Zen 4
    rev: 2 cache: L1: 512 KiB L2: 8 MiB L3: 32 MiB
  Speed (MHz): avg: 3266 high: 5140 min/max: 3000/5388 boost: enabled cores:
    1: 2907 2: 3000 3: 3000 4: 3000 5: 3000 6: 3000 7: 3800 8: 5140 9: 3000
    10: 3000 11: 3000 12: 3000 13: 3000 14: 3000 15: 3610 16: 3808
    bogomips: 121364
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: AMD Navi 33 [Radeon RX 7700S/7600S] vendor: Gigabyte
    driver: amdgpu v: kernel arch: RDNA-3 bus-ID: 03:00.0
  Device-2: AMD Raphael vendor: Gigabyte driver: amdgpu v: kernel
    arch: RDNA-2 bus-ID: 12:00.0 temp: 42.0 C
  Display: x11 server: X.Org v: 1.21.1.7 driver: X: loaded: amdgpu
    unloaded: fbdev,modesetting,radeon,vesa dri: radeonsi gpu: amdgpu
    resolution: 1920x1080~60Hz
  API: OpenGL v: 4.6 Mesa 22.3.6 renderer: AMD Radeon Graphics (gfx1102
    LLVM 15.0.6 DRM 3.49 6.1.0-30-amd64) direct-render: Yes
Audio:
  Device-1: AMD Navi 31 HDMI/DP Audio driver: snd_hda_intel v: kernel
    bus-ID: 03:00.1
  Device-2: AMD Rembrandt Radeon High Definition Audio driver: snd_hda_intel
    v: kernel bus-ID: 12:00.1
  Device-3: AMD Family 17h/19h HD Audio vendor: Gigabyte
    driver: snd_hda_intel v: kernel bus-ID: 12:00.6
  API: ALSA v: k6.1.0-30-amd64 status: kernel-api
  Server-1: PulseAudio v: 16.1 status: active
Network:
  Device-1: Realtek RTL8852CE PCIe 802.11ax Wireless Network vendor: AzureWave
    driver: N/A port: d000 bus-ID: 09:00.0
  Device-2: Realtek RTL8125 2.5GbE vendor: Gigabyte driver: r8169 v: kernel
    port: c000 bus-ID: 0a:00.0
  IF: enp10s0 state: up speed: 1000 Mbps duplex: full mac: <filter>
Bluetooth:
  Device-1: IMC Networks Bluetooth Radio type: USB driver: btusb v: 0.8
    bus-ID: 1-9:4
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter>
Drives:
  Local Storage: total: 5.46 TiB used: 84.44 GiB (1.5%)
  ID-1: /dev/nvme0n1 vendor: Kingston model: SFYRD2000G size: 1.82 TiB
    temp: 47.9 C
  ID-2: /dev/sda vendor: Western Digital model: WD40EZAX-00C8UB0
    size: 3.64 TiB
Partition:
  ID-1: / size: 1.76 TiB used: 54.8 GiB (3.0%) fs: ext4 dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 299.4 MiB used: 5.8 MiB (2.0%) fs: vfat
    dev: /dev/nvme0n1p1
Swap:
  ID-1: swap-1 type: partition size: 33.51 GiB used: 0 KiB (0.0%)
    dev: /dev/nvme0n1p3
Sensors:
  System Temperatures: cpu: 51.2 C mobo: N/A
  Fan Speeds (RPM): N/A
  GPU: device: amdgpu temp: 38.0 C fan: 0 device: amdgpu temp: 42.0 C
Info:
  Processes: 379 Uptime: 0m Memory: 30.47 GiB used: 3.32 GiB (10.9%)
  Init: systemd target: graphical (5) Compilers: gcc: 12.2.0 Packages: 2552
  Shell: Bash v: 5.2.15 inxi: 3.3.26
```

a5a5aa555oo, 2025-02-08:
Larry passed away last year, so this repo is no longer maintained.
You can try my rtw89 driver, it works with RTL8852BE on Arch Linux (kernel version: 6.1.123-1-lts61)

https://github.com/a5a5aa555oo/rtw89/tree/6.6-lts

a5a5aa555oo, 2025-02-08:
There is another easier and better solution available: update your kernel to 6.12 and use the built-in rtw89 driver in the kernel.

Steps:
1. Add this line to /etc/apt/sources.list
`deb http://deb.debian.org/debian bookworm-backports main`
2. install kernel 6.12
`sudo apt update`
`sudo apt install -t bookworm-backports linux-image-amd64 linux-headers-amd64`

Reboot.

https://backports.debian.org/Instructions/

closed 2026-09-03 by jack5github