lwfinger/rtw89 #377: Loading 8852BE causes issue in Linux kernel 5.17.13 in Debian Installer (d-i) netboot for Bookworm.

Metztli, 2025-02-05:
Utilized your modules during an older kernel build with GCC12 for Bookworm -- including the Debian Installer (d-i) UDEBs to create a custom netboot image.

Fact is, when d-i loads the module for the 8852BE WiFi 6 Realtek card the user interface installation hangs. Opening a virtual tty and doing:
`lspci -vnn -d::02xx` output is:
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller [10ec:b852]
        Subsystem: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller [10ec:b852]
        Flags: bus master, fast devsel, latency 0, IRQ 68, IOMMU group 9
        I/O ports at f000 [size=256]
        Memory at fce00000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, IntMsgNum 0
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-88-52-01
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Kernel driver in use: rtw89_8852be
        Kernel modules: rtw_8852be

`lsmod | grep -i ".*rtw.*"` yields:
rtw_8852be             16384  0
rtw_8852b             360448  1 rtw_8852be
rtw89pci               81920  1 rtw_8852be
rtw89core             598016  2 rtw89pci,rtw_8852b
mac80211             1081344  2 rtw89pci,rtw89core
cfg80211             1060864  3 rtw89core,mac80211,rtw_8852b

`dmesg` relevant section:
[  242.689149] INFO: task kworker/4:1:140 blocked for more than 120 seconds.
[  242.689151]       Tainted: G            E     5.17.0-3+reizer4.0.2-amd64 #1 Debian 5.17.13-1+reizer4.0.2
[  242.689153] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  242.689154] task:kworker/4:1     state:D stack:    0 pid:  140 ppid:     2 flags:0x00004000
[  242.689157] Workqueue: events_power_efficient reg_check_chans_work [cfg80211]
[  242.689180] Call Trace:
[  242.689181]  <TASK>
[  242.689183]  __schedule+0x35b/0x9b0
[  242.689188]  schedule+0x4e/0xc0
[  242.689189]  schedule_preempt_disabled+0x14/0x20
[  242.689191]  __mutex_lock.constprop.0+0x2ea/0x490
[  242.689193]  reg_check_chans_work+0x2d/0x400 [cfg80211]
[  242.689211]  ? finish_task_switch.isra.0+0xc1/0x2f0
[  242.689214]  ? __switch_to+0x108/0x430
[  242.689216]  process_one_work+0x1c7/0x380
[  242.689220]  worker_thread+0x4d/0x380
[  242.689222]  ? _raw_spin_lock_irqsave+0x25/0x50
[  242.689224]  ? process_one_work+0x380/0x380
[  242.689225]  kthread+0xe9/0x110
[  242.689227]  ? kthread_complete_and_exit+0x20/0x20
[  242.689229]  ret_from_fork+0x22/0x30
[  242.689232]  </TASK>
[  242.689233] INFO: task kworker/8:1:144 blocked for more than 120 seconds.
[  242.689233]       Tainted: G            E     5.17.0-3+reizer4.0.2-amd64 #1 Debian 5.17.13-1+reizer4.0.2
[  242.689234] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  242.689235] task:kworker/8:1     state:D stack:    0 pid:  144 ppid:     2 flags:0x00004000
[  242.689236] Workqueue: events linkwatch_event
[  242.689239] Call Trace:
[  242.689239]  <TASK>
[  242.689240]  __schedule+0x35b/0x9b0
[  242.689242]  schedule+0x4e/0xc0
[  242.689243]  schedule_preempt_disabled+0x14/0x20
[  242.689244]  __mutex_lock.constprop.0+0x2ea/0x490
[  242.689246]  linkwatch_event+0xe/0x40
[  242.689247]  ? __linkwatch_run_queue+0x220/0x220
[  242.689248]  process_one_work+0x1c7/0x380
[  242.689250]  worker_thread+0x4d/0x380
[  242.689252]  ? _raw_spin_lock_irqsave+0x25/0x50
[  242.689253]  ? process_one_work+0x380/0x380
[  242.689255]  kthread+0xe9/0x110
[  242.689256]  ? kthread_complete_and_exit+0x20/0x20
[  242.689258]  ret_from_fork+0x22/0x30
[... 107 lines cut ...]
[  484.353210]       Tainted: G            E     5.17.0-3+reizer4.0.2-amd64 #1 Debian 5.17.13-1+reizer4.0.2
[  484.353211] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  484.353212] task:kworker/8:1     state:D stack:    0 pid:  144 ppid:     2 flags:0x00004000
[  484.353213] Workqueue: events linkwatch_event
[  484.353214] Call Trace:
[  484.353215]  <TASK>
[  484.353215]  __schedule+0x35b/0x9b0
[  484.353217]  schedule+0x4e/0xc0
[  484.353218]  schedule_preempt_disabled+0x14/0x20
[  484.353219]  __mutex_lock.constprop.0+0x2ea/0x490
[  484.353221]  linkwatch_event+0xe/0x40
[  484.353222]  ? __linkwatch_run_queue+0x220/0x220
[  484.353223]  process_one_work+0x1c7/0x380
[  484.353225]  worker_thread+0x4d/0x380
[  484.353227]  ? _raw_spin_lock_irqsave+0x25/0x50
[  484.353228]  ? process_one_work+0x380/0x380
[  484.353230]  kthread+0xe9/0x110
[  484.353231]  ? kthread_complete_and_exit+0x20/0x20
[  484.353232]  ret_from_fork+0x22/0x30
[  484.353234]  </TASK>

[image]
[image]

Any insight would be appreciated.

a5a5aa555oo, 2025-02-08:
Larry passed away last year, so this repo is no longer maintained.
You can try my rtw89 driver, it works with RTL8852BE on Arch Linux (kernel version: 5.15.174-1-lts515)

https://github.com/a5a5aa555oo/rtw89/tree/6.6-lts

Metztli, 2025-02-12:
> Larry passed away last year,

I am sorry to hear that!

>so this repo is no longer maintained. You can try my rtw89 driver, it works with RTL8852BE on Arch Linux (kernel version: 5.15.174-1-lts515)
> 
> https://github.com/a5a5aa555oo/rtw89/tree/6.6-lts

Thank you, @a5a5aa555oo ! I hacked your modules into my build of kernel 5.17.13-2+reiser4.0.2, replacing the original/'official' rtw89 module set directory with yours, and then I created my custom Debian Installer (d-i), [Metztli Reiser4](https://sourceforge.net/projects/metztli-reiser4/). This d-i then was enabled to successfully detect the rtw89 wireless device and I was able to install Debian from the network. Hard to believe, as I had spent many days in this task, but it worked like magic!

Thank you once again!

closed 2025-02-12 by Metztli